MarlinFirmware/Marlin #5005: G92 Actual Value?

sarf2k4, 2016-10-12:
I often use "g92 z10" to lower my printer head, apparently this command does nothing. Although the value returned is correct which is z = 10mm from g92 z10 earlier, but issuing "g0 z0" after that does nothing. 

I tried with g90 and g91, both with no luck.

I suspect that there are 2 separate codes that holds their own set of value, When the g92 were issued, it gets displayed and returned correctly, but the "real" value are still at 0 value, in summary, not changed

manianac, 2016-10-17:
There were changes a few versions back that enforced software endstops, so even if you change current position with G92, you can't go outside where the endstops reported.

I don't know your printer setup, but if you have a Z probe, you should look into all the new options there to prevent your head from crashing.

thinkyhead, 2016-10-22:
If you want to be able to lower the nozzle below the Z home coordinate (e.g., 0), then you can set `Z_MIN_POS` to a number lower than your `MANUAL_Z_HOME_POS` setting. This will move the Z min software endstop to a lower position.

closed 2016-10-22 by thinkyhead