MarlinFirmware/Marlin #5007: New Gcode For Priming?

sarf2k4, 2016-10-12:
Is it relevant to add a new gcode to prime the nozzle on to the bed? 

For example lowers the nozzle down to 0.2mm, move certain distance while extruding plastic outside of the printable area region.

Of course, several extra options would be added, more like a macro to prime the nozzle, probably g30 or g32 something?

FHeilmann, 2016-10-12:
Marlin strives to keep as much backwards compatibility with the original intent of G-Code (mostly CNCs) as possible, and I believe G30 and G32 are already taken.

Moreover, the time it would take you to configure the location of the prime, compile, flash firmware, or come up with the correct parameters to a new command is better spent in writing those ~3 lines of G-code to do it manually.

I've done it once, if you need help let me know!

Roxy-3D, 2016-10-12:
If you load up the devel_ubl branch, there is a GCode to produce a Mesh Validation Pattern.  Its purpose is to leave a visual indicator on the bed to help the user edit the Mesh to perfection.   One of its parameters is a 'P' for 'Prime' function.    Do a G28, and then say G26 P     That will prime until you stop the action with an LCD Click.   You can also specify the amount of desired prime with a G26 P 2.5   to indicate 2.5mm of filament.

If you want just a Prime function and not an entire Mesh Validation Pattern, it would not be difficult to move that part of the code to any suitable GCode number.

thinkyhead, 2016-10-22:
I do this just by adding extra prefix GCode to my slicer (Cura) profile. The extra code simply makes a long extrusion line across the front of the bed as soon as the nozzle heating is done.

closed 2017-10-31 by thinkyhead