MarlinFirmware/Marlin #5009: Homing Bump / BLTouch (G28, G29, M48)

gcormier, 2016-10-12:
(Continuation of PR 4002)

Behavior I am seeing

> #define Z_CLEARANCE_DEPLOY_PROBE    5 // Z Clearance for Deploy/Stow
> #define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
> ..
> #define Z_HOME_BUMP_MM 2
 
**G28 works correctly. Plate strikes on G29/M48.**

> #define Z_CLEARANCE_DEPLOY_PROBE    5 // Z Clearance for Deploy/Stow
> #define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
> ..
> #define Z_HOME_BUMP_MM 5

**All three function correctly, with enough clearance before pin is deployed.**

However, according to the code
> #if HOMING_Z_WITH_PROBE
>   (axis == Z_AXIS) ? max(Z_CLEARANCE_BETWEEN_PROBES, home_bump_mm(Z_AXIS)) :
> #endif

This shouldn't be the case. From my understanding, the value used in the code should be 5.

thinkyhead, 2016-10-22:
That would imply that `G28` is setting the Z position wrong, because `G29` and `M48` don't even use `Z_HOME_BUMP_MM`. Do you see a different current Z position at the end of `G28` when using `Z_HOME_BUMP_MM` of `2` versus `5`?

boelle, 2016-11-05:
@gcormier 

"Do you see a different current Z position at the end of G28 when using Z_HOME_BUMP_MM of 2 versus 5?"

gcormier, 2016-11-05:
Sorry - I've uninstalled the BLTouch. I have a replacement on the way, so I can try to hook it up and see how it performs when I get it, but that's probably 3-5 weeks as it's coming from Korea. If @bgort still has his he might be able to test.

Roxy-3D, 2016-11-05:
>  I have a replacement on the way, so I can try to hook it up and see how it performs when I get it,

Please let us know how the new BLTouch performs.   There are a number of us that are very interested in how good they are, and how they typically fail.

boelle, 2016-11-09:
@bgort you still have yours?

bgort, 2016-11-09:
I do.  It's working fine, afaik.

I can check this out in a bit, though.

boelle, 2016-11-09:
@Roxy-3D @thinkyhead does this looks like its solved and can be closed?

or should he retest on latest RCBugfix if he have not allready done so?

bgort, 2016-11-09:
Testing ca7275a (old build I'd been using) with:

```
#define Z_CLEARANCE_DEPLOY_PROBE   5 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
#define PROBE_DOUBLE_TOUCH
#define Z_HOME_BUMP_MM 5
```

And all is working fine.  

`#define Z_HOME_BUMP_MM 2` works as expected, too.

Testing with latest RCBugFix (06f2282) yields the same.  Everything seems to work as advertised, for me.  No strikes anywhere.

FWIW, I'm seeing <5um repeatability, generally, today.  Mostly ~4um on successive `M48`s, with some 2s and 3s sprinkled in.  Really happy with my BLTouch now that the kinks are ironed out.

Roxy-3D, 2016-11-09:
> FWIW, I'm seeing <5um repeatability, generally, today. Mostly ~4um on successive M48s, with some 2s and 3s sprinkled in. Really happy with my BLTouch now that the kinks are ironed out.

Was it difficult to get them to replace the bad unit?   Did you have to send the failed unit back to them?

bgort, 2016-11-09:
>Was it difficult to get them to replace the bad unit? Did you have to send the failed unit back to them?

My unit didn't go bad, but their support was good when I used it.  I believe @gcormier is waiting on a replacement.

gcormier, 2016-11-09:
I've just received my replacement yesterday! I'm not overly enthused about wiring it back up as my inductive sensor has been performing flawlessly... I will try to get to it this weekend however for the greater good :)

boelle, 2016-11-12:
@gcormier please let us know what you figure out :-D

would be nice if we could close this one before RC8 is going to be let loose to the public

gcormier, 2016-11-12:
Looks like it's all fixed up.

closed 2016-11-12 by gcormier