MarlinFirmware/Marlin #5010: M428 does not correctly take account of MANUAL_Z_HOME_POS

andyjbm, 2016-10-12:
This is true for any axis but for me it was on the Z. The code in question handles all axis so fixing it fixes for all.

If there is a MANUAL_Z_HOME_POS value set in configuration.h (as well as using M206 to set offsets), then M428 does not include MANUAL_Z_HOME_POS in the calculation of the new offset.

In Marlin_main.cpp in:
`inline void gcode_M428()`

a few lines in the line:
`diff = current_position[i] - LOGICAL_POSITION(base, i);`

Should be:
`diff = current_position[i] - LOGICAL_POSITION(base, i) - base_home_pos(i);`

as only the value for M206 (set_home_offset) is going to be changed. We do not want to add MANUAL_Z_HOME_POS to set_home_offset unless we are also going to set MANUAL_Z_HOME_POS to zero, which as the latter is hardcoded in firmware, isn't going to happen when M428 is called.

Thanks,

Andy.

thinkyhead, 2016-10-22:
I don't doubt there's a problem here, but I'm not sure that's the full solution. The intent is to set the offsets based only on the software endstops closest to the current position. This is made a little more complicated with the way `G92` can now move the software endstops around. It would seem to make the most sense to use the `soft_endstop_min` and `soft_endstop_max` values directly. However, these values both include the current value of `home_offsets`, so that needs to be considered too.

I'll look at this more closely and see if I can come up with a patch that does the right thing in all cases.

andyjbm, 2016-11-04:
But shouldn't M428 just be changing the offset (as set by M206) by the difference in current position from the home position? 

With the axis at the home position (ie there should be no change to M206) executing M428 adds (subtracts?) MANUAL_?_HOME_POS to the M206 offset (changing M206 value by this amount) when it should leave the M206 value as is!

I was trying to use this via the LCD menu option to adjust offsets (which just calls M428) as a quick way of modifying M206 Z without having to hook up the laptop and issue M206 Z? directly.

So what I was trying to do all via LCD was:
G28 Z (Autohome)
Move Z by small amount (Control menu? can't remember now) say 0.2 mm.
Execute Adjust Offset menu option.
Save EEPROM (menu option)

But because I had a MANUAL_Z_HOME_OFFSET of 25 set in firmware this was not working as it was outside the M428 2cm limit. When I found that limit and removed it I discovered the MANUAL_Z_HOME_OFFSET was being included in the new value for M206 Z rather than just changing the M206 Z by the 0.2mm I was after.

Ideally I'd just like to be able to edit M206 on the LCD but that's a feature request, not a bugfix!

Andy.

thinkyhead, 2016-11-04:
> difference in current position from the home position?

Well, I designed it to use the offset from the nearest end, but presumably most will use it in close proximity to their configured endstops.

andyjbm, 2016-11-16:
Ok maybe I misunderstand the use of m428 then and should be using G92? But M428 is accessible through the LCD Menu and G92 isn't!

Let me explain what I'm doing - and along the wayI think I discovered another bug but at the time I thought I just didn't understand what I should be doing.

So I have a Cartesian XY bot, Z has dual steppers (another story) and dual MIN endstops (a longer story).

The Z MIN endstops operate at about Z=11mm from the bed (I'll stick with Z=0 being the physical print bed). So when I first set this all up I thought I could use M206 Z11 to position the Z home endstop at Z=11 making Z=0 correctly put the nozzle at the print bed. This all seemed to display correctly on the LCD with Z=11 after G28 Z was complete, but sending G1 Z0 (or anything less than G1 Z11) did nothing - it would appear the software min Z endstop of 0 was stopping this from working. Setting the software Z min endstop to <-11 didn't help - no difference. But I discovered that using MANUAL_Z_HOME_POS instead of M206 worked and now G1 Z0 was working properly.

Now changing MANUAL_Z_HOME_POS to tweek the bed z=0 position is tedious as it requires a firmware flash every time, so I set MANUAL_Z_HOME_POS to 25 (above and beyond what I needed) and flashed that, then used M206  Z-14 to get an endstop position of 25-14 = 11mm.

Even better, this M206 could be tweeked via M428 (and the LCD without hooking up the laptop)  - but  that's where I stumbled across this problem I opened this issue for.

SInce I've been getting used to this approach I also think the sign of diff is wrong. Say The nozzle is homing too high and Z=0 is actually 1mm lower than G1 Z0 would place it,  I would home the printer, then move axis Z to 12mm (from current home pos of 11mm) and then execute M428 (adjust offsets). But the 1mm change goes the wrong way. The new position of the Z endstop is now Z=10mm, not Z=12mm.

So changing the sign of diff from my original change suggestion would fix this - or I'm using and abusing a feature that was intended for something completely different?

To help you in your blind code changes I can verify that the mods I first posted are working for me as I have described the use case above.

Thanks for your help,

Andy.

thinkyhead, 2016-11-17:
`M428` is meant to be an automated variant of `M206`, and in fact it does "tweak" it if it's been applied before. I'll peek at the code tomorrow night.

thinkyhead, 2016-11-17:
> The Z MIN endstops operate at about Z=11mm from the bed

Are these opto endstops, then?

andyjbm, 2016-11-18:
Ok So:
>>Are these opto endstops, then?
No but for all intents and purposes they work in the same way, allowing the Z to keep moving after the switch has been triggered.

>>The correct way to have a Z min endstop (in your case, 11) different from the Z min position (in your >>case, 0) is to set Z_MIN_POS to 0 and MANUAL_Z_HOME_POS to 11.

..and this is exactly how I started out **but** tweeking the Z=0 position then requires a firmware flash every time to change this. I'm finding that I need to tweek this +/- 0.3mm or so every so many prints to get the best first layer, and using M206 is much much faster. BUT (again) M206 requires the laptop be hooked up to change it. I'm using SDCards and an LCD and was using M428 via the LCD menu as a mechanism to change M206. There is no current way to change M206 via LCD other than M428 is there?

>>Note, however, that your setup breaks homing because the Z carriage can go below the Z endstops! 
Yes/No/Maybe?
So what is the difference between an endstop (software or Hardware) and the home position?
I was thinking that the mechanical endstop = the home position which might be displaced from the Z=0 position. the software endstop would be at Z=0 thus preventing mechanical damage.

>> Z carriage can go below the Z endstops!
Yes that's right. Isn't this the home position, which isn't necessarily the Z=0 position?
So the offset in M206 adds an ... well... offset, so that once homing is complete the firmware knows it is at Z=11 and G28 can only step down as far as Z=0 and not beyond. The software endstop which is at Z=0 and NOT the homing position (hardware endstop) stops the nozzle crashing into the bed if G28 Z-1 were possible. This is what is happening in practice.

>>Therefore G28 cannot assume that moving Z down will eventually reach the endstops. If you do G28 >>while Z is under 11mm, the steppers will just grind away with the nozzle pressing against the bed.
Yes absolutely. That was another mod I've done which I'll post when I get a moment. I modded homing so that if G28 was executed and the endstops are already triggered (say the head is at Z=6mm) the carriage is moved in the +ve direction first (or -ve for MAX endstops) to untrigger the endstops, then homing begins for real. I think it used to just step away by a few mm or so and then stop thinking it was homed. Now it actively looks for the endstop untriggering. I do this by flipping the endstop_inverting flag etc. I'll post the whole mod to main.cpp and endstops.cpp as another thread when I get chance. (unless you'd like it here?)

>> You should move your endstops (or the "flag") so that they trigger when Z is at the bed level.
Yes I figured this is the way it is intended to use endstops in the firmware but mechanical adjustment is tedious and nothing like as accurate as modding M206 Z? by +/-0.1mm every time I want to print.
Each time I change the tape/prittstick glue or whatever on the bed I'm checking the level, looking at the Z=? on the LCD and if its not 0.00mm then changing the home position via LCD by this amount. Its a one step fix and bang on accurate every time, rather than mucking about with mechanical adjustment.

 Here are some pics of the Z endstop and of the very 1st print! The Z drive has since been changed from fishing line (which kept slipping) to threaded bar (much better).

Yes I know it looks horrendous (we call it the tierap as most of it is held together by tiewraps) but as bad as it looks it has already printed two printrbots (last pic) which are currently being assembled.

[image]
[image]
[image]
[image]
[image]
[image]

thinkyhead, 2016-11-19:
Now that's a real RepStrap.

thinkyhead, 2016-11-19:
> but tweeking the Z=0 position then requires a firmware flash every time

`M206` will be the better option, then. It does get saved in EEPROM with `M500` also. How often do you need to adjust your endstops, really?

> There is no current way to change M206 via LCD other than M428

Correct. And I'm not so thrilled about `M428` even though I wrote it. It's "weird" compared to most codes.

> if G28 was executed and the endstops are already triggered…

I see. So the endstop is triggered at all points below ~11mm?

> So what is the difference between an endstop (software or Hardware) and the home position?

The hardware endstops _are_ the "home position," and they can be wherever. As long as your GCode is in agreement with your bed coordinates, everything will work as it should. The "software endstops" (i.e., movement limits) simply define the physical bounds beyond which carriages should not move.

thinkyhead, 2016-11-19:
Yep. As I examine M428 it seems antiquated, with poor assumptions. I'll see what I can do to make it smarter over the weekend.

andyjbm, 2016-11-21:
>>Now that's a real RepStrap.

Oh Yeah!

>>M206 will be the better option, then. It does get saved in EEPROM with M500 also. How often do you need to adjust your endstops, really? If it's fairly frequent then perhaps you can/should improve the physical "repeatability" of the endstops.

Being able to tweak m206 for Z via LCD would be utopia! ie to see the current value and just move up/down with the jog wheel... And yes I am saving to EEPROM via the LCD as well.

>>How often do you need to adjust your endstops, really? If it's fairly frequent then perhaps you can/should improve the physical "repeatability" of the endstops.

Well in all honestly its getting less and less as I get familiar with the machine. Initially I was using fishing wire on the Z steppers which turned out to be slipping as the z steppers got hot over time. I got involved with dual Z min endstops using M666 which eluded me into thinking I'd fixed it for a while but really it was a disaster. Then having switched to threaded rod I've seen a big improvement in consistency and the correction applied with M666 has not changed since.. I'm still using the dual ZMIN endstops and m666 as its working well and correcting the difference in Z steppers after fingers have been playing with them. 

 I think the remaining variation between endstop position and bed position is down to layers of prittstick or whatever building up over time and possibly a bit of flex in the hot end carriage. I'm only needing to change by +/- 0.2mm if at all, but in an effort to get that 1st layer to goddamn stick to the bed, tweaking that Z=0 position even by this small amount has been a useful tool.

>> ...So the endstop is triggered at all points below ~11mm?

Yes. Exactly. So my way of thinking is that the hardware endstop just gives us a reference of things physical relative to the bed position and then that relativity can be configured in software. (eg M206).
Now I realize the hardware endstop is intended as more of an absolute Z=0 indicator and isn't intended to be as arbitary as I've coearced it to be.

>>The hardware endstops are the "home position," and they can be wherever...

>>The "software endstops" (i.e., movement limits) simply define the physical bounds beyond which carriages should not move.

So how is the relationship set between the endstop "home position", G1 Z0 and "software endstops"?

I tried to wind Z below zero with the adjust axis option on the LCD menu and that allowed Z to go negative but I don't think G1 Z-1 did. I need to try that and confirm.

I do know that when I was first starting out and the hardware endstop was at Z=0 I tried to have a negative software endstop (Z=-11)  but I couldn't go below Z=0. I figured the hardware endstop was preventing this and that's when I discovered M206 and MANUAL_Z_HOME_POS. The rest is history!

andyjbm, 2016-11-21:
Just trying to upload my complete firmware for you @thinkyhead  but the thread is not allowing me to attach .zip even though it's indicating that should work:

>>We don’t support that file type.  Choose Files Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP.

Andy.

boelle, 2016-11-23:
@andyjbm would configuration.h and configuration_adv.h not be enough?

if you only changed a few files then use a site like pastebin.com and post the links from there?

andyjbm, 2016-11-23:
My Marlin c/w all changes. Here we go:

https://www.dropbox.com/sh/3hiz6f4oqf2puuj/AADEDUWbyXlbmBjUsKFmO9Gja?dl=0

There is much more than pertains to this thread here but I hope it's all useful. Please raise them in another thread if that is the way to go. I've just not had the time to offer this all back properly so here it is as is.

Ok so the changes that matter:

conditionals_post.h
Configuration.h
Configuration_adv.h

Endstops.h
Endstops.cpp
language.h
Marlin_main.cpp

It includes:

Fix to be able to use Zmax plug as a second Zmin endstop. The disabling of Zmax was also disabling Z2min.

Fixes to make dual min endstops work correctly. 
There was something amiss with Endstops::test_dual_z_endstops(?,?) in endstops.cpp that wouldn't allow for the use of Zmax as a second Zmin endstop.
I think this boils down to the fact that dual Z endstops were only ever coded to be used as a ZMAX. I wanted to use dual endstops as a ZMIN. So  #define Z2_IS_MAX  is added to configuration_adv.h. If it's undefined then Z2 is taken as being a min endstop.
Yes, Yes I should have done this the other way round so as not to break existing configurations that won't have it defined and are using Z2 as a max endstop! Hey Ho...

An issue where unused endstop pins were #UNDEFfed and re #define as -1 would take the Z2 pin definition with it. (macro pointing to a macro which was then changed).

Mod to allow the carriage to move away from a triggered endstop until it is untriggered, and then commence homing as usual. (Explained in this thread.) There's also a #define ENDSTOP_CLEAR_MM as a sanity just to limit how far the motors will go looking for untriggered endstops. Remember they are probably open circuit in the triggered state, which can't be distinguished from them not being connected to the ramps board at this point! So this prevents imminent disaster at the other end of the Z travel.

Mod to M428 as discussed in this thread.

Mod to M119 to include Z2min value.

As far as I'm able I think the changes I made to homing would work on other axis (X and Y) as well although I've only tested for Zmin.

Also my bot is a XY cartesian (dual Z steppers) no auto bed levelling or mesh. So I don't know how things would fair for deltas or coreXY.

Thanks,

Andy.

thinkyhead, 2016-11-26:
Dude, I'm not going to look at your source code. A better way to deal with it is to fork Marlin, apply your changes as commits, and then you can point to the interesting parts of those commits.

boelle, 2016-12-11:
@jbrazio @thinkyhead should this one not be closed due to inactivity?

closed 2016-12-13 by jbrazio