MarlinFirmware/Marlin #5011: EXTRUDER_AUTO_FAN_SPEED help?

gcormier, 2016-10-12:
I'm trying to turn down my extruder fan speed, as I've upgraded it to something with better specs, so it does not need to run at full speed. (It will be quieter).

At a first glance, it look like EXTRUDER_AUTO_FAN_SPEED in Configuration_adv.h will get me there.

My config is RAMPS1.4. The hotend fan is connected to a RAMPS Fan Extender. (E3D Chimera, so 1 fan for 2 hotends). The Fan Extender is shifted 2 pins away from the top, which lets me still use 2 servo pins - one for heated bed (solid state relay), one for BLTouch.

> #define HEATER_BED_PIN   11   // Servo pin 11 to drive the SSR
> #define EXTRUDER_0_AUTO_FAN_PIN 5 // Servo pin 5 for E3D Fan
> #define EXTRUDER_1_AUTO_FAN_PIN 5 // Servo pin 5 for E3D Fan

With EXTRUDER_AUTO_FAN_SPEED at 255, the fan comes on at 50 degrees as expected. There is a RED led inside my RAMPS case that lights up solid.

When I change EXTRUDER_AUTO_FAN_SPEED to 245, the fan no longer comes on. The red led flickers for a split second - at what I assume is as the probe hits 50 degrees - but remains off.

I've tried looking around for this quite a bit, both on the internet, and inside the code, but I can't seem to put it together.

manianac, 2016-10-12:
I'm not too familiar with the fan extender, but looks like two simple MOSFET extender.  Have you tried 
`M42 P5 S128` to attempt setting pin 5 (without auto fan turned on)?

Or have you tried test code with Arduino IDE to analogWrite the pin?

gcormier, 2016-10-12:
Hmm, that's a good idea. To effectively allow that M42 command, would I have to comment any any reference to that pin for fans?

Bob-the-Kuhn, 2016-10-13:
Foobar and double rats ---- the detailed explanation I left several hours ago has disappeared.  :-(

I think the problem is your system is using hardware PWMs for the fan and the heat bed but the timers that control those PWMs are in use by interrupt service routines so pins 5 & 11 can't only be used as general purpose digital I/O.

FYI - the BLTouch is a servo based probe so using it is why pin 5 can't be used as a hardware PWM.

The options I see are:
-  move to pins 6-10 or 44-46
-  use a software PWM  (disable FAST_PWM_FAN and enable FAN_SOFT_PWM)

If you use the software PWM then the fans & heaters can use any digital I/O pin. 

There's also a SLOW_PWM_HEATERS option meant for use with relays.  I expect that this is for mechanical relays.  Your solid state relay should be fast enough to work with the standard software PWMs.

gcormier, 2016-10-13:
Aha! Thanks Bob. That is something I dug up from an old thread but couldn't quite put it all together.

Is there any pros/cons in using SOFT PWM?

I could always move my RAMPS case fan (D8) to one of the non-PWM pins and then put the hotend fan on D8... but that's some rewiring. If SOFT PWM comes for "free" then I'll just change some bits.

Bob-the-Kuhn, 2016-10-13:
I'm into CNC so I've never used a fan.  

The only pros/cons I know of are what's mentioned in the config file.

thinkyhead, 2016-10-22:
Since the auto fans are not meant to be PWM or run at an alternate speed, they are simply connected to digital on/off pins, and as you noted, this means that 255 is on, while all other values are off. If you simply want the fan to run slower, you could solder a resistor inline with the fan.

gcormier, 2016-10-22:
Just to confirm, when I commented out the PIN for the extruder fan, I was then able to use PWM properly to control it.

thinkyhead, 2016-10-22:
@gcormier What are the exact changes you made to get it working? Should we regard this as a bug or just a matter of how some settings work?

gcormier, 2016-10-22:
Hmm... let me do the testing again and comment back while it's fresh in my mind. I'll try to get some results for you after lunch!

gcormier, 2016-10-22:
1) Stock Configuration
> //#define FAN_SOFT_PWM
> #define EXTRUDER_0_AUTO_FAN_PIN 5 
> #define FAN_PIN            		6

2) Test manual - PWM works. M106 P0 S255 - Full speed. S130 - slower.
> //#define FAN_SOFT_PWM
> //#define EXTRUDER_0_AUTO_FAN_PIN 5 
> #define FAN_PIN            		5

3) Test EXTRUDER_AUTO_FAN_SPEED. - PWM does not work. Fan does not engage.
> //#define FAN_SOFT_PWM
> #define EXTRUDER_0_AUTO_FAN_PIN 5 
> #define FAN_PIN            		6
> adv:
> #define FAN_KICKSTART_TIME 250
> #define EXTRUDER_AUTO_FAN_SPEED   180

This could be a bug? Unless @Bob-the-Kuhn can find a problem with my testing methodology? I leave it up to smarter people to decide. To me, Given test case #2 worked, that shows PWM should work regardless of the interrupt stuff mentioned?

thinkyhead, 2016-10-22:
@gcormier What is your board? Megatronics? Azteeg X3 Pro?

The reason I ask is because you cannot override `FAN_PIN` in `Configuration_adv.h` unless you also comment it out (or change it) in the pins file. However, you can override `FAN_PIN` in the `Configuration.h` file, because it is processed before `pins.h`.

Anyway, from your testing it makes perfect sense that the `FAN_PIN` can do PWM. There's actually code in Marlin for that. However, there is no software PWM implemented for `EXTRUDER_0_AUTO_FAN_PIN`. If you want variable speed, you have to attach it to a pin that can do hardware PWM.

gcormier, 2016-10-22:
RAMPS 1.4... but for my build (VORON) there is another header file that overrides some of the RAMPS settings.

I see, so FAN_PIN is always software PWM?

In that case, should we do something about EXTRUDER_AUTO_FAN_SPEED in sanity check if it is on certain pins which it would not work? If not supported, it should disallow anything other than 255 - or, ignore that setting, rather than have the fan not working?

thinkyhead, 2016-10-22:
I don't know of any easy way, given the data in our headers (e.g., `fastio.h`) and support for various chips, to test that the selected pin is PWM-able. But I will look into it. If nothing else, I could add that data for this purpose.

In response, all we can really do is throw a compiler warning, as I think throwing an error would be too severe. Unfortunately, most users will not see the warning because Arduino IDE hides them by default. So I think we may just have to add a note to the configuration file about that limitation.

The other thing we can do is to pre-assign auto fan pin numbers in the pins files. We've been gradually converting the pins files to set the recommended pin numbers, then using the configured options to decide whether or not to use them. That seems like a good idea for this case.

Blue-Marlin, 2016-10-22:
In the variants pins_arduino.h there is an array `digital_pin_to_timer_PGM[]`. If you do not get a 'NOT_ON_TIMER' the pin is hardware PWMable. The other problem is: Is this timer used for some other purpose?
Arduino Mega example:
```
const uint8_t PROGMEM digital_pin_to_timer_PGM[] = {
	// TIMERS		
	// -------------------------------------------		
	NOT_ON_TIMER	, // PE 0 ** 0 ** USART0_RX	
	NOT_ON_TIMER	, // PE 1 ** 1 ** USART0_TX	
	TIMER3B	, // PE 4 ** 2 ** PWM2	
	TIMER3C	, // PE 5 ** 3 ** PWM3	
	TIMER0B	, // PG 5 ** 4 ** PWM4	
	TIMER3A	, // PE 3 ** 5 ** PWM5	
	TIMER4A	, // PH 3 ** 6 ** PWM6	
	TIMER4B	, // PH 4 ** 7 ** PWM7	
	TIMER4C	, // PH 5 ** 8 ** PWM8	
	TIMER2B	, // PH 6 ** 9 ** PWM9	
	TIMER2A	, // PB 4 ** 10 ** PWM10	
	TIMER1A	, // PB 5 ** 11 ** PWM11	
	TIMER1B	, // PB 6 ** 12 ** PWM12	
	TIMER0A	, // PB 7 ** 13 ** PWM13	
	NOT_ON_TIMER	, // PJ 1 ** 14 ** USART3_TX	
	NOT_ON_TIMER	, // PJ 0 ** 15 ** USART3_RX	
	NOT_ON_TIMER	, // PH 1 ** 16 ** USART2_TX	
	NOT_ON_TIMER	, // PH 0 ** 17 ** USART2_RX	
	NOT_ON_TIMER	, // PD 3 ** 18 ** USART1_TX	
	NOT_ON_TIMER	, // PD 2 ** 19 ** USART1_RX	
	NOT_ON_TIMER	, // PD 1 ** 20 ** I2C_SDA	
	NOT_ON_TIMER	, // PD 0 ** 21 ** I2C_SCL	
	NOT_ON_TIMER	, // PA 0 ** 22 ** D22	
	NOT_ON_TIMER	, // PA 1 ** 23 ** D23	
	NOT_ON_TIMER	, // PA 2 ** 24 ** D24	
	NOT_ON_TIMER	, // PA 3 ** 25 ** D25	
	NOT_ON_TIMER	, // PA 4 ** 26 ** D26	
	NOT_ON_TIMER	, // PA 5 ** 27 ** D27	
	NOT_ON_TIMER	, // PA 6 ** 28 ** D28	
	NOT_ON_TIMER	, // PA 7 ** 29 ** D29	
	NOT_ON_TIMER	, // PC 7 ** 30 ** D30	
	NOT_ON_TIMER	, // PC 6 ** 31 ** D31	
	NOT_ON_TIMER	, // PC 5 ** 32 ** D32	
	NOT_ON_TIMER	, // PC 4 ** 33 ** D33	
	NOT_ON_TIMER	, // PC 3 ** 34 ** D34	
	NOT_ON_TIMER	, // PC 2 ** 35 ** D35	
	NOT_ON_TIMER	, // PC 1 ** 36 ** D36	
	NOT_ON_TIMER	, // PC 0 ** 37 ** D37	
	NOT_ON_TIMER	, // PD 7 ** 38 ** D38	
	NOT_ON_TIMER	, // PG 2 ** 39 ** D39	
	NOT_ON_TIMER	, // PG 1 ** 40 ** D40	
	NOT_ON_TIMER	, // PG 0 ** 41 ** D41	
	NOT_ON_TIMER	, // PL 7 ** 42 ** D42	
	NOT_ON_TIMER	, // PL 6 ** 43 ** D43	
	TIMER5C	, // PL 5 ** 44 ** D44	
	TIMER5B	, // PL 4 ** 45 ** D45	
	TIMER5A	, // PL 3 ** 46 ** D46	
	NOT_ON_TIMER	, // PL 2 ** 47 ** D47	
	NOT_ON_TIMER	, // PL 1 ** 48 ** D48	
	NOT_ON_TIMER	, // PL 0 ** 49 ** D49	
	NOT_ON_TIMER	, // PB 3 ** 50 ** SPI_MISO	
	NOT_ON_TIMER	, // PB 2 ** 51 ** SPI_MOSI	
	NOT_ON_TIMER	, // PB 1 ** 52 ** SPI_SCK	
	NOT_ON_TIMER	, // PB 0 ** 53 ** SPI_SS	
	NOT_ON_TIMER	, // PF 0 ** 54 ** A0	
	NOT_ON_TIMER	, // PF 1 ** 55 ** A1	
	NOT_ON_TIMER	, // PF 2 ** 56 ** A2	
	NOT_ON_TIMER	, // PF 3 ** 57 ** A3	
	NOT_ON_TIMER	, // PF 4 ** 58 ** A4	
	NOT_ON_TIMER	, // PF 5 ** 59 ** A5	
	NOT_ON_TIMER	, // PF 6 ** 60 ** A6	
	NOT_ON_TIMER	, // PF 7 ** 61 ** A7	
	NOT_ON_TIMER	, // PK 0 ** 62 ** A8	
	NOT_ON_TIMER	, // PK 1 ** 63 ** A9	
	NOT_ON_TIMER	, // PK 2 ** 64 ** A10	
	NOT_ON_TIMER	, // PK 3 ** 65 ** A11	
	NOT_ON_TIMER	, // PK 4 ** 66 ** A12	
	NOT_ON_TIMER	, // PK 5 ** 67 ** A13	
	NOT_ON_TIMER	, // PK 6 ** 68 ** A14	
	NOT_ON_TIMER	, // PK 7 ** 69 ** A15	
};
```

Blue-Marlin, 2016-10-22:
In fastio.h the check s the other way around. (section "// timers and PWM")
```
// __AVR_ATmega2560__
  // timers and PWM
  #define OC0A        DIO13
  #define OC0B        DIO4
  #define OC1A        DIO11
  #define OC1B        DIO12
  #define OC2A        DIO10
  #define OC2B        DIO9
  #define OC3A        DIO5
  #define OC3B        DIO2
  #define OC3C        DIO3
  #define OC4A        DIO6
  #define OC4B        DIO7
  #define OC4C        DIO8
  #define OC5A        DIO46
  #define OC5B        DIO45
  #define OC5C        DIO44
```
If the timer is defined you will get the pin.

Blue-Marlin, 2016-10-22:
```
int8_t digital_pins[] = {
  #ifdef OC0A 
    DIO13,
  #endif
  #ifdef OC0B 
    DIO4,
  #endif
...
  #ifdef OC5C 
    DIO44,
  #endif
  -1
};
```
As far as i know no 8bit AVR has more timers then the 2560. Maybe with other names - just look into fastio.h

Blue-Marlin, 2016-10-22:
But i guess a more effective software pwm will do a better job.

Make a list containing pins and duty cycles.
When setting a new duty cycle - sort the list by duty cycle. (this is rare)

  set all pin in the list high
  when time has come for the shortest duty cycle shut it off. Go to next list item.
 ...
 when cycle time is over set all pons high.

Seems to be much more effective than looking for every pin if it's time to switch off on every tick.

thinkyhead, 2016-10-23:
I found a pretty simple way to check if a pin is PWM-able.
```cpp
  #if HAS_AUTO_FAN_0
    static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0_AUTO_FAN_PIN is not a PWM pin.");
  #endif
```
…and #5702 is the set of changes required to modernize this feature.

thinkyhead, 2016-10-23:
> i guess a more effective software pwm will do a better job.

In the longer run this would be good. The current software PWM code is hard to follow, and tedious to extend.

dcwalmsley, 2016-10-23:
Hello All, 
I recently built a Cartesian 3D printer utilizing a Azteeg 3 Pro board and a Diamond Hotend extruder.  My current issue is with the extruder fan which is connected to D9.  

I originally pinned out on port 5 (pin 5) and when the autotemp feature in Configuration_adv.h activated the fan came on and the temperature stopped climbing to desired operator selected level.  After trouble shooting, I discovered that as stated earlier in the blog, the PWM is either 255 or 0 and there was no speed adjusting possible.

So, I pinned the fan to D9 and found I had success with autofan on/off and I was able to input from PWM 40 to 255.   But there is an issue I cannot figure out. The fan pushes air down and through the 3 extruder cooling vents but even at a PWM of 40 (which barely moves the fan) I still find the temperature won’t reach desired operator level.  

I adjusted the speed and I tried modifying the Configuration_adv.h section for the extruder to 185 C and PWM of 40 and once the temp was reached the fan came on at desired speed.  However, I commanded a higher temperature level and the built-in safety code triggers because the heating stops climbing.

I was successful in AutoPID tuning the extruder up to 240C with no fan connected but don’t advise this to anyone as the whole extruder unit was quite hot.  I did this to establish a Kp Ki and Kd configuration for Configuration.h extruder heating section.

On a side note, I have a thermal blanket separator placed between the extruder head, thermistor, and heater for the cooling vents and fan assembly.  This was recommended to ensure the fan wasn’t defeating the heater and thermistors job.

What am I doing wrong that I’m unable to get the fan to auto speed adjust during heat up of extruder or to get the extruder to reach desired temperatures?

thinkyhead, 2016-10-24:
It sounds peculiar. Almost as though the fan is stealing current from the heater. Have you tried a different fan?

Bob-the-Kuhn, 2016-10-24:
Sounds like there could be a heater issue.  Is the LED for the hot end behaving as expected?  It should be on brightly when heating up and then dim when it's just maintaining the temperature.  If it's on brightly all the time then you're not getting enough power to the heater.  It could be wiring, running a 24V heater at 12V or a bad heater.

Time to double check the wiring.  I suggest removing & re-installing the wires at the HE1 connector and the power input 2 connector.  Also remove & re-install fuse F4.

If you have a voltmeter I suggest measuring the voltage on the + terminal and the - terminal on the HE1 connector. The + terminal is from the power input 2 connector (through fuse F4) so it should be steady at your power supply level.   The - terminal should read a low voltage when heating up (0.6V or lower) and higher when maintaining temperature. 

I have the same controller but haven't hooked up the extruder yet.  If the above doesn't fix the issue then I'll try to reproduce the issue.  I'll need the following files: 
- conig
- config adv
- pins_AZTEEG_X3_PRO.h
- pins_RAMPS.h

dcwalmsley, 2016-10-25:
Hello Bob,

So I am running 24v power supply and my heater cores have been swapped out with other new 24v heater cores with same results.  The HE1 red LED is lit solid and bright throughout heating process.  If I disconnect the fan and perform a heat up to selected power the LED will extinguish upon reaching desired temp and come back on when heating is needed.  My next step is to give Repetier a shake to see if it's repeatable and if not, then I know it's Marlin firmware or something I have configured incorrectly. 

I will upload the requested files later today when I get home from work.

----------------------------------------------------------------------------------------
All,

I did move the extruder fan around to HE3 (PWM pin 16), HE7 (PWM Pin 7), and HE6 (PWM pin 6) and all showed the same result.  Once desired temperature was reached as set in Configuration_adv.h, the fan would only come on if PWM was 255.  Nothing else would work and fan remained off.

I currently have it pinned out on HE6 with the parts fan on HE7.

Here is my files for your review.

Thanks.

V/R,
Doug W.
[DCW-Azteeg X3 Pro - Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/551018/DCW-Azteeg.X3.Pro.-.Marlin.zip)

dcwalmsley, 2016-10-25:
Version is 1.1.0 RC7

dcwalmsley, 2016-10-25:
All,

I did move the extruder fan around to HE3 (PWM pin 16), HE7 (PWM Pin 7),
and HE6 (PWM pin 6) and all showed the same result.  Once desired
temperature was reached as set in Configuration_adv.h, the fan would only
come on if PWM was 255.  Nothing else would work and fan remained off.

I currently have it pinned out on HE6 with the parts fan on HE7.

Here is my files for your review.

Thanks.

V/R,
Doug W.

dcwalmsley, 2016-10-25:
Hello Bob,

So I am running 24v power supply and my heater cores have been swapped out
with other new 24v heater cores with same results.  The HE1 red LED is lit
solid and bright throughout heating process.  If I disconnect the fan and
perform a heat up to selected power the LED will extinguish upon reaching
desired temp and come back on when heating is needed.  My next step is to
give Repetier a shake to see if it's repeatable and if not, then I know
it's Marlin firmware or something I have configured incorrectly.

I will upload the requested files later today when I get home from work.

brainscan, 2016-10-25:
Have you tried auto tune with the fan running? If it's going to be running while printing then it needs to be on while you tune it. That's my experience anyway.

Bob-the-Kuhn, 2016-10-25:
Please see if you still have the problem using the RCBugFix branch.  There's been a lot of bug fixes & improvements since RC7.

dcwalmsley, 2016-10-25:
So,  I now am on hold.  Snapped heater block at the break and now awaiting more parts.  :-(

I will post my status early next week when parts arrive. 

Ugh

gcormier, 2016-10-26:
I'm going to close this issue as I think the changes are complete on the original topic...

closed 2016-10-26 by gcormier

Bob-the-Kuhn, 2016-10-26:
Email me at bob.kuhn@att.net when you get your parts in.  We can take this private for now.

thinkyhead, 2016-10-26:
@dcwalmsley Another thing you can try is to run the fan from a separate power supply, then you can still test heatup with the fan, but making sure that the fan isn't leeching current from the electronics.