MarlinFirmware/Marlin #5012: New Board GT2560 printer build

23chaos23, 2016-10-12:
ok so i have a few things i need to trouble shoot, but im not a wizard with arduino so my printer build is at a standstill.
Im building a custom 3d printer using OpenParts hardware and GEEEtech extruders and their GT2560 board http://www.geeetech.com/wiki/index.php/GT2560 

i need the board to talk to a LCD12864 http://www.geeetech.com/reprap-smart-controller-lcd12864-version-led-turn-on-control-p-690.html
dual extruder dual nozzle 32mm offset
min/max endstops
zmin on pwm servo arm for auto bed leveling

i can read and understand enough of the config.h tab to set up all the extruder x offsets and z probe positions and servo degrees, but the base code wont compile for this board

please help

Bob-the-Kuhn, 2016-10-12:
Please supply the following:
- The error message(s) from the failed compile (just past it into a text file & attach)
- your config files
- the pins_xxx.h file you're using
- which branch of Marlin that you're using

Bob-the-Kuhn, 2016-10-12:
You need to supply the **pins_ULTIMAKER.h** file.  That's the one used when motherboard is set to 7.

23chaos23, 2016-10-12:
[000.txt](https://github.com/MarlinFirmware/Marlin/files/525748/000.txt)

Bob-the-Kuhn, 2016-10-12:
The Marlin file(s) I downloaded from [](http://www.geeetech.com/wiki/index.php/GT2560) compiled without error.  I'm using the same Arduino IDE as you are.  I didn't make any changes to any of the files.

From the error messages it looks like the Configuration.h file has been modified .  The **#define MOTHERBOARD**  line in it has been changed.  It needs to say exactly **#define MOTHERBOARD 7** .  All the other errors are probably due to the wrong motherboard being defined.

You'll also need to edit the Configuration.h & Configuration_adv.h files to match your hardware.

The Configuration.h & Configuration_adv.h files are several layers down in the Marlin-Marlin_v1 directory that you created when you unzipped the file you downloaded.

Bob-the-Kuhn, 2016-10-12:
Sorry - I just re-read the your last comment.

I need to see the ones that you pointed the Arduino IDE at.  It looks like they're all in **C:\Users\eaalb_000\Desktop\Marlin 3D printer\Marlin-RC\Marlin\**

The directory names tell me that you're not using the geeetech Marlin software.  

Please include the **version.h** file also so I know what Marlin branch you're using. 

The error message says the motherboard is a RAMPs board.  That's one specified in an unmodified configuration.h file.  You'll need to change it to **BOARD_ULTIMAKER**

23chaos23, 2016-10-12:
[Marlin 3D printer.zip](https://github.com/MarlinFirmware/Marlin/files/525838/Marlin.3D.printer.zip)
ok so the marlin RC is the one from here the other 3 folders are versions for this board/the sungulolo board, these are from the suppliers link http://www.geeetech.com/forum/viewtopic.php?f=20&t=17046

all 3 of those did not compile. also i changed the define board 7 change and it compiles now, but will it work with the GT2560?

Bob-the-Kuhn, 2016-10-13:
You'll need to modify the new (RC) config files to work with the GT2560 printer.

It looks like the _v1 config files were already setup for a specific printer.  I'm assuming the it was the GT2560 since they came from the GT2560 support area.

I took the liberty of trying to move all the _v1 settings to the new config files (attached).  You'll need to look them over because it's easy to make mistakes.

The new config files have a lot more options in them, some things have been moved and some names have been changed.  Take your time and you'll figure most of it out.  One thing I noticed is the _v1 files want a custom M581 code.  That's now one of the standard ones in your RC branch.

There's also more cross checking done.  The new system didn't like temp_sensor_1 enabled when extruder_1 wasn't.

I did successfully compile the RC directory with the modified config files.  You could probably upload as is and be able to move the steppers.

You'll need to check how your system works and modify the config files accordingly.  One thing I did notice was that the steps per unit were different in the _v1 files vs. the setup instructions on the GT2560 web page.

Almost forgot - you're running branch RC7 (technically 1.1.0-RC7).

[Config.mod.zip](https://github.com/MarlinFirmware/Marlin/files/526016/Config.mod.zip)

23chaos23, 2016-10-13:
cool thankyou so much for the help, how do i add these new files? do i just take off the .mod at the end and replace the corresponding .h files

Bob-the-Kuhn, 2016-10-13:
Yes

23chaos23, 2016-10-13:
cool i replaced the files and compiled, A OK! ill upload to the board and see what happens

23chaos23, 2016-10-13:
no dice, tried to upload and it gives me this

Sketch uses 54,812 bytes (21%) of program storage space. Maximum is 253,952 bytes.
Global variables use 3,075 bytes (37%) of dynamic memory, leaving 5,117 bytes for local variables. Maximum is 8,192 bytes.
avrdude: verification error, first mismatch at byte 0x001e
         0x4e != 0x46
avrdude: verification error; content mismatch
avrdude: verification error; content mismatch

Bob-the-Kuhn, 2016-10-13:
OUCH!!!

Is this repeatable?

That's an error where the data that was uploaded, written to FLASH but didn't verify.  

I'm hoping that you had noise on the com line, not a stuck bit in the FLASH memory.

23chaos23, 2016-10-13:
yes its repeatable

Bob-the-Kuhn, 2016-10-13:
Is the Arduino IDE set to mega2560 or mega ADK?

The original error message you sent said "Arduino Mega ADK".  Try setting it to "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

23chaos23, 2016-10-13:
ok switched to Arduino/Genuino Mega or Mega 2560 and it flashed fine
after power down a restart the lcd is blank, i geet a ticking sound from the buzzer on the lcd.

i can connect with repetier host but i get a MAXTEP trig sys stop heater ID: 0 and it hangs all commands to the board

Bob-the-Kuhn, 2016-10-13:
The LCD display isn't enabled in the config file.

Do you have your thermistor/thermocouple hooked up?  Sounds like TEMP_SENSOR_0 has problems.

A temp work around is to change the following line in configuration.h from -1 to 998
   #define TEMP_SENSOR_0 -1   to #define TEMP_SENSOR_0 998

23chaos23, 2016-10-13:
ok ill set the tempsens to debug for now, how do i enable and set up the lcd?

23chaos23, 2016-10-13:
do you have team speak? i think we could work this out faster that way. you can find me on my team speak Nix gravis gaming, IP: 66.85.77.146

Bob-the-Kuhn, 2016-10-13:
The geeech wiki says:

1. INSTALL U8glib to your ARDUINO library folder: 
2. In "Configuration.h” Change “#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER” (just remove the // at the beginning) 

To install U8glib go to the Arduino IDE, click **SKETCH**, then **include library**, then **manage libraries...**

That'll popup a window.  In the **filter your search** area type **U8glib**.  That'll show a couple of libraries.  The U8glib has a pull down to select the version.  Click on either one and it'll do the rest.

Bob-the-Kuhn, 2016-10-13:
It's 12:19 here.  Time for me to call it a night.

23chaos23, 2016-10-13:
No problem Bob, thanks for the help so far, if your free again tomorrow ill be online all day again.

Bob-the-Kuhn, 2016-10-13:
Let's take this offline.  My email is:
  bob.kuhn@att.net

My cell is 512 934-1943

23chaos23, 2016-10-13:
email sent

Bob-the-Kuhn, 2016-10-13:
didn't get it

23chaos23, 2016-10-13:
sent it to 	bob.kuhn@att.net,

23chaos23, 2016-10-13:
you can send me one at e.a.albertyn@gmail.com

closed 2016-10-14 by 23chaos23