MarlinFirmware/Marlin #5014: Disable "Set home offsets"

spilz87, 2016-10-13:
hello,
it's not really an issue, more a request, but i don't know where I can put it.
if it's really not the good location for this tell me and show me where I should suggest it.

I have often this problem and I found people who have the same : 
- I never need to use "Set home offsets"
- but I select it sometimes by error

is there a way to disable it in LCD menu ?

Sorry for my bad English

regards

FHeilmann, 2016-10-13:
For a quick fix:

https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd.cpp#L1198

change

```C   
//
// Set Home Offsets
//
MENU_ITEM(function, MSG_SET_HOME_OFFSETS, lcd_set_home_offsets);
//MENU_ITEM(gcode, MSG_SET_ORIGIN, PSTR("G92 X0 Y0 Z0"));
```

to

```C   
//
// Set Home Offsets
//
//MENU_ITEM(function, MSG_SET_HOME_OFFSETS, lcd_set_home_offsets);
//MENU_ITEM(gcode, MSG_SET_ORIGIN, PSTR("G92 X0 Y0 Z0"));
```

spilz87, 2016-10-17:
that works 
thanks a lot

closed 2016-10-22 by thinkyhead