MarlinFirmware/Marlin #5015: Building on arm (arduino 1.6.0 + core 1.6.0) in cli fails

ngarafol, 2016-10-13:
I am using arm (orange pi) and using arduino-core 1.6.0 building sketch fails, some sanity check complains about not using arduino 1.6.0...
I will post logs once I get to the orange pi, but was wondering does someone build Marlin on arm (running octoprint on it, is directly connected to arduino mega2560). Older version, MArlin 1.0.0 I managed to build using arduino-core 1.0.5 for reference.

kevans91, 2016-10-13:
Hi,

I believe it actually requires 1.6.8, but are you invoking arduino or
arduino-builder? With the latter, you have to set the core version higher
manually.

Thanks,

Kyle Evans

>
> I am using arm (orange pi) and using arduino-core 1.6.0 building sketch
fails, some sanity check complains about not using arduino 1.6.0...
> I will post logs once I get to the orange pi, but was wondering does
someone build Marlin on arm (running octoprint on it, is directly connected
to arduino mega2560). Older version, MArlin 1.0.0 I managed to build using
arduino-core 1.0.5 for reference.

ngarafol, 2016-10-13:
I am invoking make (from arduino-mk package IIRC), see here: http://hardwarefun.com/tutorials/compiling-arduino-sketches-using-makefile

And thats how I updated to arduino 1.6.0: http://duinorasp.hansotten.com/arduino-1-6-x-for-raspberry-pi-jessie/ 

I use debian 8 aka jessie.

kevans91, 2016-10-13:
Alright, looks like you need to figure out what ARDUINO_VERSION is getting set to, then, because it defaults to 105 if not otherwise specified -- updating Arduino itself generally won't count as 'otherwise specified.'

If it's defaulted to 105, then you can set it to 10600 manually given that you've done your due diligence with updating arduino-core.

ngarafol, 2016-10-13:
Seems like arduino_version is ok.. 

[http://pastebin.com/9QQEz0VE](pastebin)
[http://pastebin.com/knrCNW1z](makefile used)
[http://pastebin.com/22LbCbCC](Marlin modified makefile)

If I used modifed makefile shipped with marlin, I get following error:

`make: *** No rule to make target '/root/bager/wiring.o', needed by '/root/bager/Marlin.elf'.  Stop.`

kevans91, 2016-10-14:
Ahh, arduino_version is in fact off -- see:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/SanityCheck.h#L33

Core version for 1.6.0 = 10600

thinkyhead, 2016-10-22:
Probably best to work with `RCBugFix` because it has an updated Makefile.

bgort, 2016-10-22:
An option:  you may want to consider platformio for command-line compiling.  I've been using it for 4-5 months on Rpi with Marlin and other Arduino sketches without any trouble.  RCBugFix has a working platformio.ini and compiles without a hiccup with a single command (`platformio run`) from the Marlin directory.

I've not used Orange Pi, but imagine most everything would be the same if you're running Jessie.

ngarafol, 2016-10-26:
Thanks bgort for the info, it works using platformio!

closed 2016-10-26 by ngarafol