MarlinFirmware/Marlin #5019: G29 Bed leveling mirrored

chrs37, 2016-10-13:
I'm not 100% sure if it's my mistake (bad wiring or config) or a actual bug, but when I try to auto level my printer the tilt compensation is mirrored in both axis. It worsens the bed tilt.

My printer uses delta kinematics and I have tried switching around the axis but this doesn't seem to make  any difference.  

Does anyone here know what to do?

thinkyhead, 2016-10-14:
Hi @chrs37 – Are you using Marlin `RCBugFix` or some other version?

Delta uses bilinear (mesh) bed leveling, so if there's an issue with the wrong Z height, that's the code we would need to look at. It raises and lowers Z according to the height of the bed at the probed points, rather than just applying a single tilted plane, and was created to help deltas with the problem of bowl-shaped movements (as when the geometry isn't perfectly calibrated).

Do you find that there's Z goes off as you go closer to the edges of the bed. Is it worse at some points than at others?

You can enable `DEBUG_LEVELING_FEATURE` to get diagnostic information in more detail. Before starting, do `M111 S32` or `M111 S247` to enable debugging of homing and leveling. Then do `G28` followed by `G29 V4` and post the log here. Hopefully from that we can figure out where the problem lies.

chrs37, 2016-10-14:
Thanks for your answer!

I'm using RCBugFix. And the further I move away from the center, the bigger the error seems to be. As far as I can tell the offset seems very linear and doesn't appear to be random.

Here's the log:
23:04:05.065 : echo:N34 M111 S247*87
23:04:05.068 : echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
23:04:13.717 : echo:N38 G28*40
23:04:13.717 : >>> gcode_G28
23:04:13.719 : Machine Type: Delta
23:04:13.720 : Probe: FIX_MOUNTED_PROBE
23:04:13.722 : Probe Offset X:30 Y:18 Z:-8.00 (Right-Back & Below Nozzle)
23:04:13.724 : Auto Bed Leveling: BILINEAR
23:04:13.724 : reset_bed_level
23:04:13.726 : current_position=(0.00, 0.00, 183.46) : setup_for_endstop_or_probe_move
23:04:13.728 : > endstops.enable(true)
23:04:13.730 : current_position=(0.00, 0.00, 183.46) : >>> home_delta
23:04:13.733 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
23:04:15.533 : >>> homeaxis(X)
23:04:15.533 : Home 1 Fast:
23:04:15.536 : >>> do_homing_move(X, 360.00, 0.00)
23:04:15.537 : current_position=(0.00, 275.22, 275.22) : sync_plan_position
23:04:15.541 : <<< do_homing_move(X)
23:04:15.542 : Move Away:
23:04:15.544 : >>> do_homing_move(X, -2.00, 0.00)
23:04:15.547 : current_position=(0.00, 275.22, 275.22) : sync_plan_position
23:04:15.591 : <<< do_homing_move(X)
23:04:15.592 : Home 2 Slow:
23:04:15.594 : >>> do_homing_move(X, 4.00, 40.00)
23:04:15.596 : current_position=(0.00, 275.22, 275.22) : sync_plan_position
23:04:15.652 : <<< do_homing_move(X)
23:04:15.653 : <<< homeaxis(X)
23:04:15.653 : >>> homeaxis(Y)
23:04:15.655 : Home 1 Fast:
23:04:15.656 : >>> do_homing_move(Y, 360.00, 0.00)
23:04:15.659 : current_position=(4.00, 0.00, 275.22) : sync_plan_position
23:04:15.660 : <<< do_homing_move(Y)
23:04:15.662 : Move Away:
23:04:15.663 : >>> do_homing_move(Y, -2.00, 0.00)
23:04:15.666 : current_position=(4.00, 0.00, 275.22) : sync_plan_position
23:04:15.710 : <<< do_homing_move(Y)
23:04:15.711 : Home 2 Slow:
23:04:15.712 : >>> do_homing_move(Y, 4.00, 40.00)
23:04:15.715 : current_position=(4.00, 0.00, 275.22) : sync_plan_position
23:04:15.770 : <<< do_homing_move(Y)
23:04:15.771 : <<< homeaxis(Y)
23:04:15.772 : >>> homeaxis(Z)
23:04:15.773 : Home 1 Fast:
23:04:15.774 : >>> do_homing_move(Z, 397.83, 0.00)
23:04:15.777 : current_position=(4.00, 4.00, 0.00) : sync_plan_position
23:04:15.778 : <<< do_homing_move(Z)
23:04:15.780 : Move Away:
23:04:15.781 : >>> do_homing_move(Z, -2.00, 0.00)
23:04:15.783 : current_position=(4.00, 4.00, 0.00) : sync_plan_position
23:04:15.828 : <<< do_homing_move(Z)
23:04:15.829 : Home 2 Slow:
23:04:15.831 : >>> do_homing_move(Z, 4.00, 20.00)
23:04:15.833 : current_position=(4.00, 4.00, 0.00) : sync_plan_position
23:04:15.934 : <<< do_homing_move(Z)
23:04:15.936 : <<< homeaxis(Z)
23:04:15.937 : >>> set_axis_is_at_home(X)
23:04:15.937 : For X axis:
23:04:15.938 : home_offset = 0.00
23:04:15.938 : position_shift = 0.00
23:04:15.940 : soft_endstop_min = -120.00
23:04:15.941 : soft_endstop_max = 120.00
23:04:15.942 : > home_offset[X] = 0.00
23:04:15.943 : current_position=(0.00, 4.00, 4.00) :
23:04:15.945 : <<< set_axis_is_at_home(X)
23:04:15.946 : >>> set_axis_is_at_home(Y)
[... 825 lines cut ...]
23:04:58.082 : (done)
23:04:58.082 : Extrapolate [ 0+  3-]
23:04:58.083 : (done)
23:04:58.083 : Extrapolate [ 4-  3-]
23:04:58.083 : (done)
23:04:58.084 : Extrapolate [ 0+  0+]
23:04:58.084 : (done)
23:04:58.086 : Extrapolate [ 4-  0+]
23:04:58.086 : (done)
23:04:58.087 : Extrapolate [ 0+  4-]
23:04:58.087 : (done)
23:04:58.088 : Extrapolate [ 4-  4-]
23:04:58.088 : (done)
23:04:58.089 : Bilinear Leveling Grid:
23:04:58.091 : 0     1     2     3     4
23:04:58.094 : -1.25  -0.64  -0.25  +0.06 +0.39
23:04:58.095 :  -0.81 -0.39 -0.03 +0.32 +0.67
23:04:58.095 :  -0.49 -0.05 +0.31 +0.67 +0.99
23:04:58.096 :  -0.15 +0.27 +0.64 +0.97 +1.31
23:04:58.098 :  +0.00 +0.69 +1.07 +1.38 +1.63
23:04:58.099 : G29 uncorrected Z:28.00
23:04:58.099 : corrected Z:26.79
23:04:58.100 : <<< gcode_G29
23:04:58.103 : X:88.00 Y:80.00 Z:26.79 E:0.00 Count X: 10661 Y:22434 Z:24703
23:04:58.107 : current_position=(88.00, 80.00, 26.79) : sync_plan_position_kinematic

Roxy-3D, 2016-10-14:
That bed is very tilted.  From the back left to the front right you have almost 3mm shift in height.  
We do need to figure out the firmware error that is causing you problems.  But you really should try to get your bed physically more level.  The Auto Bed Leveling can do a better job if it doesn't have to deal with very extreme situations.

thinkyhead, 2016-10-22:
I've been patching the bed leveling code a bit more, recently. Please give the latest `RCBugFix` a try and post another log for comparison.

boelle, 2016-11-05:
@chrs37 did you retest with the latest RCBugfix?

chrs37, 2016-11-05:
I haven't gotten around to testing it yet. I needed to spend a bit more time on school instead of my 3D printer.... But after I enabled the debug feature and leveled my bed a bit more, the auto leveling seemed to work fine. When I have some spare time I'll test the latest RCBugfix.

nzinov, 2016-11-06:
I was experiencing some odd behavior with bilinear leveling today at 1614c9d. I hasn't managed to troubleshoot it but after seeing this issue I suspect the same problem (mirroring). Will be able to test it closely in a day.

thinkyhead, 2016-11-07:
@nzinov Delta?

nzinov, 2016-11-07:
@thinkyhead Yes

nzinov, 2016-11-08:
I tested carefully with the bed tilted at artificially big angle. Moving on both X and Y goes fine with no mirroring (at 1614c9d). So no problem on my delta printer.

boelle, 2016-11-12:
@thinkyhead @nzinov @chrs37 

So this one seems to be solved? Can we close it?

closed 2016-11-17 by thinkyhead