MarlinFirmware/Marlin #5020: Auto Bed Leveling, Z not compensating for unlevel bed

Skylark7, 2016-10-14:
Hey,

I've just recently finished building my Smartrapcore (Wooden) and pretty much everything works except the auto bed leveling. I'm using Marlin 1.0.2-1.

Technically it woks (I've done prints), I mean after Z homes in the center, the probe goes to all 4 corners and probes and I see the probe's red light turning on and off as this happens. The problem is, the Z axis doesn't move at all to compensate and level the bed during the print.
One side of my first layer is completely squished down, very thin layer. While the other side is round and barely sticking to the bed! I used [Tom's guide](https://www.youtube.com/watch?v=EcGFLwj0pnA
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/530172/Configuration.zip)

) and tried to follow his advices pretty closesly.
My print bed is pretty solid and doesnt vibrate or wobble up and down and I'm pretty sure the glass bed I'm using is flat. Because I used it on my first printer until I got an aluminum bed for it, which was working well.
I use an inductive sensor with aluminum tape on the glass bed.

Here is my config file:
http://pastebin.com/u7L2N6GY

I've been fiddling with config file and my setup for a few days and I couldn't figure out the problem, so I'd appreciate some some help! :)

Grogyan, 2016-10-14:
Please try using Marlin RCBugFix

WOOKIE111, 2016-10-15:
possibly related to #4948

thinkyhead, 2016-10-22:
@WOOKIE111 No bilinear leveling for non-delta in older Marlin.

closed 2016-10-22 by thinkyhead