MarlinFirmware/Marlin #5021: Garbled character of RCBugFix with REPRAP_DISCOUNT_SMART_CONTROLLER and without SDSUPPORT

esenapaj, 2016-10-14:
In case of newest RCBugFix, when `REPRAP_DISCOUNT_SMART_CONTROLLER` is enabled and `SDSUPPORT` is disabled,
'⤴' is garbled to '／'.
RC7 looks like no problem.

RCBugFix without `SDSUPPORT` (garbled) | RCBugFix with `SDSUPPORT` (ok) | RC7 without `SDSUPPORT` (ok)
--- | --- | ---
[image] | [image] | [image]

Blue-Marlin, 2016-10-14:
It's commit https://github.com/MarlinFirmware/Marlin/commit/e47c8b98293630e958bd2fd07c6949ce946db2b2
Custom char `uplevel` has to be defined always - not only `#if ENABLED(SDSUPPORT)`
Shift `uplevel` up to just below `clock`. Don't forget to also shift up https://github.com/MarlinFirmware/Marlin/blame/RCBugFix/Marlin/ultralcd_impl_HD44780.h#L322

esenapaj, 2016-10-15:
Thanks for the analyzing and how to fix.
I've submitted a patch just now.

closed 2016-10-24 by thinkyhead