MarlinFirmware/Marlin #5023: what speed should Temperature::isr() be running at?   1KHz or 62.5KHz  (RCBugFix)

Bob-the-Kuhn, 2016-10-15:
Either the comments are wrong or Temperature::isr() is running at the wrong rate or I'm up too late (again).

The comments say it's running at 62.5KHz and the slow fan PWM speed comment  **// 488.28 Hz (or 1:976.56, 2:1953.12, 3:3906.25, 4:7812.5, 5:7812.5 6:15625, 6:15625 7:31250)** calculates out to be 62.5KHz.

Print statements (correlated by the host time stamp) indicate it's actually running at 1KHz.

I put this code right at the start of the ISR
``    bob_count += 1;
  if (bob_count == 625) {
    SERIAL_PROTOCOLLNPAIR("SHOULD BE EVERY 10:", millis());
    bob_count = 0;
  }``

and got this result.  The millis() count is incrementing by 639 and the time stamp is incrementing by 641 mS.

```
00:18:59.682 : SHOULD BE EVERY 10:12222
00:19:00.322 : SHOULD BE EVERY 10:12861
00:19:00.964 : SHOULD BE EVERY 10:13500
00:19:01.604 : SHOULD BE EVERY 10:14139
00:19:02.244 : SHOULD BE EVERY 10:14778
00:19:02.886 : SHOULD BE EVERY 10:15417
00:19:03.526 : SHOULD BE EVERY 10:16056
00:19:04.166 : SHOULD BE EVERY 10:16695
00:19:04.806 : SHOULD BE EVERY 10:17334
00:19:05.446 : SHOULD BE EVERY 10:17973
00:19:06.088 : SHOULD BE EVERY 10:18612
00:19:06.728 : SHOULD BE EVERY 10:19251
00:19:07.368 : SHOULD BE EVERY 10:19890
00:19:08.008 : SHOULD BE EVERY 10:20529
00:19:08.649 : SHOULD BE EVERY 10:21168
00:19:09.289 : SHOULD BE EVERY 10:21807
00:19:09.929 : SHOULD BE EVERY 10:22446
```

Sebastianv650, 2016-10-15:
Maybe my comment in #4971 answers your question, i was also confused by the comment.

Blue-Marlin, 2016-10-15:
Out processor is running with 16MHz..
The prescaler for timer 0 is set to 256. Timer 0 will that for increment with 62500Hz (0.000016 s).
The compare register of timer 1 channel b is set to 128. The counter counts from 0 to 128, resets to 0 and throws the interrupt. We get interrupts at a rate of 976,5625Hz - 1/0.001024 s. About 1ms.
In this interrupt service routine we establish our software pwm with a base of 127 ticks per cycle. 7.625Hz  - 1/0.1311 s.
Slow soft pwm uses the interrupt as its timebase and adds an kind of next prescaler to get its real ticks (slow_pwn_count)
The state machine for the temperatire masurements is called every tick (976,5625Hz). It has 12 states, so after 12 cyles we have values for all ADCs (81Hz). And need OVERSAMPLE (16) samples to complete a measurement (~5Hz). (The comments in the code are based on less (10) states.) New values for the PWMs are also calculated at 5Hz (when we have a newly measured temperature.)

Bob-the-Kuhn, 2016-10-15:
The two comments are outdated.  Much better than the alternative.

  thanks
   Bob

closed 2016-10-15 by Bob-the-Kuhn

thinkyhead, 2016-10-22:
Changed the comment on `Stepper::isr` to read:
```cpp
/**
 * Stepper Driver Interrupt
 *
 * Directly pulses the stepper motors at high frequency.
 * Timer 1 runs at a base frequency of 2MHz, with this ISR using OCR1A compare mode.
 *
 * OCR1A   Frequency
 *     1     2 MHz
 *    50    40 KHz
 *   100    20 KHz - capped max rate
 *   200    10 KHz - nominal max rate
 *  2000     1 KHz - sleep rate
 *  4000   500  Hz - init rate
 */
```