MarlinFirmware/Marlin #5024: warning about u8g with ca7275a6

ruggb, 2016-10-15:
I just installed the latest bug fix ca7275a6 and Arduino 1.6.12. Previous builds (62fd176 & ccf8001) do not give any warnings/errors, but I get this warning on compile now;

d:\Users\Bill\Documents\Arduino\libraries\U8glib\src\clib\u8g_rot.c:48:1: warning: initialization from incompatible pointer type

 u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };

 ^

d:\Users\Bill\Documents\Arduino\libraries\U8glib\src\clib\u8g_rot.c:48:1: warning: (near initialization for 'u8g_dev_rot.dev_fn')

what's up with that?

thinkyhead, 2016-10-22:
Try a newer U8glib and see if it helps.

ruggb, 2016-10-22:
it isn't that. I had a problem with the Arduino driver not loading. Got that to work and now that error is gone. Another ghost.
thx

closed 2016-10-22 by ruggb