MarlinFirmware/Marlin #5026: Requested approval for a third-party application

Roxy-3D, 2016-10-15:
I received this message from GitHub:

> @emartinez167 has requested approval for a third-party application to access MarlinFirmware organization resources via the GitHub API:
> 
>   "CodeHub" from Dillon Buchanan
> 
> Until it is approved, this application will have no access to private resources and will have read-only access to public resources belonging to your organization.

@emartinez167 Can you explain to us what is going on?

@thinkyhead @jbrazio   I presume you both got a request also?  From a security perspective, I feel we need to move slowly and carefully on these types of requests.

emartinez167, 2016-10-15:
Hi all,

I was trying to get the iPad app re-authorized, and I must have inadvertently clicked on the button to request access. Please ignore it.

Regards,

Ernesto

Roxy-3D, 2016-10-15:
OK!  Thank You for the explanation!

closed 2016-10-15 by Roxy-3D