MarlinFirmware/Marlin #5027: Is it possible to read data from the arduino back to the PC?

GrevanGER, 2016-10-15:
Greetings. 

I know, this is not a directly Marlin related issue, yet I wasn't sure if someone back at the arduino forums could help me, since this might be a bit to specific. Or not... I am actually a bit to confused where to start looking for help since google research didn't give me anything that is even in the same ballpark as my problem. 

Long story short: 
I haven't touched my printer in about a year do to a new job and other stuff that keept me busy. In the meantime I had to reset my computer about three times, meaning that I have lost all previous files. 

Since I have got a Frankenstein printer (custom silicon heatbead, custom thermisters, nozzle ... etc.) it would be a real pain to call every single company I have bought stuff from and gather all information again and adjust all the scripts to my needs. 

So my question is: Is it somehow possible to recover the data I have loaded up to my Arduino Mega and kind of "download" the files from it?

Roxy-3D, 2016-10-15:
Depending upon how old your Marlin is in the printer...   You might be able to get most of the information with a M503 command.   The information that M503 provides has been expanded greatly in the last 18 months.  So depending upon what you have running, you might be able to benefit from that.

GrevanGER, 2016-10-15:
Thanks for your help, but it looks like I have had a version running, that is not giving out information, that Pronterface wasn't able to gather after connection with my printer. 
At least I was able to safe me from having to calculate the steps per unit. 
But I am afraid  I have to dig for all the heater and thermister specs again, which is the real bother.

Someone else suggested to me, that I should use AVRDude to recover my data, but that kind of stuff seems to requiere an IT-degree to make it even run on a windows computer.

Roxy-3D, 2016-10-15:
> Someone else suggested to me, that I should use AVRDude to recover my data, but that kind of stuff seems to requiere an IT-degree to make it even run on a windows computer.

You could grab the firmware off of the printer and disassemble things to get your numbers.  But I can assure you, that won't be less effort.

GrevanGER, 2016-10-15:
Well, at least this teraches me to always hide a flashdrive somewhere with all the files fro mthe last installation. ^^

Roxy-3D, 2016-10-15:
Or fork the Marlin repository and change the Configuration.h file on GitHub (in your own account).   It will just sit there waiting for you to come back and use it again.

thinkyhead, 2016-10-22:
To capture data off the Arduino using `avrdude` (in Windows) use the following syntax:
#### Flash (program) data:
```
avrdude -c arduino -P com9 -p ATMEGA328P -b 19200 -U flash:r:%temp%\backup_flash.bin:r
```
#### EEPROM data:
```
avrdude -c arduino -P com9 -p ATMEGA328P -b 19200 -U eeprom:r:%temp%\backup_eeprom.bin:r
```
Since they are .bin files you should be able to open them in a text editor and see any text strings embedded in the code.

Instead of %temp% you can use the full path, such as `\Users\owner\AppData\Local\Temp\`

GrevanGER, 2016-10-22:
@thinkyhead: 
Well, I have the problem, that AVRDude even refuses to boot under win10 64bit. Or at least it does so on my system. I am just getting the good old direct input screen, it collapses and that's about it. 
To be honest, I have got no concept on how that programm is supposed to be opperated and the internet is not realy helpful when it comes to that.

thinkyhead, 2016-10-23:
Did you try these? https://drive.google.com/drive/folders/0B0LHhwkPhcG8M2RtSURJMGptdjQ

closed 2017-10-31 by thinkyhead