MarlinFirmware/Marlin #5030: Linking error with recent Arduino AVR Boards tools update

Fulg, 2016-10-15:
Hello,

Using current RCBugfix, ca7275a67a429d557d369041c7643128d4a032c4, trying to compile with EXTRUDERS 2 and BLTouch enabled at the same time will yield a linking error:

`unable to find a register to spill in class 'POINTER_REGS'`

Minimum amount of changes required to reproduce this (attached):
`#define MOTHERBOARD BOARD_RAMPS_14_EEB`
`#define EXTRUDERS 2`
`#define TEMP_SENSOR_0 998`
`#define TEMP_SENSOR_1 998`
`#define TEMP_SENSOR_BED 998`
`#define BLTOUCH`

The workaround is to manually downgrade the `Arduino AVR Boards` package of the Arduino IDE to version 1.6.11 (from the `Tools | Board: | Board Manager...` menu). This error occurs from 1.6.12 and up on Windows (I didn't try other platforms).

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/531277/Configuration.zip)

Fulg, 2016-10-15:
Duplicate of #5004? I guess I fail at searching. Nevertheless a workaround wasn't identified over there so maybe my report is still useful.

thinkyhead, 2016-10-22:
These are more-or-less spurious compiler errors. There is no definitive solution. Shuffling the code around sometimes helps. Ultimately, all we can do to avoid these altogether is break the code up into smaller units. That will be done in earnest for 1.2.x.

closed 2016-10-22 by thinkyhead