MarlinFirmware/Marlin #5031: Babystepping - do not idle out

gcormier, 2016-10-15:
I found it annoying getting idled out of babystepping. Staring intently at the bed nozzle... trying to tweak, wait for some more skirt to print... see how it's going... get booted out.

I have a patch I can put a PR in for that disables the idle timeout during babysteps.

(BTW, not sure how to contribute with things like this. Should I just submit the PR and we talk about it then? Or ask here first?)

Roxy-3D, 2016-10-16:
The default is 15 seconds.  You can change it in Configuration_adv.h:
```cpp
 #define LCD_TIMEOUT_TO_STATUS 15000
```
And if you set it long and need to get back to the status screen, the Baby Stepping respects an Encoder Wheel click and will bailout early if you tell it to.

gcormier, 2016-10-16:
I don't mind the timeouts on other screens though.. just babystepping :) But I can just set it to 30 or 60 and then that's fine. I rarely use my LCD anyways.

Roxy-3D, 2016-10-16:
Also...   I don't have the 'Press and Hold' of the encoder wheel working yet.  But the 'Kill Switch' over ride to get into Z-BabyStepping is working.   It doesn't matter if the screen times out because with one easy button push, you are back into it.   You might want to give that a try.

gcormier, 2016-10-16:
Which branch is that in?

KevMags, 2016-10-16:
I am finding "#define LCD_TIMEOUT_TO_STATUS 15000" in ultralcd.h.

Roxy-3D, 2016-10-16:
It is in the https://github.com/MarlinFirmware/Marlin/tree/devel-ubl branch.   This branch was forked from RCBugFix on July 4th 2016.   It works, but it is a 'Development Branch'.  With that said, once you learn the process to define a High Resolution Mesh, you can get a new piece of glass fully tuned in in under 30 minutes.  (And much of that time, you are not doing any work.)

closed 2016-10-16 by gcormier

thinkyhead, 2016-10-22:
Here's the solution. Add `defer_return_to_status = true;` to the babystep menu items:
```cpp
    #if ENABLED(BABYSTEP_XY)
      static void _lcd_babystep_x() { _lcd_babystep(X_AXIS, PSTR(MSG_BABYSTEPPING_X)); }
      static void _lcd_babystep_y() { _lcd_babystep(Y_AXIS, PSTR(MSG_BABYSTEPPING_Y)); }
      static void lcd_babystep_x() { babysteps_done = 0; lcd_goto_screen(_lcd_babystep_x);  defer_return_to_status = true; }
      static void lcd_babystep_y() { babysteps_done = 0; lcd_goto_screen(_lcd_babystep_y);  defer_return_to_status = true; }
    #endif
    static void _lcd_babystep_z() { _lcd_babystep(Z_AXIS, PSTR(MSG_BABYSTEPPING_Z)); }
    static void lcd_babystep_z() { babysteps_done = 0; lcd_goto_screen(_lcd_babystep_z); defer_return_to_status = true; }
```