MarlinFirmware/Marlin #5032: UBL on CoreXY with inductive probe

mtodman, 2016-10-16:
Firstly, I apologise if this is the wrong forum to post this type of question on. If so, please feel free to direct me to the correct location.
I have a CoreXY with 300 x 300 aluminium bed & inductive probe. I've been trying to get the UBL Marlin tree working with my printer however I'm having issues getting the firmware to compile.

I have started with the Roxy-3D config files which I can compile without any issues. My problem arises when I uncomment #define FIX_MOUNTED_PROBE (which I assume I need to do to use an inductive probe?).
I then get the following error: "Please define only one type of probe: Z Servo, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or FIX_MOUNTED_PROBE.".
I can't figure out how to undefine one of the other probe types.

Does anyone have their printer working with UBL and indictive probe?
Does this dev release support inductive probes?

Any help would be appreciated.

Roxy-3D, 2016-10-16:
I presume you have your printer working with RC-7, right?   If not...  That is the first step.  Get your printer running with RC-7.  The devel_ubl branch was forked from RC-7 on July 4th.   A few names in the Configuration.h file have changed since then, but if you have a Configuration.h file that works with RC-7, we will be very close to getting you going on the UBL branch.

> Does anyone have their printer working with UBL and indictive probe?

I believe @adamfilip is using an inductive probe.   Adam, can you post your Configuration.h file as a reference to help mtodman ?

> Does this dev release support inductive probes?

Yes.  For sure it does.   But I don't have an inductive probe and have never used one.   So looking at a working Configuration.h file from somebody that is using an inductive probe is probably the best way to go.

> Any help would be appreciated.

You should have some answers soon.

moth4017, 2016-10-20:
hi I'm have an issue with my core xy where the nozzle height is around 8mm above the bed on the first layer, the mesh bed is working as I can see the z axis moving.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/541048/Configuration.txt)

Roxy-3D, 2016-10-20:
For right now...   The answer is to adjust your Z_PROBE_OFFSET_FROM_EXTRUDER  to get the nozzle at the correct height.  Be sure to do a M502 and M500 each time you make a change in your Configuration.h file and rebuild the firmware.

You should also turn on the Z-BABYSTEPPING in Configuration_adv.h  and the QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN in Configuration.h   That will allow you to very easily position the nozzle where it needs to be.

moth4017, 2016-10-25:
HI Roxy

Making headway and now pleased with the Mesh leveling function.

So here is my dilemma:-

How do i do an auto mesh level before printing, as my bed changes shape depending on how hot the bed is  and its a 6mm slab of aluminium but 400mm x 400mm

Regards

Mark

Roxy-3D, 2016-10-25:
I've seen the same thing with my FolgerTech i3-2020.    I'm not sure it is the bed 'changing shape' but for sure I need to preheat things to get consistency.   I turn on the nozzle and the bed and just wait 5 or 10 minutes before I do a first print.   And I purge some filament before I start the print.

I'm not sure what to tell you.   But it seems the dimensions of the machine are changing more than we realize as it heats up.    In my case, I have aluminum rails for everything.   It maybe these are absorbing some of the waste heat and changing size.   I don't know.

adamfilip, 2016-10-25:
Yes preheat bed and nozzle. before doing UBL, there is some expansion for sure happening.
If you print stuff at with large differences in temperture.. example PLA is much cooler then ABS
you could preheat for PLA.. do UBL and Save as Mesh #1 and do the same with ABS and save as Mesh #2 
and then just load the proper mesh in your start up gcode.

moth4017, 2016-10-26:
ok  what i would really like was the option to do a auto mesh bed level before printing. a bit like the prusa Mk2

Roxy-3D, 2016-10-26:
> ok what i would really like was the option to do a auto mesh bed level before printing. a bit like the prusa Mk2

The 3-Point leveling can be applied to the Mesh and used before printing.   The Grid Based version will show up when there is some free time.   But really...   The 3-Point is fine for now because the Mesh handles the bulk of the corrections.   You really don't even need the 3-Point leveling applied to the Mesh.

moth4017, 2016-10-26:
Hi any way to put in the startup script  a G28, g29s1 Auto instead of having to manual set the levels or use Gcode for each spot, Not found a way to auto run mesh bed leveling yet

Regards

Mark

Roxy-3D, 2016-10-26:
> Hi any way to put in the startup script  a G28, g29s1 

Well...   If you were going to put a G29 into the startup script, I think it makes more sense to LOAD a mesh instead of saving a mesh.   You might want to do a G28, G29 L1   

Here is the point...   It takes a little bit of work to generate (and validate) a Mesh.   That is something that needs some human control and judgement behind it.   Once you understand how the system works, you can generate a perfect mesh for a new piece of glass using the various tools in 10 or 15  minutes of human time.   (The auto probing the bulk of the mesh takes 40 minutes for my slow Folger Tech i3-2020.  Printing the Mesh Validation Pattern can take 10 minutes)    Once you have this Mesh defined, you would save it in a slot number that makes sense.    

And from then on, you can pull up that mesh out of that slot number and use it.    And the system will automatically pull up what ever the last saved mesh is.    So it is very possible that once you get your Mesh defined how you want it, you never give the system another G29 until you need to change something on your mesh.

moth4017, 2016-10-26:
ok let me try and explain another way

saving a mesh and reloading  it is an option but this assumes that your using the same temperatures  for the nozzel and bed  and also that if the bed was removed it returns back to the same place.

So m thought was that as i have an inductive sensor and when i use the reputability test   it comes out at 0.001  . i could use an auto bed mesh script with auto save before printing as this would only take a few minutes which is nothing to a 20 hour print.

i thourght about a script at start up like this but does not seem to work

g28
g29 s1
g28 z0
g29 s2
g28 z0

etc etc

Roxy-3D, 2016-10-26:
> saving a mesh and reloading it is an option but this assumes that your using the same temperatures for the nozzel and bed and also that if the bed was removed it returns back to the same place.

It is important to put the glass back in the same location as much as possible.   It isn't super critical, but I have black triangles on the bottom of my glass so it can be put in alignment with the triangles on the MK2 heat bed.

I don't know how much variance comes from printing at 205 versus 235.   But I do know I have to let my printer warm up if I want it to put down a first layer without any Z-BabyStepping.

The G29 Z is there to adjust the mesh compared to the nozzle height.   That isn't measuring new points for the mesh and isn't doing what you are hoping it to do.

Please try getting a mesh accurately dialed in and saved and try it with the system in an Active state.  That way, the printer will power up and just use the mesh.    I don't know this for a fact, but I think it is very possible that printing at different temperatures isn't going to be a problem.

closed 2017-10-31 by thinkyhead