MarlinFirmware/Marlin #5034: Bed Heating issues with LIN_ADVANCE

wstallwoods421, 2016-10-16:
I've noticed with LIN_ADVANCE enabled on RC7 that smaller object prints with short layer times seem to result in the printer not being able to keep the bed warm, (hardware confirmed ok as bed heats and stays heated when printer not printing or printing larger more complex layers)

This seems to be related to layer complexity rather than setup or slicer because printing 5 of the same item at the same time on the bed solves the problem and the print can complete. Otherwise the print aborts eventually

https://s3-eu-west-1.amazonaws.com/ws-share/Printer_crash.tiff

Discoverd on a E3D BigBox , Marlin running on a Rumba, running at 4500.0 mm/min, 11500 baud as suggested when enabling LIN_ADVANCE, PID's calibrated for Hotend and Bed (Bed set to PID mode)

Blue-Marlin, 2016-10-16:
My crystal ball let me see this story.
Your slicer detects a short layer time and orders the part fan to run. Because the nozzle is still very deep the bed is cooled, the heater can't keep up and the temperature is dropping. After a while you get a theater protection error including a kill.
Likely this is completely independent from LIN_ADVANCE.
Please retry without advanve.
Please try with longer min layer times.

Roxy-3D, 2016-10-16:
If indeed it is a small part with a short layer time...   Try moving it some where else on the bed so the fan is not cooling the bed location with the thermistor that measures the bed temperature.

Blue-Marlin, 2016-10-16:
@Roxy-3D 
If the sensor is in the middle of the bed your way is mostly counter-productive - because the part usually is shielding the sensor area.

Roxy-3D, 2016-10-16:
Mostly, I was hoping to get another data point in order to understand the problem    I don't know the shape of the object being printed, or where the sensor is.   We also do not know if the part is covering up the sensor but that would be good to know.  If it is covering up the sensor, the fan is probably not entering into the equation.

wstallwoods421, 2016-10-16:
Sorry I was trying to attach some files to help with this (including the gcode I printed with) Part does cover the sensor and fan is set very very low (20% in this case) I've not seen problems with my fan cooling the bed before like this and my 24v bed usually has enough heating power to cope (80 is nowhere near what it can cope with) but it's a good thought. I'll retest tomorrow without fan and close the issue if I can't reproduce it like that.

AndyZuerich, 2016-10-18:
Hi,

just another idea (but I am not really sure if this applies at all, therefore it is just a thought, and possibly my thought is completely wrong):
Please have a look at this issue [https://github.com/MarlinFirmware/Marlin/issues/4931]
Could it be possible that the thermal manager is not working properly when the layer complexity is too high (as you described)? When you print 5 of the same items, you always have the movement between the items (which might be high speed, but without complex calculation), and then the printer has time to call the heater manager.
As thinkyhead described in the other issue, the heater manager is possibly not called regularly enough.

boelle, 2016-11-05:
@thinkyhead  what is the status on this one?

thinkyhead, 2016-11-06:
> what is the status 

Mysterious. But the advance code has gotten a lot of good tweaks recently. I'll close this for now and if the problem is still lingering we can always re-open it.

closed 2016-11-06 by thinkyhead