MarlinFirmware/Marlin #5039: RCBugFix won't compile - planner.cpp syntax issue

Bob-the-Kuhn, 2016-10-18:
The latest RCBugFix won't compile because of a syntax issue in planner.cpp

Lines 1023-1026 now read
```
    #define LIMIT_ACCEL(AXIS) do{ \
      if (max_acceleration_steps_per_s2[AXIS] < (accel * block->steps[AXIS]) / block->step_event_count) 
        accel = (max_acceleration_steps_per_s2[AXIS] * block->step_event_count) / block->steps[AXIS]; 
    }while(0)

```
1024 & 1025 need a "\" at the end.  They should read
```
    #define LIMIT_ACCEL(AXIS) do{ \
      if (max_acceleration_steps_per_s2[AXIS] < (accel * block->steps[AXIS]) / block->step_event_count) **\**
        accel = (max_acceleration_steps_per_s2[AXIS] * block->step_event_count) / block->steps[AXIS]; **\**
    }while(0)
```

Blue-Marlin, 2016-10-18:
Fix for the commit dc2cb84 (Revert acceleration limiting to avoid overflow) #5037

Bob-the-Kuhn, 2016-10-18:
Great

closed 2016-10-18 by Bob-the-Kuhn