MarlinFirmware/Marlin #5041: About DIY printer getting two times AUTOHOMING and also filament is too excessly extruding

Shrikant888, 2016-10-19:
Hi All!!

I have trying to solve a problem.

When I am selecting AUTOHOME option in LCD and then given a PRINT option, After that nozzle is reaching reference temperature i,e 205 Degree and again its getting AUTOHOMING and then starts to print.

I am not getting. Please help... I wanted that manually I will select AUTOHOME from LCD and then it has to start PRINT after reaching desired temperature.

and also during AUTOHOME position at (0,0) Nozzle is extruding too much.. Is there any thing I have to change is Marlin.....???

Please reply.

FHeilmann, 2016-10-19:
About the double homing:

Check the start script of your slicer. Maybe it adds another G28 which is executed as you start your print.

About the extruding during homing:

Your nozzle is heating up, which causes the material that is still left in the nozzle to slowly ooze out. Not much you can do about it, you can try retracting some at the end of your prints so it doesn't ooze on the next.

Shrikant888, 2016-10-19:
Hi there!!!

I am using CURA which has AUTOHOME gcode in software. But the point is I have already taken some prints before 10 days. But I didn't get DOUBLE HOMING and also excess extruded nozzle.

Please tell where I can make modification regarding DOUBLE AUTOHOME and also Excess Nozzle.... Please tell in both SOFTWARE and also in FIRMWARE.

Thanks in Advance.

FHeilmann, 2016-10-19:
These things don't change by themselves. Have you updated anything since your prints were working ok? If so, look at these updates.

Shrikant888, 2016-10-19:
I am generating gcode from CURA and storing in SD CARD. then giving AUTOHOME from LCD. Upto 205 degrees for model cube PLA. So here is whole process....

1. Generate Gcode from CURA put in SD Card
2. LCD options AUTOHOME
3. Print PLA cube model 205 degrees LCD.
4. At nearly 200 to 205 degrees exceess EXTRUDE at HOMING
5. After reaching 205 degrees again its AUTHOMING..
6. Nozzle move to center of platform..

Double AUTOHOMING is OK for me. I am worried about excess filament nozzle at HOMING position..

Please tell me where can I change..

Shrikant888, 2016-10-20:
Hi all !!!

I have attached a file to under stand clearly. Excess extrusion at HOMING and Bottom layer of Model is thicker. Please advice what shall I do now?
[image]

thinkyhead, 2016-10-22:
Marlin doesn't "know" anything about any "first layer." It just obeys the GCode.
So this is definitely a slicer issue. Check your Cura profile settings.
If you need more help, http://forums.reprap.org/

closed 2016-10-22 by thinkyhead