MarlinFirmware/Marlin #5042: [Minor cosmetic bug] When doing Manual Mesh Leveling, if one has to "search" for more than 10mm the shown distance on the display wraps around

FHeilmann, 2016-10-19:
Probable because the leading 1 is overwritten.

To be more precise, if I start manual mesh leveling and have to search for more than 10mm to find my bed the displayed value will look something like this

`-9.950`
`-9.975`
`-0.000`

while it should be -10.000 in the last row. Also, which speed is Marlin using when moving between probing points? It seems to me that it is using the maximum speed for at least the Z-Axis, because I found out while Mesh leveling that my Z-Axis will skip steps at that speed. Wouldn't it make more sense to use the Homing Feedrates (or at least not the maximum feedrate) for this?

Blue-Marlin, 2016-10-19:
Ähhh - If your z-axis is loosing steps at DEFAULT_MAX_FEEDRATE it is configured too high.

FHeilmann, 2016-10-19:
@Blue-Marlin yes, I fixed that already. I simply wanted to put the matter up for discussion. Of course this report is about the display error.

thinkyhead, 2016-10-22:
Why is your search distance so large? At each point the starting value should be zero, with the nozzle already close to the bed.

FHeilmann, 2016-10-22:
I had to set the value this large because my nozzle ends up quite high above the bed after the homing process.

I assume this is because of these two parameters (?):

```
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```

thinkyhead, 2016-10-22:
Well, this procedure has been screwed up, then. I had the behavior working pretty well at one point. Someone must have meddled with bad assumptions.

First and foremost, the nozzle should always return to the bed level at each point so you can quickly adjust it. It should not remain sitting above the bed.

FHeilmann, 2016-10-22:
I'll try once I get back to my printer (I think you can relate to that problem ;))

Thanks for the assistance so far!

mosh1, 2016-11-19:
I noticed a similar problem after upgrading to latest RCBugFix where the nozzle just sits too high after homing. When I do the mesh leveling probe I also need to lower the nozzle over the default of 4mm.  The last time this worked was before #4821 was implemented, so either that changed the behavior or something between then and now changed.

My Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES values were kept the same from the previous RC7 Z_PROBE_DEPLOY_HEIGHT & Z_PROBE_TRAVEL_HEIGHT values.

Not sure if a separate bug needs to be filed or if this is intentional behavior and i need to change some other setting to keep the nozzle closer to the bed.

mosh1, 2016-11-20:
In the end I had to set both Z_CLEARANCE_BETWEEN_PROBES and Z_CLEARANCE_DEPLOY_PROBE to 0 get the same behavior as RC7. The default of Z_CLEARANCE_DEPLOY_PROBE to 10 has the nozzle over 10mm above my bed and I can't do the MBL without changing MESH_HOME_SEARCH_Z to a large value.

thinkyhead, 2016-11-21:
@mosh1 If you could post a detailed log of `G28` then we should look at how MBL behaves there. First enable `DEBUG_LEVELING_FEATURE` and restore the settings you were using before. Enter `M111 M247` to enable all debug options except for `DRYRUN`. Perform your usual manual bed leveling and finally do a `G28`. Post the log here and we'll see if we can figure out where the issue lies.

mosh1, 2016-11-22:
Here is the output of G28

Recv: echo:G28
Recv: >>> gcode_G28
Recv: Machine Type: Cartesian
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X:-65 Y:0 Z:0.00 (Left-Center & Same Z as Nozzle)
Recv: Mesh Bed Leveling (enabled)
Recv: MBL Adjustment Z0.00
Recv: MBL was active
Recv:   current_position=(80.00, 15.00, 4.00) : Set Z to pre_home_z
Recv:   current_position=(80.00, 15.00, 4.00) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -352.50, 0.00)
Recv:   current_position=(80.00, 0.00, 4.00) : sync_plan_position
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, 5.00, 0.00)
Recv:   current_position=(80.00, 0.00, 4.00) : sync_plan_position
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, -10.00, 25.00)
Recv:   current_position=(80.00, 0.00, 4.00) : sync_plan_position
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 235.00
Recv: > home_offset[Y] = 0.00
Recv:   current_position=(80.00, 0.00, 4.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv:   current_position=(80.00, 0.00, 4.00) : sync_plan_position
Recv:   current_position=(80.00, 0.00, 4.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv:   current_position=(80.00, 0.00, 4.00) : > homeY
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -450.00, 0.00)
Recv:   current_position=(0.00, 0.00, 4.00) : sync_plan_position
Recv: echo:busy: processing
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, 5.00, 0.00)
Recv:   current_position=(0.00, 0.00, 4.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, -10.00, 25.00)
Recv:   current_position=(0.00, 0.00, 4.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 300.00
Recv: > home_offset[X] = 0.00
Recv:   current_position=(0.00, 0.00, 4.00) :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position=(0.00, 0.00, 4.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 4.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv:   current_position=(0.00, 0.00, 4.00) : > homeX
Recv: Z_SAFE_HOMING >>>
Recv:   current_position=(0.00, 0.00, 4.00) : sync_plan_position
Recv:   destination=(80.00, 15.00, 4.00) : Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(80.00, 15.00, 4.00)
Recv: <<< do_blocking_move_to
Recv: >>> homeaxis(Z)
Recv:   current_position=(80.00, 15.00, 4.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(3.00)
Recv: >>> do_blocking_move_to(80.00, 15.00, 4.00)
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -450.00, 0.00)
Recv:   current_position=(80.00, 15.00, 0.00) : sync_plan_position
Recv: echo:busy: processing
Recv: <<< do_homing_move(Z)
Recv: Move Away:
Recv: >>> do_homing_move(Z, 3.00, 0.00)
Recv:   current_position=(80.00, 15.00, 0.00) : sync_plan_position
Recv: <<< do_homing_move(Z)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Z, -6.00, 1.00)
Recv:   current_position=(80.00, 15.00, 0.00) : sync_plan_position
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: <<< do_homing_move(Z)
Recv: >>> set_axis_is_at_home(Z)
Recv: For Z axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 300.00
Recv: *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
Recv: > zprobe_zoffset = 0.00
Recv: > home_offset[Z] = 0.00
Recv:   current_position=(80.00, 15.00, 0.00) :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position=(80.00, 15.00, 0.00) : sync_plan_position
Recv:   current_position=(80.00, 15.00, 0.00) : > AFTER set_axis_is_at_home
Recv:   current_position=(80.00, 15.00, 0.00) : set_probe_deployed
Recv: deploy: 0
Recv: do_probe_raise(3.00)
Recv: >>> do_blocking_move_to(80.00, 15.00, 3.00)
Recv: <<< do_blocking_move_to
Recv: >>> do_blocking_move_to(80.00, 15.00, 3.00)
Recv: <<< do_blocking_move_to
Recv: <<< homeaxis(Z)
Recv: <<< Z_SAFE_HOMING
Recv:   current_position=(80.00, 15.00, 3.00) : > (home_all_axis || homeZ) > final
Recv:   current_position=(80.00, 15.00, 3.00) : sync_plan_position
Recv: MBL has mesh
Recv: MBL Z homing
Recv:   current_position=(80.00, 15.00, 4.00) : sync_plan_position
Recv:   current_position=(80.00, 15.00, 4.00) : MBL adjusted MESH_HOME_SEARCH_Z
Recv:   current_position=(80.00, 15.00, 4.00) : clean_up_after_endstop_or_probe_move
Recv: <<< gcode_G28
Recv: X:80.00 Y:15.00 Z:4.00 E:0.00 Count X: 12800 Y:1350 Z:6400
Recv: ok
Send: M105

mosh1, 2016-11-22:
Notes: I couldn't MBL since the default is 4mm of searching and the nozzle was higher than 4mm. I went all the way to 4mm on all points.

Z_CLEARANCE_DEPLOY_PROBE & Z_CLEARANCE_DEPLOY_PROBE both set to 3 as before. The 'issue' perhaps is the last do_probe_raise at the end which didn't happen before.

Again not sure if the old behavior was correct or if this is now the correct behavior and I should tweak my Z_CLEARANCE_DEPLOY_PROBE & Z_CLEARANCE_DEPLOY_PROBE to compensate.

thinkyhead, 2016-11-26:
According to the log you have a probe. Why not use ABL?

mosh1, 2016-11-26:
I was thinking of giving ABL a try. I may be wrong but there didn't seem to be a way to do ABL just once  and then save it (like you can with MBL). I don't want to run ABL every print because not much changes and my IR probe can be inconsistent of the bed glass is not fully clean (such as with gluestick layer). If I can do ABL once and save the mesh then I will switch to it.

Roxy-3D, 2016-11-26:
> If I can do ABL once and save the mesh then I will switch to it.

With the UBL branch, you can automatically probe the bed to form the bulk of the mesh.  You can save the mesh so it is active every time you power up the printer.  You can also have the machine print a Mesh Validation Pattern that can be used as input to quickly edit the mesh to perfection.   And if your Z-Probe is inconsistent...  It doesn't even matter.   It's easier to use a probe, but you can build a mesh without a Z-Probe if you want to.

mosh1, 2016-11-26:
@Roxy-3D your UBL work seems awesome and that's what I intend on using in the end but I really make use of all the LINEAR ADVANCE work that's gone into the latest RCBugFix branch. It makes a huge difference to the print quality and the current MBL is livable. Once UBL is merged in, I will switch to it immediately.

thinkyhead, 2016-11-26:
@mosh1 As it happens I need to make it possible to save the bed leveling to EEPROM for one of my clients anyways. When I have that ready I'll let you know.

closed 2017-06-12 by FHeilmann