MarlinFirmware/Marlin #5046: [Request] Support for MAX31856

Grogyan, 2016-10-19:
While the MAX31855 is still very useful for "ordinary" K-type thermocouples, Maxim Integrated their device with a more versatile device which can handle a greater vareity of thermocouple types, including K type.
It also has a SPI data in, for configuring.

The method which data is read is also different from he MAX6675 and MAX31855, and isn't a drop in replacement, and therefore will need to be catered for in Marlin

thinkyhead, 2016-10-22:
At some point pretty soon I will put in the time to implement this. Currently focused on fixing a few minor issues to prep for the RC8 release.

esenapaj, 2016-10-22:
for reference:
adafruit/Adafruit_MAX31856: Arduino Library for Adafruit MAX31856
https://github.com/adafruit/Adafruit_MAX31856

PlayingWithFusion/PWFusion_MAX31856: Maxim MAX31856 Thermocouple to Digital Converter Arduino Library
https://github.com/PlayingWithFusion/PWFusion_MAX31856

engineertype/MAX31856: Arduino library and example code for Maxim's MAX31856 Thermocouple IC
https://github.com/engineertype/MAX31856

derletztename, 2016-10-23:
I am willing to test your coding, as a non programming user, as I already ordered the MAX31856 Board and a professional thermocouple type S :) Both is supposed to arrive next week.

Grogyan, 2016-10-24:
It will be quite some time until I can get a board, so I am in no rush

derletztename, 2016-10-24:
But I am in rush, I need to print steel :p 
Actually I need to test/proof that one can print metal via the FDM/FFF process and that the layers will adhere properly. After that I can dive into the second iteration and make the printer a little less ugly and more durable and usable for non-engineers ;)

esenapaj, 2016-10-24:
@derletztename 
"1.1.1 milestone" label was added to this thread, it means that maybe official implementation of MAX31856 will be during in next year.
(1.1.0-RC (current) → 1.1.0 Release (winter or next year)→ 1.1.1 (next year or later?))

Therefore for the moment, I handle about your wish. 
I'm proceeding with implementation gradually. (This doesn't works completely yet)
https://github.com/esenapaj/Marlin/tree/support-max31856

But it's slowly, therefore I'm not able to assure that I can make working version by 27.10.16.

derletztename, 2016-10-25:
No worrys. If it works at all, I will be quite happy :)
Although I forgot to pay for the breakout board and have to wait for about another weak :(
But at least the thermocouple is here and looks very promising in it's quality :)

thinkyhead, 2016-10-26:
1.1.1 will be the first patch to 1.1.0, which will be released pretty soon.

derletztename, 2016-10-26:
As much as I like your post, I'd love to know what's pretty soon for you ;)
Is it days or weeks or month?
Maybe it's the same for esenapaj...

derletztename, 2016-10-26:
Does the little green bar on the right indicates the current status of 1.1.1?

thinkyhead, 2016-10-26:
The more testing and feedback we get from enthusiastic users, the more quickly it will be released. My plan is to release RC8 once the most pressing current issues are fixed. At the moment two issues are pressing: CoreXY `G29` is wonky and dual Z endstops are wonky.

The green bar indicates the number of milestone issues closed versus the number of milestone issues still open. These are new features that we want to put off until after 1.1.0.

derletztename, 2016-10-26:
Thanks for the explanations. I'm learning more and more about git and I like it :)

derletztename, 2016-10-30:
I received my MAX31856 board and wanted to do some testing now. How did you wire it up?
Regarding [](https://learn.adafruit.com/adafruit-max31856-thermocouple-amplifier/wiring-and-test) you need one more cable than for the MAX31855; one for SDI. Which Pin did you choose for that? I couldn't find it in the code...

Blue-Marlin, 2016-10-30:
Here you nead to talk to the board, for example to set the right type of sensr. So MOSI (Master Out Slve In) on the other boards is not needed. On RAMPS AUX3-4 = D51.

esenapaj, 2016-10-30:
@derletztename 
I'm terribly sorry, Implementation in [my branch](https://github.com/esenapaj/Marlin/tree/support-max31856) is still work in progress, it's still doesn't work yet.
My [quad-MAX31856 board](http://www.ebay.com/itm/331762435793) arrive at end of next week.

About wiring, like this.
SDI of MAX31856 is connected to D51 (MOSI) of RAMPS.
[image]

@Blue-Marlin 
In case of MAX31856, it needs command (access from host to register of MAX31856) at initialization and reading temperature.
Therefore, Marlin needs MOSI connection at using a MAX31856.

derletztename, 2016-10-31:
Thanks for the beautiful explanation esenapaj :)
And thanks for doing the coding job for us :)

esenapaj, 2016-11-20:
I changed the plug and start test and integration, but It's not stable at all.
My first integration of MAX31856 for Marlin caused MINTEMP Err.
So I tested sample programs of libraries in [previous comment](https://github.com/MarlinFirmware/Marlin/issues/5046#issuecomment-255541040), results are extremely unstable and strange...
<details>
<summary>log of adafruit's library</summary>
```
Cold Junction Temp: 16.00
Thermocouple Temp: MAX31856 thermocouple test
Thermocouple type: Unknown
Cold Junction Temp: 16.00
Thermocouple Temp: 127.99
Thermocouple Range Fault
Cold Junction High Fault
Cold Junction Low Fault
Thermocouple High Fault
Thermocouple Low Fault
Over/Under Voltage Fault
Thermocouple Open Fault
Cold Junction Temp: 16.00
Thermocouple Temp: 15.99
Cold Junction Temp: 16.00
Thermocouple Temp: 15.99
Cold Junction Temp: 16.00
Thermocouple Temp: -112.20
Cold Junction Temp: 16.00
Thermocouple Temp: -16.01
Cold Junction Temp: 16.00
Thermocouple Temp: 15.87
Cold Junction Temp: 16.00
Thermocouple Temp: 15.99
Cold Junction Temp: 16.00
Thermocouple Temp: 15.87
Cold Junction Temp: 16.00
Thermocouple Temp: 15.99
Cold Junction Temp: 16.00
Thermocouple Temp: 15.94
Cold Junction Temp: 16.00
Thermocouple Temp: 15.94
Cold Junction Temp: 16.00
Thermocouple Temp: 15.94
Cold Junction Temp: 16.00
Thermocouple Temp: 15.96
Cold Junction Temp: 15.62
Thermocouple Temp: 15.98
Cold Junction Temp: 16.00
Thermocouple Temp: 127.98
Cold Junction High Fault
Cold Junction Low Fault
Thermocouple High Fault
Thermocouple Low Fault
Thermocouple Open Fault
Cold Junction Temp: 16.00
Thermocouple Temp: 127.99
Cold Junction High Fault
Cold Junction Low Fault
Thermocouple High Fault
Thermocouple Low Fault
Over/Under Voltage Fault
Thermocouple Open Fault
Cold Junction Temp: 16.00
Thermocouple Temp: 127.98
Cold Junction Range Fault
Thermocouple Range Fault
Cold Junction High Fault
Cold Junction Low Fault
Over/Under Voltage Fault
Thermocouple Open Fault
Cold Junction Temp: 16.00
Thermocouple Temp: 127.99
Cold Junction Temp: 16.00
[... 288 lines cut ...]
TC Temp = 31.28
Thermocouple 0: no faults detected
Tint = 23.86
TC Temp = 31.78
Thermocouple 0: no faults detected
Tint = 23.86
TC Temp = 31.78
Thermocouple 0: no faults detected
Tint = 23.86
TC Temp = -0.22
Thermocouple 0: no faults detected
Tint = 23.83
TC Temp = 31.75
Thermocouple 0: no faults detected
Tint = 23.83
TC Temp = 31.75
Thermocouple 0: no faults detected
Tint = 23.83
TC Temp = 31.75
Thermocouple 0: no faults detected
Tint = 23.86
TC Temp = 31.78
Thermocouple 0: no faults detected
```
</details>

Grogyan, 2016-11-20:
The fault saying under/over voltage might have something to do with it.
What does a multimeter say for the 5V rail?

esenapaj, 2016-11-21:
Thanks for the advice.
My METEX P-10 says 4.99V - 5.00V. It seems to be very accurate and stable.

...But just now, I __blew up__ my quad-MAX31856 board by elementary mistake.
I remembered that my power unit has 5V output, thus I tried to take the board power supply from there.
But I connected a board to __400V__ connector instead of a 5V output, then board was blown up in a moment.
The line pattern evaporated, and a crater was pierced in the regulator.
[image]

...But, I never falter.
I've ordered [Adafruit's  board](https://www.adafruit.com/products/3263) just now.

@derletztename 
I'm sorry for the delay and delay.
However, the minimal implementation was completed yesterday.
you're able to test your board.
https://github.com/esenapaj/Marlin/tree/support-max31856

derletztename, 2016-11-21:
Thank you very much, I will test this tonight :)
Sorry for your loss, though. But it's a great picture for "blown up electronics". You could use it on wikipedia or something, it's really representable.

Grogyan, 2016-11-21:
Ah the magic smoke makes an appearance

Roxy-3D, 2016-11-21:
I always get depressed when I do something like that.

derletztename, 2016-11-23:
I tested your code today and got a min temp error which led to a system halt :(
I checked the wiring again but it's just as you have shown on you picture. I did not change anything else than maxtemp and the -4 for MAX31856 in you code. Should I change something else maybe?

esenapaj, 2016-11-23:
If you're using type __S__ thermocouple, additional changing is needed.
https://github.com/esenapaj/Marlin/blob/support-max31856/Marlin/Configuration.h#L258

```diff

// Select thermocouple type when use a MAX31856
// :{ '0': B Type, '1': E Type, '2': J Type, '3': K Type, '4': N Type, '5': R Type, '6': S Type, '7': T Type }
-#define THERMOCOUPLE_TYPE 3
+#define THERMOCOUPLE_TYPE 6

// Dummy thermistor constant temperature readings, for use with 998 and 999
```

derletztename, 2016-11-26:
Still got a mintemp error and the printer was halted :(
I also checked the wiring of the thermocouple again and white is at minus and orange at plus.
Is there something else I need to adjust in the software maybe?

esenapaj, 2016-11-26:
If you're using type S thermocouple and set `#define TEMP_SENSOR_0 -4` and `#define THERMOCOUPLE_TYPE 6`, 
settings are completed, and there is nothing else to do.

In my case [I got also MINTEMP Err, furthermore hardware was unstable for some reason.](https://github.com/MarlinFirmware/Marlin/issues/5046#issuecomment-261764547) (and finally I blew up it.)

So, why don't you try the sample program provided by the board maker, if you are interested.
I modified a samples a little, you can use it on your MEGA2560 + RAMPS + MAX31856 system.
When testing, please set a baud rate on serial monitor of Arduino IDE to 115200.

By adafruit: https://github.com/esenapaj/Adafruit_MAX31856
By PlayingWithFusion: https://github.com/esenapaj/PWFusion_MAX31856 (please use a SEN30005_MAX31856_example.ino)

Anyway, my new MAX31856 board is still in transit.
But, I'll check the implementation tomorrow.

esenapaj, 2016-11-27:
It seems that Hardware SPI + MAX31856 had a problem, and it was solved yesterday in Adafruit's library.
Adafruit customer service forums • View topic - Max31856 and Hardware SPI
https://forums.adafruit.com/viewtopic.php?f=19&t=104926&p=535630&hilit=max31856

derletztename, 2016-11-27:
Sorry I got you wrong in the first message. I understood it's suppose to work already.
That's great news, that the (or an) error was at the adafruit site. I am seeing forward to test again :)
Meanwhile I concentrate on the hardware if you don't mind. I am really not that good with electronics and software and it always eats up a lot of time to get into it.

esenapaj, 2016-11-27:
>Meanwhile I concentrate on the hardware

 I don’t mind. Nothing is so precious as time, thus time should be used efficiently and effective.

esenapaj, 2016-11-30:
Now my Adafruit's MAX31856 works, tentatively.
https://github.com/esenapaj/Marlin/tree/support-max31856

But implementation is really really ugly (at least SD Card read is broken) and still needs works, therefore I don't recommend that you test my branch at this time.

esenapaj, 2016-12-01:
Now temperature measurement of MAX31856 is stable.

But, the error detection function is completely malfunctioning.
When error detection function is enabled, all kinds of errors are thrown...

Blue-Marlin, 2016-12-01:
When i scanned over the MAX31856 data sheet i registered a lot of registers like max/min temperatures for internal reference and external sensor. But i can't remember if they are set to a reasonable values at boot time, or if one has to set them explicitly.

derletztename, 2016-12-04:
I am sorry for always being so slow with testing. I just was able to test the new version and it's running without errors so far :) :) :)
But it's measures to low values. I touched the tip of the thermocouple Type S and nothing happend. I put it into a flame and it slowly crawled up to about 40°C. Nevertheless before starting the tests the value was very near to the real roomtemperature value.
Maybe it's just extremely slow because of SPI related issues or something?

esenapaj, 2016-12-04:
Did you set a `#define THERMOCOUPLE_TYPE S`?
At newest version of my branch, setting was changed from number to alphabet.
https://github.com/esenapaj/Marlin/blob/support-max31856/Marlin/Configuration.h#L262

```diff
//
// Select thermocouple type when using MAX31856
//
// Options are: B, E, J, K, N, R, S, or T
//
// :[ 'B', 'E', 'J', 'K', 'N', 'R', 'S', 'T' ]
-#define THERMOCOUPLE_TYPE K
+#define THERMOCOUPLE_TYPE S
// Dummy thermistor constant temperature readings, for use with 998 and 999
#define DUMMY_THERMISTOR_998_VALUE 25
```

And, Could you try turning the power off completely?
I'm realizing that it looks like that MAX31856 keep wrong setting when it was given or an incorrect operation is performed.

esenapaj, 2016-12-04:
>without errors so far

Currently, MIN / MAX temp error detection function of Marlin works, however MAX31856's own error detection function is omitted.
Because it's completely malfunctioning at this time.
If you interested in MAX31856's own error detection function, when you uncomment a `#define MAX31856_ERROR_DETECTION`
https://github.com/esenapaj/Marlin/blob/support-max31856/Marlin/Conditionals_post.h#L229
, it will be effective.

And if you'd like to test a another method of SPI communication, when you uncomment a `#define USE_SPI_LIB_FOR_THERMOCOUPLE`
https://github.com/esenapaj/Marlin/blob/support-max31856/Marlin/Conditionals_post.h#L227
, it will be effective.

esenapaj, 2016-12-04:
By the way this is unrelated to MAX 31856, [you have a MAX31855SASA soldered board](https://github.com/MarlinFirmware/Marlin/issues/4938#issuecomment-254874626).
If you combine it with a new (expensive) S type thermocouple, is there a possibility that you can measure over 1400 degrees?

derletztename, 2016-12-04:
Thank you for the fast answer. Yes I did choose the type S thermocouple this time ;)
What do you mean by turning off the power, for the software testing there is no external power on the RAMPS. 
And no, I don't need the error function for now, I just meant no mintemp or other software errors.
And yes I still have the MAX31855SASA soldered board, I can try combining it with the high quality thermocouple, I thought there were other issues, but why not trying and find out :) Thanks for the hint
I'll come back with results soon

esenapaj, 2016-12-04:
>What do you mean by turning off the power, for the software testing there is no external power on the RAMPS.

In that case, removing the USB cable that connecting between the RAMPS and the PC, and leaving it for about 20 to 30 seconds then the charge stored in the capacitor is escaped.

derletztename, 2017-04-15:
Hey Guys
It has been quite a while. Sorry for that.
So in the meantime, I got a new professional thermocouple, the first one had an issue and the manufacturer made me a new one. This seems to work find now :)
Though, still with the MAX31856 I am measuring very slow raising numbers and too low temperatures. Should be in the hundreds, but is only about 45°C.
And with the modified MAX31855 board, I again face the 1000°C max limit, which I fail to solve again. I am not sure which version was the working one, but the RCBugFix won't help me with the 1000°C limit. Though I remember we were beyond this, the last barrier we face was 1400°C and this was an issue with the thermocouple(s). Could you maybe help me again with this, esenapaj?

Despistado, 2017-12-04:
Good day, im new here , and i really need your help

i was testing the code above from esenapaj  for my adafruit universal MAX31856 but cant go more than 440 celsius, and  it rise very slow the sensor get really red hot when  i heat it with a map pro torch , i dont think that is real temperature, **my thermocouple "K"** is this [https://www.amazon.com/Temperature-Thermocouple-Ceramic-Furnace-CR-01/dp/B01N5MTOES/ref=sr_1_5?ie=UTF8&qid=1509944895&sr=8-5&keywords=High+Temperature+K+Type+Thermocouple+Sensor+Ceramic+Kiln+Furnace+2372%C2%B0F+1300%C2%B0C&dpID=41NDtxlpwwL&preST=_SX342_QL70_&dpSrc=srch](url)

**Im using arduino MEGA2560 and ramp 1.4** 

**and 2 extruders**

ambient temperature reading is 26 degrees and im sure that real temperature because my second extruder sensor PT100 its reading 24

im testing adafruit max31856 because my old ad597 cant get more than 395-400 celsius degrees, i set max temp to 700 degrees, and -4 for the sensor  in configuration.h but nothing change, can someone help me please? , is there a solution for this? what im doing wrong

BTW. i test the example file from adafruit  and works great i was reading more than 700 degres when sensor get red hot, what im trying to do is 3d printing PEEK for delamination issue but im stuck, sorry for my bad english.

Thank you in advanced

closed 2018-02-10 by thinkyhead