MarlinFirmware/Marlin #5047: LCD displays two rows of blocks when using premade Delta Configuration.h

silverx10, 2016-10-20:
As the title suggests: When I try to use the Configuration.h and Configuration_adv.h from the Delta\Kossel Mini folder, my REPRAP_DISCOUNT_SMART_CONTROLLER won't show anything more than two horizontal rows of blocks. Using the default Configuration.h file, the display works, but obviously is configured for cartesian vs. delta builds.

I've ensured that my motherboard is specified as the RAMPS_14_EFB and that the respective LCD is uncommented.

esenapaj, 2016-10-20:
If you comment this line, perhaps problem is solved.
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/example_configurations/delta/kossel_mini/Configuration.h#L1184

silverx10, 2016-10-21:
Nailed it. Wow, I wish that I'd been looking for things that _should've_ been commented out instead of just looking for things that I could uncomment.

Thank you!

closed 2016-10-21 by silverx10

thinkyhead, 2016-10-22:
Pretty sure `RCBugFix` will catch this error and prevent compilation.