MarlinFirmware/Marlin #5048: Bottom layer thicker and Excess nozzle extrude at HOMING position

Shrikant888, 2016-10-20:
Hi All!!!

I am trying to take models. At [b]HOMING[/b] position its excess extruding. I have been used [b]CURA[/b] earlier but now its extruding nealy 3-4mm at 200-205 degrees. Also getting thick layers for 2-3 layer at bottom. Please help out with this.

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000 ; Put printing message on LCD screen
M 117 Printing...
[image]

Blue-Marlin, 2016-10-20:
Please learn to use you __CURA__. Marlin is just doing what it's told to do.
```G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops     // the second homing x/y
G28 Z0 ;move Z to min endstops             // z
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock // the extrusion
G92 E0 ;zero the extruded length again
G1 F9000 ; Put printing message on LCD screen
M 117 Printing...
```
The 'over-extrusion' is so much that likely 'raft' or 'brim' is selected.

Shrikant888, 2016-10-20:
Thanks for replying.

Ok now I am getting intial layers bit thicker and remaining layer as defined. How to solve that problem?

Blue-Marlin, 2016-10-20:
In this (Marlin) bug tracker you will find the Marlin experts. For questions about CURA ask there.

thinkyhead, 2016-10-22:
@Shrikant888 Keep playing with your slicer settings. The initial layer can be thicker and have more extrusion, and often that's the default. But you can tweak that however you like. Best of luck!

closed 2016-10-22 by thinkyhead