MarlinFirmware/Marlin #5049: QUESTION: Default EEPROM Values when flashing

Kaibob2, 2016-10-20:
Just a short question, with, i hope a short answer. I've been on a Juli RCBugFix Version since yesterday with enabled EEPROM support.
Yesterday i downloaded the recent RCBugFix from here and edited the configuration files to match my settings.
After flashing the updated Version i checked the settings with M501 and the printer responded something like: "V24 Default Values...received" and the parameters where not the ones which i had in the EEPROM before but some default parameters.
After storing them with M500 i reflashed the new version again and the EEPROM values are correct. Serial answer is something like "stored settings received..."
Why is this? I thought the EEPROM values always override the default parameters and **stay persistent even when flashing**.

Blue-Marlin, 2016-10-20:
They do - except the EEPROM layout has changed. 
It did change between then and now.

Kaibob2, 2016-10-20:
I see. How do I know before I flash? This time I made a Screenshot with the M501 response, so I had a backup. But what if I flash without knowing that the eeprom will be overwritten?

Blue-Marlin, 2016-10-20:
When this (https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/configuration_store.cpp#L39) number changes.

Kaibob2, 2016-10-20:
Great. This is exactly what i needed to know. Thanks for the fast response.

closed 2016-10-20 by Kaibob2