MarlinFirmware/Marlin #5050: CONTROLLER_FAN turning on when bed is heating?

FHeilmann, 2016-10-20:
I have the following setup:

MKS_BASE with the following `pins_MKS_BASE_CUSTOM.h`

#define MOSFET_D_PIN 7

```c
#include "pins_RAMPS.h"

#ifdef X_MAX_PIN
  #undef X_MAX_PIN
  #define X_MAX_PIN -1
#endif

#ifdef Y_MAX_PIN
  #undef Y_MAX_PIN
  #define Y_MAX_PIN -1
#endif
```

and the controller fan defined like so:

```c
#define CONTROLLERFAN_PIN 2 //Pin used for the fan to cool controller (-1 to disable)
#define CONTROLLERFAN_SECS 60 //How many seconds, after all motors were disabled, the fan should run
#define CONTROLLERFAN_SPEED 100  // == full speed
```

I basically use thee X_Max pin, which i don't need to drive a simple fan extender, which drives the fan.
The problem I have now is, whenever the bed turns on, but the stepper motors aren't, the CONTROLLER FAN turns on as well with (as far as I can tell, it's set speed). It also has a pretty nasty whine which I can't seem to eliminate by switching between SOFT_PWM FAST_PWM and the other parameters that are supposed to cure this problem.

Is this intended?
I am using PIDTEMPBED if that makes any difference.

thinkyhead, 2016-10-22:
Just a side-note… This is all you need to undefine the X and Y max pins:
```cpp
#define MOSFET_D_PIN 7

#include "pins_RAMPS.h"

#undef X_MAX_PIN
#undef Y_MAX_PIN
```
…though you shouldn't really need that for any reason, as long as other settings are ok.

thinkyhead, 2016-10-22:
Apparently this is the normal designated behavior. If you want to change it:
```diff
- if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_bed > 0
+ if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON
```

thinkyhead, 2016-10-22:
> It also has a pretty nasty whine

At all speeds? What about when turned on by a non-PWM pin? (You should be able to plug it into any 5V pin.)

FHeilmann, 2016-10-24:
Yes, at all speeds. If I use a non-PWM pin on the latest RCBugfix (synced today, used Z-Max (Pin 19) instead of X-Max which, according to the arduino website is not a PWM pin) I get the same issues as described in #5011 where I can either turn the fan to full speed (which I don't really want) or the fan won't turn on at all.

thinkyhead, 2016-10-24:
> Yes, at all speeds.

If the whine doesn't go away when you use a non-PWM pin, then it's just a whiny fan, and all you can do is replace or physically adjust it. Perhaps a rubber gasket between the fan and its housing will reduce acoustic amplification.

FHeilmann, 2016-10-24:
on the non-pwm pin i can only go full speed and the whine is not present there. on the pwm pin the whine is there at all speeds.

thinkyhead, 2016-10-24:
>> What about when turned on by a non-PWM pin?

> on the non-pwm pin the whine is not present

I see…

Well, currently there's no simple way —apart from adding custom code— to change the PWM rate. It can be done, but the pin's associated interrupt must not be in use.

FHeilmann, 2016-10-25:
Yes it happens at all speeds. I know this may sound stupid but how would I
determine which PIN is NON-PWM and which isn't?

thinkyhead, 2016-10-26:
@FHeilman In Marlin we can use the `GET_TIMER(pin)` macro to see if a pin has a timer.

closed 2016-11-14 by FHeilmann