MarlinFirmware/Marlin #5051: Given print at 205 degrees but machine printing at 207 degrees

Shrikant888, 2016-10-21:
Hello All!!!

I am giving a CUBE print 20x20x10mm at PLA 205 degrees but model is been printing at 207 degrees and then decreasing to 205 degrees normally...

Any one can help me ...?

Kaibob2, 2016-10-21:
I would suggest reading [this guideline for PID tuning](http://reprap.org/wiki/PID_Tuning)

Shrikant888, 2016-10-21:
Hi again!!!

I run the commands M303 S200 C5 and 

Kp: 28.22
Ki: 2.25
Kd: 88.31

But whats the value for PID-C...? C for CYCLES...? That is 5...?

Kaibob2, 2016-10-21:
Have you even read the guideline?
Run M303 E0 S200 C8

E0 is the extruder number
S200 is the target temperature
C8 is the number of cycles

This will take some time. When it's finished there will be a response in your serial monitor which will look something like this:

> bias: 92 d: 92 min: 196.56 max: 203.75
Ku: 32.59 Tu: 54.92
Clasic PID
Kp: 19.56
Ki: 0.71
Kd: 134.26
PID Autotune finished ! Place the Kp, Ki and Kd constants in the configuration.h

Insert your determined Kp, Ki, and Kd values in your [configuration.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration.h#L289) file and upload it.

If you're using EEPROM settings you have to upload them with:
`M301 P19.56 I0.71 D134.26 `
And store them with
`M500`

Shrikant888, 2016-10-21:
Ok thanks for replying. I have done by command 

M303 E0 S205 C3

Got AUTOTUNE value and uploaded.

But also getting as 208 degrees... Model is given at 205 from CURA...

Its going upto 208 and then little extruding at HOMING and then reducing at 205 then starts to print....

Please help...

Kaibob2, 2016-10-21:
A little overshoot on initial heating up is fairly normal. If the Extruder temperature stays within a small range around you're given target temperature during printing everything is fine. 
It's quite hard to get the temperature to "land" exactly at the given value and stay there.
You can experiment with the "No-overshoot" settings in the guideline:
```
    Kp=Kp_classic/3 
    Ki=Ki_classic 
    Kd=Kd_classic*8/3
```

Kaibob2, 2016-10-21:
There is also not really a suitable way of avoiding some filament drooping out of the nozzle during heating up or homing. As long as there is molten filament in the extruder it will slowly droop out of the nozzle, even rectrating will not completely eliminate this. After some time all filament will be drooped out. This makes priming or skirt lines absolutely necessary before the print of the actual model starts.

Blue-Marlin, 2016-10-21:
Since some time you can use the 'U' parameter to immediately __Use__ the tuned values. (https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L6393-L6400)

Some degree overshot are pretty normal wit PID. You can fine tune this a bit with `PID_FUNCTIONAL_RANGE`

Shrikant888, 2016-10-21:
Thanks for all supports...

All right instead of CLASSIC, I wanted to use NO Over shooting...

Here are the my values...

M303 E0 S205 C3

For 205 Degrees... Values are KP=27.61    KI=2.08    KD=91.83     C=3

Please tell how can avoid over shooting.....That is I wanted exactly 205 degrees... It should not go beyond that.... Like depending on Model temperature is should take exactly... It might be 205 or 208 etc...

Please help me....

Shrikant888, 2016-10-21:
OK I have calculated the NO OVERSHOOT VALUES as

  Kp=Kp_classic/3 
  Ki=Ki_classic 
  Kd=Kd_classic*8/3

Values are KP=27.61 KI=2.08 KD=91.83 C=3

New values are 9.20, 208 and 91.83....But still 205 Degree model is going upto 208 degrees and ruining the model....Please the model....
[image]

Kaibob2, 2016-10-21:
To me this looks not really temperature dependent. If this is the first solid layer on top of the cube it will close with the second one. I'm printing OrbiTech PLA at 210°C and this sometimes happens, especially at low infill percent.

Kaibob2, 2016-10-21:
Btw. "No overshoot" value should be:
Kp = 27.61 / 3 = 9.20
Ki = 2.08 
Kd = (91.83 * 8) / 3 = 244.88

Shrikant888, 2016-10-21:
After putting these values also no change... its going upto 208 degrees with 
Kp = 27.61 / 3 = 9.20
Ki = 2.08 
Kd = (91.83 * 8) / 3 = 244.88

Somebody help me...........please

Shrikant888, 2016-10-21:
Also changed 

#define PID_FUNCTIONAL_RANGE 0

from 

#define PID_FUNCTIONAL_RANGE 30

Shrikant888, 2016-10-21:
Also how to store this C value..... PID-C...?
I am using C=3......

Kaibob2, 2016-10-21:
Hmm, this is getting a little ridiculous. Are you really serious??
There is no PID-C value. C=3 is only used for telling the PID autotune how often it should cycle around the given S value (205°) to determine exact value  for Kp,Ki and Kd.

Like isaid, if the temperature goes up to 208° it is no problem and absolutely normal on heating up!

Blue-Marlin, 2016-10-21:
Why the hell 0? 0 is switching off PID. Try something about 10 - then 12 - then 8. Observe the differences and find something good.

Shrikant888, 2016-10-21:
thanks to all....!!!!

If I get 208 then model is getting disturb... and what is this 

#define TEMP_HYSTERESIS 0 in CONFIGURATION.h file...

Can I make it 0? then I might be get 205....

Please help...

Shrikant888, 2016-10-21:
Please help me...My model is 205 degrees... I wanted to print that at same temperature....

Help help !!!!

Blue-Marlin, 2016-10-21:
TEMP_HYSTERESIS has nothing to do with PID - only with limit switching.
Limit switching is __NOT__ better then PID.

Begging for help more than once is counter-productive.

Kaibob2, 2016-10-21:
For real, I know no filament where 3 degree make a difference between complete failure and perfection. Just set cura to print the model at 202 degrees and it will be perfect.

Shrikant888, 2016-10-21:
@Kaibob2 
Now I am getting really PROBLEMATIC...

Model called in CURA a CUBE 20x20x10 at PLA 205 degrees, For printing its going upto 208 degrees and then coming to 204 degrees then 206 then 205 degrees...

I am really worried now...

Any one can solve problem?

Shrikant888, 2016-10-21:
Hello!!

Now I am getting really PROBLEMATIC...

Model called in CURA a CUBE 20x20x10 at PLA 205 degrees, For printing its going upto 208 degrees and then coming to 204 degrees then 206 then 205 degrees...

I am really worried now...

Any one can solve problem?

Kaibob2, 2016-10-21:
To be honest I suppose this thread is fake.

Shrikant888, 2016-10-21:
No sir, Its real. I dont know why told like that. I am using MARLIN V1.0 with sanguino board. Reprap LCD.

Kaibob2, 2016-10-21:
I'm sorry, i can't help you anymore. 
Maybe reading this
http://www.marlinfw.org/docs/development/configuration.html
and this
http://reprap.org/wiki/PID_Tuning
and the really well written comments in Configuration.h and Configuration_adv.h
will help you. At least it's also some kind of try and error involved. Be creative and check out what changing a parameter does. This is the way you learn and understanding the nature of Marlin and your printer.

Shrikant888, 2016-10-21:
I think you all are experts.. Try to solve my problem. I have attached a image in previous message...

Due to change in temperature MODEL is over heating and not coming. Simply telling FAKE thread is not a solution and also closing this issue. What you all told in previous messages I have tried and correspondly updated here. I am always eager to learn. So, asking here at forums by thinking that you all experinced in this field. Try to show us a light by sharing knowledge.

If I get good model then no reason to ask here.

Kaibob2, 2016-10-21:
I'm sorry for supposing fake, but it just doesn't seem to me that you try to think by yourself and just hoping someone here will tell you "Change parameter XYZ and everything will be fine" 
In some cases this will work, in your case it won't.
And by the way: You closed this issue. I'm not able to do so.

Shrikant888, 2016-10-21:
Yeah thats ok.

Alright I might ask lastly. I have AUTOTUNE PID from REPETIER HOST. But I am getting Kp, Ki and Kd values only at C=3. Not for C=1,2 and also tried to store into MEMORY by LCD manually. Again its resetting to C=1. EEPROM is not taking that.

Is it neccessary to keep C=5 or what sir?

Difference between  M303 E0 S205 C8 and  M303 E0 S205 C3...?

Kaibob2, 2016-10-21:
The difference between M303 E0 S205 C8 and M303 E0 S205 C3 is, that with C=8 it will heat up and cool down eighttimes and with C=3 it will heat up and cool down three times. This parameter is ONLY for figuring out the PID values. You cannot write C to the Eeprom. It is only a parameter for the M303 command. Like S205 sets the temperature, C8 sets the cycles.
You can also use M303 E0 S250 C10 or M303 E0 S200 C4
The more cycles, the more accurate will the be PID values.

Shrikant888, 2016-10-21:
I got it. 

In CURA bottom and top layers its 0.6...

Kaibob2, 2016-10-21:
0.6mm? At which layer height?

Shrikant888, 2016-10-21:
Please find the images
[image]

Shrikant888, 2016-10-21:
[image]

Shrikant888, 2016-10-21:
[image]

Shrikant888, 2016-10-21:
[image]

Shrikant888, 2016-10-21:
These are CURA settings for single extruder.....

Kaibob2, 2016-10-21:
I'm using Simplify3D and can't estimate these values, maybe someone else here can. The picture you've sent before doesn't show a cube with 6 top layers. Did you interrupt the print?

Kaibob2, 2016-10-21:
The travel speed appears to be quite slow. But this is something different.

Shrikant888, 2016-10-21:
No I didn't stopped it. I am really worrying only for that temperature changing. I will wait if some one or you can help...

Kaibob2, 2016-10-21:
The way the cube looks in your picture it is quite common when there is not enough infill the bridging over the holes of infill fail. BUT when this is only the case for the first layer to close the top. The second (of 6) or at leeast the third (of 6) should close the holes. 
As i said before, in my opinion this is not a 3° temperature issue but maybe a massive underextrusion, or printing too fast or the measured temperature is not equal to the real temperature -> nozzle too cold ->underextrusion or the Extruder stepper skips steps due to undercurrent...

Shrikant888, 2016-10-21:
Ok I will check it again. 

So, let me ask sir if you agree with this...

What all are the points releted to temperature like AUTOTUNING PID or CURA etc... Please tell if you feel free...

Shrikant888, 2016-10-21:
Also see this
[image]

Kaibob2, 2016-10-21:
The M303 command has nothing to do with CURA. With the PID values you only tell marlin how your hotend reacts on temperature changes and how "aggressive" the regulation has to be. The above images are not temperature related.
Like Blue-Marlin already told you in another thread:

> In this (Marlin) bug tracker you will find the Marlin experts. For questions about CURA ask there.

thinkyhead, 2016-10-22:
I see you've all met @Shrikant888 now.

thinkyhead, 2016-10-22:
This is a non-issue. **PLBKAC** Your prints will be fine as long as the temperature is within a reasonable range of the filament's melting temperature. Some overshoot is normal in heat-up.

closed 2016-10-22 by thinkyhead

emartinez167, 2016-10-25:
Shrikant,

Without trying to sound rude, I believe this forum is more for reporting bugs and software issues, rather than providing this kind of guidance, as there are better forums for that (and the team being so small as it is). Can I recommend you to try first reading the documentation available at www.reprap.org/wiki/G-code or making these questions in a more suitable place? Try the Google+ 3D Printing community or the RepRap forums. At least try reading the documentation before asking these questions here as the resources are limited.

Regards,

Ernesto

zanthor, 2016-10-25:
I still think the problem isn't a temperature issue - I see +/- 3 degree
swings when printing all the time without issues...

zanthor, 2016-10-25:
So I just want to understand the problem a little better.... a 3 degree
overshoot before it starts printing is ruining your TOP LAYER?

Kaibob2, 2016-10-25:
I think this is a massive lack of understanding how to operate the machine in detail. From my point of view Shrikant is a little lazy when it comes to reading instructions, googling function principles etc.
I don't want to sound disrespectful, but 6 months ago i was a complete newbie, too. In this 6 months i build the printer from zero and read (approx.) the whole repository and almost every line of code :)

> So I just want to understand the problem a little better.... a 3 degree overshoot before it starts printing is ruining your TOP LAYER?

If we take a close look at this résumé from zanthor i think it is clear what i mean. There is no connection between issue, observation and so on.