MarlinFirmware/Marlin #5052: M1 won't always continue when button pressed

Kaibob2, 2016-10-21:
I made myself a few Filament change routines for my dual extruder setup in ultralcd.cpp
The code is:
```
void filament_change_E0_PLA() {
lcd_return_to_status();
enqueue_and_echo_commands_P(PSTR("M109 S210 T0\nG4 P10000\nM300 P1000 S2500\nM1 Retract...\nT0\nG91\nG0 E-600.0000 F5000\nM300 P1000 S2500\nM1 Insert new Filament...\nM109 S210 T0\nG0 E520.0000 F3000\nG1 E70.0000 F240\nG90"));
  }
```

This works great and is intended to change filaments before printing, not like the filament exchange routine which is regularly implemented.
Like you can see the routine get's interrupted by M1.
Sometimes when i press the button on my Smart discount full graphic board it beeps and continues with the routine. This is how it should be.
Most of the time it beeps and the routine won't continue. When i press again it beeps the routine continues and the display switches away from the info screen to the settings screen.

This is not really a problem because it works anyway, but it would be nice to understand why this happens.

Roxy-3D, 2016-10-21:
> Most of the time it beeps and the routine won't continue. When i press again it beeps the routine continues and the display switches away from the info screen to the settings menu screen.  This is not really a problem because it works anyway, but it would be nice to understand why this happens.

I'm working to add a "Press and Hold" function to the Encoder Wheel right now to provide a fast way into the Z-BabyStepping.   I'm seeing some Wheel Clicks disappear and get lost.  I've added some debug code to find the issue but I still need to add some more debug code find the root cause of the problem.

What I'm chasing might be related to what you are seeing.

Kaibob2, 2016-10-21:
@Roxy-3D I've read about this press&hold thing. Greatly appreciated! Good to know i'm not the only one who experiences this.

thinkyhead, 2016-10-22:
Try this in `gcode_M0_M1` and see if it makes any difference:
```diff
      if (codenum > 0) {
        codenum += previous_cmd_ms;  // wait until this time for a click
        KEEPALIVE_STATE(PAUSED_FOR_USER);
        while (PENDING(millis(), codenum) && !lcd_clicked()) idle();
        lcd_ignore_click(false);
      }
      else if (lcd_detected()) {
        KEEPALIVE_STATE(PAUSED_FOR_USER);
        while (!lcd_clicked()) idle();
+       lcd_ignore_click(false);
      }
      else return;
```
…or maybe…
```diff
      if (codenum > 0) {
        codenum += previous_cmd_ms;  // wait until this time for a click
        KEEPALIVE_STATE(PAUSED_FOR_USER);
        while (PENDING(millis(), codenum) && !lcd_clicked()) idle();
        lcd_ignore_click(false);
+       while (lcd_clicked()) idle();
      }
      else if (lcd_detected()) {
        KEEPALIVE_STATE(PAUSED_FOR_USER);
        while (!lcd_clicked()) idle();
+       lcd_ignore_click(false);
+       while (lcd_clicked()) idle();
      }
      else return;
```

Kaibob2, 2016-10-24:
@thinkyhead I tried both attempts, but i'm struggling with the not reproduceabilty (is this a word?) of the issue itself. All three versions work, but they behave different...i think. So, i'm not sure which one is best. Sorry, this doesn't help at all

thinkyhead, 2016-10-24:
@Kaibob2 If you enable `EMERGENCY_PARSER` are you able to break out of `M1` reliably using `M108`?

thinkyhead, 2016-10-24:
When it fails, does it also make no clicky sound?

Kaibob2, 2016-10-24:
Puh, this is a tough one. What is "a clicky sound" for you :) 
I would say that it does beep and i feel the pressure point of the micro switch.
I'm an electrical engineer, so hardware failure can be excluded...i hope :)

thinkyhead, 2016-10-24:
> i feel the pressure point of the micro switch.
> hardware failure can be excluded

@Kaibob2 All Marlin does to read the button is to look at the raw pin state. It's not doing any fancy debouncing. So it's mysterious. If you have a spare endstop, try attaching it and switching the `BTN_ENC` to its pin. See if it still has the same issue.

Kaibob2, 2016-10-24:
@thinkyhead I will try to narrow this down some more and report back.
Btw. I'm not a coding genius, would you say this solution with `enqueue_and_echo_commands_P` is okay or is there a smarter way to do this?
My controller:
RepRapDiscount Full Graphic Smart Controller
[image]

thinkyhead, 2016-10-24:
> would you say this solution with `enqueue_and_echo_commands_P` is okay 

Seems fine to me!

Kaibob2, 2016-10-24:
> If you enable EMERGENCY_PARSER are you able to break out of M1 reliably using M108?

@thinkyhead Enabling EMERGENCY_PARSER and sending M108, M112 or M410 via Simplify3D does absolutely nothing when the printer is in "Waiting for User" state after sending M1.

When not in "Waiting for user" state sending M108, M112 or M410 via Simplify3D they work as intended.

Kaibob2, 2016-10-24:
While testing the above i had a missing click again, i can now say exactly what happens as i'm sensitized on it.

Marlin in M1 
Statusline says "Waiting for User"
I press the button and hear the mechanical click.
It beeps.
Nothing happens.
I press the button again and hear the mechanical click.
It beeps.
The display changes from Info screen to Main Menu
I press the button again and hear the mechanical click.
It beeps.
I'm back on Info Screen.
M1 is cancelled
Statusline says "Printer ready"

thinkyhead, 2016-10-26:
@Kaibob2 I've made a patched branch that _should_ work. Please test when you have a chance.

https://github.com/thinkyhead/Marlin/tree/rc_break_m1_fix

Kaibob2, 2016-10-26:
@thinkyhead I will test it tomorrow.

thinkyhead, 2016-10-26:
It is.

Kaibob2, 2016-10-27:
@thinkyhead This doesn't work as expected. It's even worse.

Marlin in M1
Statusline says "Waiting for User"
I press the button and hear the mechanical click.
Nothing happens.
I press the button again and hear the mechanical click.
Nothing happens.
I press the button again and hear the mechanical click.
Nothing happens. (and so on)
I HOLD THE BUTTON DOWN.
It beeps. (after some 100ms)
M1 is cancelled
Statusline says "Printer ready"

It seems that it depends on the moment when i hit the button. In some rare cases it reacts instantaneous. In most cases it takes some tie until it reacts. The only safe way is holding it down until it reacts.

Like in a giant loop which takes 2 seconds and only on one line it says "Hey Button Pin what's up"

thinkyhead, 2016-10-27:
@Kaibob2 I'll make more tweaks to that branch shortly. Join the thread at #5089 and we can continue there.

boelle, 2016-11-05:
@thinkyhead and status on this one?

Kaibob2, 2016-11-05:
I moved to #5089. Everything fine.

closed 2016-11-05 by Kaibob2