MarlinFirmware/Marlin #5067: Reset on heating while starting print

jaysonkelly, 2016-10-22:
I am getting the machine resetting when attempting to start a print. The heated bed will get to temp and then the screen will flash and the system will restart. Once I got it to reach bed preheat but it reset when approaching hot end temp. 

I can use the preheat in the menu to successfully start the print, but it would be nice if I didn't have to do that every time.

Blue-Marlin, 2016-10-22:
The basic questions:
What branch from about when?
What is the flashing screen telling you?
What does the serial protocol tell?

jaysonkelly, 2016-10-22:
Apologies for the scant details. I'm using RCbugfix I pulled yesterday into my fork. jaysonkelly/Marlin

By flashing I mean goes blank and then shows the splash screen. 

I'll check serial protocol in a moment

thinkyhead, 2016-10-22:
Just to eliminate one factor, try disabling `USE_WATCHDOG` during your testing.

jaysonkelly, 2016-10-29:
I think that it was partially tripping the overcurrent protection on my psu. My board has the reset on the atmega 2560 held high by the power good line on the 24v ATX style psu. Now when I preheat both bed and nozzle simultaneously it fully trips the over current protection and I have to turn off the PSU for a moment before it will turn back on. I'll test with USE_WATCHDOG disabled.

jaysonkelly, 2016-10-29:
Disabling USE_WATCHDOG had no effect on the issue. Is there a setting where I can ease into the heating instead of setting it to full heat all at once? This has not been an issue in the past. Testing with machine's stock firmware doesn't present the same issue.

Blue-Marlin, 2016-10-29:
PID_MAX and MAX_BED_POWER

thinkyhead, 2016-10-29:
When I first got into all this RepRap business Slic3r's starting code would first heat the bed, and only when bed heating was complete, heat the nozzle. Perhaps this was to ensure that not too much current was being drawn at once.

Roxy-3D, 2016-10-29:
> Perhaps this was to ensure that not too much current was being drawn at once.

So I don't know for sure...   But I always thought it was because you don't want to leave the nozzle hot and cooking the filament for too long.   I thought they were trying to prevent oooozing and the filament from crystallizing.

jaysonkelly, 2016-10-29:
It seems to have been an issue with my power supply. The backup I have doesn't present any issues. Thanks for your support!

closed 2016-10-29 by jaysonkelly