MarlinFirmware/Marlin #5069: Custom G-code problem

Mongrel-Shark, 2016-10-23:
Hi guys. This is my first post.
Been printing for a few months but so far not needed to post :D 

I was for some reason using old firmware and just upgraded to 1.0.2-1.
Its all working pretty well except for 2 issues. One seems to be very minor

The other causes the printer to stop at the end of the print with the hotend still on the print and still hot.
I have found its caused by a line of custom g-code I put in slicer to turn the fan up at the end of a print.

The custom g-code in slicer was 
M106 S200 : Fan 80%
M140 S0 ; Heatbed off
M104 S0 ; Hotend off 
G28 X0  ; home X axis
M84     ; disable motors

After looking at coms debug and comparing lines I found it was the fan 80% line it causes errors like this

SENT: N10658 G1 F8160*75
RECV: ok
SENT: N10659 G1 X80.487 Y80.977 E1.79408*104
RECV: ok
SENT: N10660 G1 E1.69408 F8400.00000*20
RECV: ok
SENT: N10661 G92 E0*71
RECV: ok
SENT: N10662 M107*38
RECV: ok
SENT: N10663 M106 S200 : Fan 80%*25
RECV: Error:No Checksum with line number, Last Line: 10662
[ERROR] Error:No Checksum with line number, Last Line: 10662

RECV: Resend: 10663
SENT: N10663 M106 S200 : Fan 80%*25
RECV: ok
SENT: N10664 M140 S0*96
RECV: Error:No Checksum with line number, Last Line: 10662
[ERROR] Error:No Checksum with line number, Last Line: 10662

RECV: Resend: 10663
SENT: N10663 M106 S200 : Fan 80%*25
RECV: ok
SENT: N10664 M140 S0*96
RECV: Error:No Checksum with line number, Last Line: 10662
[ERROR] Error:No Checksum with line number, Last Line: 10662

RECV: Resend: 10663
RECV: ok
SENT: N10663 M106 S200 : Fan 80%*25
RECV: Error:No Checksum with line number, Last Line: 10662
[ERROR] Error:No Checksum with line number, Last Line: 10662

RECV: Resend: 10663
RECV: ok
SENT: N10663 M106 S200 : Fan 80%*25
RECV: Error:No Checksum with line number, Last Line: 10662
[ERROR] Error:No Checksum with line number, Last Line: 10662

RECV: Resend: 10663
SENT: N10663 M106 S200 : Fan 80%*25
RECV: ok
SENT: N10664 M140 S0*96
RECV: Error:No Checksum with line number, Last Line: 10662
[ERROR] Error:No Checksum with line number, Last Line: 10662

RECV: Resend: 10663

So I moved the fan80% line to the end of the custom gcode and its doing everything now. Homes, turns heat off turns fan up to 80% But its clearly not running well. Coms debug keeps going on and on about the last line checksum.

Is there something wrong with my fan g-code or is it a bug? Wherever I put the fan line it causes a problem with the checksum on the previous line??? I need the fan to go up at end of print as I have an unusual setup there, where the one fan cools the heatbreak and blows some air on the print. If it goes of too soon at the end of the print I get a lot of blocked heatbreack issues.

My other issue dosent seem to be causing a problem but thought I would mention it. when I compile in Arduino 1.6.12 I get this error

sketch/temperature.cpp:565:47: warning: extra tokens at end of #ifdef directive
     #ifdef  THERMAL_RUNAWAY_PROTECTION_PERIOD & THERMAL_RUNAWAY_PROTECTION_PERIOD   > 0

It still compiles and runs etc, not sure what I did to create it. Tried turning thermal protection off by commenting out but it still comes up. I have not done anything in temp.cpp. I only changed settings in config.h
Its not a big concern for me but mentioning it in case its related to the first problem or indicative of something stupid I may have done. I'm hopeless with code...

ntoff, 2016-10-23:
>M106 S200 : Fan 80%

You have a colon rather than a semicolon (semicolon denotes a comment)

thinkyhead, 2016-10-23:
If you're getting a lot of checksum errors, try a slower `BAUDRATE` setting.

Mongrel-Shark, 2016-11-01:
Hey guys. Thanks for the help. 
Colin instead of semicolon. Facepalm. (I did mention I was useless at code)
That solved my problem.
Changing ifdef to if also helped with the compiling too.

I think I'll stick with the safe version of the firmware for now.
I just don't have the codeing skills to deal with any issues that may come up and need the printer for a business I am starting.

Thanks again for the help :D

closed 2016-11-01 by Mongrel-Shark

buinovkonstantin, 2017-06-02:
Hi, I'm not a Marlin Dev, however pure replacement of #ifdef to #if construction is not completely valid from the language point of view. It's better to replace "opening" #ifdef contruction to:
`#ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD
  #if THERMAL_RUNAWAY_PROTECTION_PERIOD > 0`
and then closing #endif to couple of #endif's:
`  #endif
#endif`
In case the macro is not defined it will turn `#if THERMAL_RUNAWAY_PROTECTION_PERIOD > 0` construction into `#if > 0`, which is not a nice thing.