MarlinFirmware/Marlin #5073: Probing offset bug

chinzw, 2016-10-23:
Hey guys, i believe this is a bug, but it would be good for someone with more knowledge of the codebase to confirm.

probe_entire_mesh( X_Pos+X_PROBE_OFFSET_FROM_EXTRUDER, Y_Pos+Y_PROBE_OFFSET_FROM_EXTRUDER, code_seen('O') || code_seen('M') );

I have my probe on the left side of the extruder, so my offset is -29, when this runs X_Pose+(-29) it goes outside my bed.

chinzw, 2016-10-23:
Also, i found the root of the problem here, as i realized, probe_entire_mesh just has the wrong lookup for the first point to probe, but the probing outside the bed problem comes from here:

#if HAS_PROBING_PROCEDURE
    // Boundaries for probing based on set limits
    #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define MAX_PROBE_X (min(X_MAX_POS, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define MIN_PROBE_Y (max(Y_MIN_POS, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
    #define MAX_PROBE_Y (min(Y_MAX_POS, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
  #endif

tofifee, 2016-10-23:
Should be an easy fix. With the use of the absolute value of the offset.

Blue-Marlin, 2016-10-23:
@tofifee 
You are confusing me. I'm quite sure about negative - not absolute.

tofifee, 2016-10-23:
| -2| == 2

-> [Compute the absolute number in Arduino](https://www.arduino.cc/en/Reference/Abs)

abs(<negative number>) == -1*<negative number>

Caveat: Should NOT be used with functions. For example abs(<negative number>++).

Here is how I think it would work.

`#define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + abs(X_PROBE_OFFSET_FROM_EXTRUDER)))`

BEWARE: I haven't tried that yet. Am at the moment trying to solve my RC7 woes myself ;)

chinzw, 2016-10-23:
It seems that at some point its using the correct position and doesn't go off the bed, but if i run G29 P1 twice in a row then i can get the probe to go outside the bed to -29

tofifee, 2016-10-23:
That is strange. Do you have EEPROM enabled?

tofifee, 2016-10-23:
Can you give replace

`X_PROBE_OFFSET_FROM_EXTRUDER`

with

`abs(X_PROBE_OFFSET_FROM_EXTRUDER)`

and see if that helps?

chinzw, 2016-10-23:
I have EEPROM enabled. It seems to be working now with the original code (which makes no sense to me). Ill do some more tests and report back.

tofifee, 2016-10-23:
It seems there is a lot of "evolved" code which interacts on one occasion and than not on another.

in conditionals.h there are definitions for probe points that make no sense for either positive or negative offsets in a universal way but seem to work intermittently. Or maybe they just get ignored depending on the circumstances.

I started a rewrite and it compiles but I will have to read through all the other code as well.

Glad to hear it works for you now!

thinkyhead, 2016-10-24:
@Roxy-3D Seems this is a UBL-related issue.

thinkyhead, 2016-10-24:
> Am at the moment trying to solve my RC7 woes myself

@tofifee `RCBugFix` is RC7 plus around 160 patches. You may have better luck with it.

tofifee, 2016-10-24:
Am on RCBugFix, will update my post accordingly

thinkyhead, 2016-10-24:
The use of `abs` is totally wrong for defining the probing boundaries. Note that the coordinates being considered here bound the probe's position, not the nozzle's position. So, taking the simple case where the offsets are -10, -10 with a typical 200x200 bed…
```cpp
    #define MIN_PROBE_X (max(0, 0 + -10))
    #define MAX_PROBE_X (min(200, 200 + -10))
    #define MIN_PROBE_Y (max(0, 0 + -10))
    #define MAX_PROBE_Y (min(200, 200 + -10))
```
Giving us:
```
MIN_PROBE_X 0
MAX_PROBE_X 190
MIN_PROBE_Y 0
MAX_PROBE_Y 190
```
Because the probe is left of the nozzle, it's no problem to probe X=0. But the probe can only reach as far as 190 on the right. Same principle applies to Y.

tofifee, 2016-10-24:
Probing at print bed x=0 or y = 0 leads to bed crash.

That's what I am trying to solve.

I will most likely hack something just for my use case starting from U.B.L..

Roxy-3D, 2016-10-24:
> It seems that at some point its using the correct position and doesn't go off the bed, but if i run G29 P1 twice in a row then i can get the probe to go outside the bed to -29

Do you still believe this is true?  Can you retry it?  I can't duplicate that failure.

> Probing at print bed x=0 or y = 0 leads to bed crash.
> That's what I am trying to solve.

Are you seeing the UBL code crash when you try probing at (X,Y) = (0,0) ???   Can you clarify exactly which firmware base you are seeing the crash on?

tofifee, 2016-10-24:
First quote is from OP chinzw .

I am not quite sure how to state it other than

if the probe is mounted left/front aka negative X/Y-value and you don't limit the X/Y-Axis during probing to a minimum postive value of X/Y-OFFSET_EXTRUDER, your print head will be forced on to the print head until you disable the motors because underneath the sensor is no metal to sense.

Btw, using a remote controlled power socket comes in quilte handy in limiting the damage.

This happens with RC7/RCBugFix/devel-ubl.

It is not an issue of the software crashing. It is an issue of probing starting at NOZZLE_X = 0 ignoring OFFSET_EXTRUDER alltogether and the sensor not being able to sense the print bed due the print bed not being under it.

adamfilip, 2016-10-24:
I did run into this aswell

I confirmed distance from Nozzle to my Probe  then i created a UBL mesh
and after the first layer gap between nozzle and bed was much larger then it should have been.. looks like over 1mm gap.. vs .25mm

tofifee, 2016-10-24:
@adamfilip I think you cross-posted ;) I seem to remember another issue with the Z_OFFSET being incorrectly applied or changing.

In essence what is needed is an absolute - excuse the pun ;) - safe area in which sensor/print head can move without causing/receiving damage.

This is my pseudo code for that:

``` 

void create_Safe_Nozzle_Positions_While_Probing_And_Not_Wasting_Print_Space(
  PRINTHEAD_CAN_MOVE_OUTSIDE_X_MIN: false/true,
  PRINTHEAD_CAN_MOVE_OUTSIDE_Y_MIN: false/true,
  PRINTHEAD_CAN_MOVE_OUTSIDE_X_MAX: false/true,
  PRINTHEAD_CAN_MOVE_OUTSIDE_Y_MAX: false/true,
  ) {

  #define X_PROBE abs(X_PROBE_OFFSET_FROM_EXTRUDER)
  #define Y_PROBE abs(Y_PROBE_OFFSET_FROM_EXTRUDER)

  #if (X_PROBE < 0)
    #define X_MIN_NOZZLE_WHILE_PROBING (
      X_MIN_NOZZLE_WHILE_PRINTING + abs(X_PROBE)
      )
    #if PRINTHEAD_CAN_MOVE_OUTSIDE_X_MAX
      #define X_MAX_NOZZLE_WHILE_PROBING (
      X_MAX_NOZZLE_WHILE_PRINTING + abs(X_PROBE)
      )
    #else
      #define X_MAX_NOZZLE_WHILE_PROBING (
        X_MAX_NOZZLE_WHILE_PRINTING
        )
    #endif
  #else
    #if PRINTHEAD_CAN_MOVE_OUTSIDE_X_MIN
      #define X_MIN_NOZZLE_WHILE_PROBING (
        X_MIN_NOZZLE_WHILE_PRINTING - abs(X_PROBE)
        )
    #else
      #define X_MIN_NOZZLE_WHILE_PROBING (
        X_MIN_NOZZLE_WHILE_PRINTING
        )
    #endif
  #endif

  #if (Y_PROBE < 0)
    #define Y_MIN_NOZZLE_WHILE_PROBING (
      Y_MIN_NOZZLE_WHILE_PRINTING + abs(Y_PROBE)
      )
  #else
    #if PRINTHEAD_CAN_MOVE_OUTSIDE_Y_MIN
      #define Y_MIN_NOZZLE_WHILE_PROBING (
        X_MIN_NOZZLE_WHILE_PRINTING - abs(Y_PROBE)
      )
    #else
      #define X_MIN_NOZZLE_WHILE_PROBING (
        Y_MIN_NOZZLE_WHILE_PRINTING
      )
    #endif
  ..
  #endif
...

probe_Print_Bed_WithZ_Probe_While_Avoiding_Print_Head_Crashing_Into_Print_Bed(
  X_START: X_MIN_NOZZLE_WHILE_PROBING,
  Y_START Y_MIN_NOZZLE_WHILE_PROBING,
  X_END X_MAX_NOZZLE_WHILE_PROBING,
  Y_END Y_MAX_NOZZLE_WHILE_PROBING,
  PROBING_STYLE: MESH/AUTO/BILINEAR/WHATHAVEYOU,
  DEBUG: FALSE/TRUE,
  ITERATIONS_TO_CALCULATE_ACCURACY: SOME_INTEGER) {

  calling U.B.L. using here

  }
```

That's it for today. Wish everyone a nice evening/morning/etc.

Roxy-3D, 2016-10-24:
> This happens with RC7/RCBugFix/devel-ubl.

This is puzzling.   UBL's bed probing is a total rewrite from the RC7/RCBugFix code base.   The Grid ZigZag code does not exist in UBL.   Instead, it tries to go to the closest unprobed location from the starting point.   And incidentally,  My Y_PROBE_OFFSET_FROM_EXTRUDER is negative, but my X_PROBE_OFFSET_FROM EXTRUDER is positive.   So, I don't have the same test scenario as you do. 

I just tried a G1 X0 Y0 followed by a G29 P1 and everything behaved correctly.  I also put the nozzle 1mm from the right side of my bed with Y=0  (199,0) and everything behaved as it should.

But...   I have a new printer with a BLTouch probe that is mounted such that both its X & Y PROBE_OFFSET_FROM_EXTRUDER are negative.   You can be assured any UBL issues with negative X & Y offsets are going to be observed and dealt with!   I'm still printing up debug panel stuff for the printer (control switches and debug LED's).   But in the next couple of days I'll be bringing up UBL on it and using it as the primary development machine for UBL.

tofifee, 2016-10-24:
Thank you for your reply Roxy-3D.

I tried U.B.L.

It homed to X/Y-EndStop and started with the Z-Axis-Probing which in turn would lead to bed crash due to sensor hanging in the air outside the print bed. 

My take is that some legacy code interferes only if certain conditions are met.

But for today I'm done. Will be at it tomorrow.

Roxy-3D, 2016-10-24:
> It homed to X/Y-EndStop and started with the Z-Axis-Probing which in turn would lead to bed crash due to sensor hanging in the air outside the print bed.

If you haven't tried this, it might be good to do the G28 and then move the nozzle to the center of the bed before doing the G29 P1.    And in fact, you can force the starting point by saying G29 P1 X100 Y100

>My take is that some legacy code interferes only if certain conditions are met.  But for today I'm done. Will be at it tomorrow.

I think you are correct about 'some legacy code' giving us unintended consequences.   While all of the previous G29 code has been taken out, it is still using the standard probing utilities.

tofifee, 2016-10-24:
> If you haven't tried this, it might be good to do the G28 and then move the nozzle to the center of the bed before doing the G29 P1

That is my issue. This should be done programmatically. 

But now I really call it a day ;)

Roxy-3D, 2016-10-24:
> That is my issue. This should be done programmatically.  But now I really call it a day ;)

Agreed.   But having a viable 'work around' is helpful for the users while the bug gets hunted down and found.   And it also helps the developers to find the bug if we know ways to avoid making it happen.

Have a good night!

adamfilip, 2016-10-25:
I would like to retract my comment.  I wasnt having a UBL issue. my steps/mm were wrong on my Z on my i3 printer and it lead to my problems.. wasnt an offset issue this time. :)

Roxy-3D, 2016-10-25:
OFF TOPIC:   
@adamfilip    How big is your print bed?   What are you using for the heat bed?   Can you include a few links of things you considered and what you finally decided to use?

adamfilip, 2016-10-25:
I use a 12" x 12" x 1/4" Alum Print bed, I had a friend cut some cheap alum plate on his plasma to a special shape. but i would buy this from Mcmaster Carr
http://www.mcmaster.com/#86825k11/=14r45my

For the heater i use a 120v 750W Silicone Heater from Keenovo, very good quality
http://www.ebay.ca/itm/271818680015?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

I use some cheap 3mm glass with a 1/16" PEI surface taped to it with 3M adhesive

I am eventually planning on upgrading my Corexy machine to 12"x24" I already have most of the pieces to do it..

thinkyhead, 2016-10-26:
> Probing at print bed x=0 or y = 0 leads to bed crash

So let me see if I understand this.

@Roxy-3D… In your UBL code, do you have it properly set up so that if someone asks to probe "0,0" it places the nozzle at "10,10" with **the probe** over "0,0"…?

Or is it the case, @tofifee, that on your machine the position 0,0 is off the bed? In other words, if you home to 0,0 and lower the nozzle, the nozzle tip is off the bed?

One of these things must be the case, if the probe is either not being moved to the correct position, or the UBL code isn't using the probe offsets.

tofifee, 2016-10-26:
sensor left and front to print head which is not that uncommon. And the print head being outside the print bed should only happen for postive offsets on the MIN side.

Problem is that the "safe" homing positions get calculated with a positive offset in mind.

I am still tinkering with the devel-ubl branch.

In essence what happens is:

The calculation which prevents sensors with positive offsets from leaving the print bed - if it's physically possible - is not doing the aquivalent part for negative offsets.

Someone took a short cut to avoid a bunch of if/else.

The easiest solution in my mind is to just put several definition blocks in there and comment them out and properly describe which to choose for the users setup.

thinkyhead, 2016-10-26:
In `RCBugFix` I believe I altered the code so that the `Z_SAFE_HOMING` positions refer to the probe position and not the nozzle position. Of course, it sounds like the `Z_SAFE_HOMING` feature is not related to your issue, which from what I understand is occurring in `G29`, right?

tofifee, 2016-10-26:
```
#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif
```

You just circumvented the whole issue and have the probe be at the middle of the print bed 👍

This is the most elegant approach imho as it works for negative and positive offset and is the place the print bed will most likely have a slight difference on the Z-Axis especially with heated beds.

thinkyhead, 2016-10-26:
> that is seems not to be used for G28 in contrast to its comment

`Z_SAFE_HOMING` surely does apply to `G28`. Are you sure it's enabled?

tofifee, 2016-10-26:
```
 #if HOMING_Z_WITH_PROBE
        destination[X_AXIS] -= X_PROBE_OFFSET_FROM_EXTRUDER;
        destination[Y_AXIS] -= Y_PROBE_OFFSET_FROM_EXTRUDER;
      #endif
```

Roxy-3D, 2016-12-01:
@thinkyhead 

> Roxy-3D… In your UBL code, do you have it properly set up so that if someone asks to probe "0,0" it places the nozzle at "10,10" with the probe over "0,0"…?

> Or is it the case, @tofifee, that on your machine the position 0,0 is off the bed? In other words, if you home to 0,0 and lower the nozzle, the nozzle tip is off the bed?

> It must be one of these things: Either the probe isn't being moved to the desired position, or the UBL code isn't using the probe offsets, or the position 0,0 is off the bed, so when the probe is moved there it can't detect it.

I think I'm doing everything right.    If the probe is going to be off the bed, it doesn't call the probe_pt() function.

```cpp
	location = find_closest_mesh_point_of_type( INVALID, X_Pos,  Y_Pos, 1, NULL);  
    	if (location.x_index>=0 && location.y_index>=0 ) {
		xProbe = blm.map_x_index_to_bed_location(location.x_index); 
		yProbe = blm.map_y_index_to_bed_location(location.y_index);
		if ( xProbe<MIN_PROBE_X || xProbe>MAX_PROBE_X || yProbe<MIN_PROBE_Y || yProbe>MAX_PROBE_Y)  {
    			SERIAL_PROTOCOLLNPGM("?Error: Attempt to probe off the bed.");
    			UBL_has_control_of_LCD_Panel = 0;
			goto LEAVE;
		}
		measured_z = probe_pt(xProbe, yProbe, ProbeStay, G29_Verbose_Level);
		z_values[location.x_index][location.y_index] = measured_z + Z_PROBE_OFFSET_FROM_EXTRUDER;
	}
```

thinkyhead, 2016-12-02:
@Roxy-3D Take a look at the `position_is_reachable` function in `RCBugFix`. You'll note that it does a little extra tweaking of the boundaries to account for the imprecision in floats…
```cpp
    return dx >= X_MIN_POS - 0.0001 && dx <= X_MAX_POS + 0.0001
        && dy >= Y_MIN_POS - 0.0001 && dy <= Y_MAX_POS + 0.0001
        && dz >= Z_MIN_POS - 0.0001 && dz <= Z_MAX_POS + 0.0001;
```

closed 2017-10-31 by thinkyhead