MarlinFirmware/Marlin #5075: Random MAXTEMP errors even with all thermistor using table #998

Fulg, 2016-10-23:
I am having a hard time figuring this one out. I started getting random MAXTEMP errors out of the blue soon after turning on the printer, but incredibly these occur even when using the testing thermistor table that always returns 25C.

Note that I started having this right after getting a legitimate MAXTEMP error while installing a new heated bed. While I was troubleshooting my thermistor and wiring, eventually the errors started. I enabled PID_DEBUG and PID_BED_DEBUG to try and spot the problem.

I cannot go back to a working state; I have deleted %TEMP% to force full rebuilds, cloned a new fresh RC7 and just added my config file, but no luck.

```
Connecting...
start
Printer is now online.
echo: External Reset
Marlin 1.1.0-RC7
echo: Last Updated: 2016-07-31 12:00 | Author: bmiller, Marlin RC7
Compiled: Oct 23 2016
echo: Free Memory: 4786  PlannerBufferBytes: 1232
echo:V24 stored settings retrieved (398 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E467.29
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X2000 Y2000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P2500.00 R2500.00 T2500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X15.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:  M304 P10.00 I0.02 D305.40
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.80
echo: PID_DEBUG 0: Input 25.00 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 724.79
echo: PID_DEBUG 1: Input 25.00 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 724.79
echo: PID_BED_DEBUG : Input 25.00 Output 0.00 pTerm -250.00 iTerm 0.00 dTerm 1941.68
Error:MAXTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: bed

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
```

The error is random, sometimes it occurs on heater #1:

```
[...]
echo: PID_DEBUG 0: Input 25.00 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 88.49
echo: PID_DEBUG 1: Input 25.00 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 88.49
echo: PID_BED_DEBUG : Input 25.00 Output 0.00 pTerm -250.00 iTerm 0.00 dTerm 237.05
Error:MAXTEMP triggered, system stopped! Heater_ID: 1
[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: 1
```

This behavior makes absolutely no sense to me. You can clearly see that the temperatures being read are indeed 25C, so how can MAXTEMP be triggered?!

[Configuration.h](https://github.com/MarlinFirmware/Marlin/files/546630/Configuration.zip)

Blue-Marlin, 2016-10-23:
Are your thermistors still connected?
Are the thermistor (ADC) pins shorted to ground or 5V?

The min/max-errors are detected from the raw values - before they are looked up in the thermistor-tables, to find the real temperature, or the dummy values.
So if the ADC is reading 16 consecutive zeros or 1023 the errors will be triggered.
The dummy thermistors are made for unconnected ADC-pins - noisy floating somewhere in the middle.

thinkyhead, 2016-10-24:
If you test with `RCBugFix` you have the advantage that it won't throw an error until you try to heat up the hot end or bed. So you can check the temperature readings with `M105` while you troubleshoot.

Fulg, 2016-10-24:
Unfortunately RCBugFix still throws on MAXTEMP (#4573 was for MINTEMP only). PR #5082 addresses this if you want it.

With this I was able to see that the errors go away as soon as I disconnect my bed thermistor, so this is definitely a hardware problem on my side. Thanks for the suggestions!

Fulg, 2016-10-24:
Confirmed, short on the thermistor legs on the bed, so the ADC was reading 0 all the time. Thank you @Blue-Marlin for the explanation!

closed 2016-10-24 by Fulg