MarlinFirmware/Marlin #5076: OVERSHOOTING NOZZLE AT 205 Degrees

Shrikant888, 2016-10-24:
Hi Everybody...

I have done PID TUNING for nozzle and trying to take print model at 205 degrees...

Tried a lot but temperature is going upto 208 which is OVERSHOOTING... Manually calculated OVERSHOOTING values...

Please help me 
Shall I change PID_MAX and BANG_MAX values from MARLIN...?

Shrikant888, 2016-10-24:
[image]

Shrikant888, 2016-10-24:
[image]

Shrikant888, 2016-10-24:
[image]

thinkyhead, 2016-10-24:
> Shall I change PID_MAX and BANG_MAX values from MARLIN

That sounds like a good plan. Try everything you can think of.

Shrikant888, 2016-10-24:
Please help me.. Since from a week I am trying. The difference is coming nearly 5-6 degrees... For 205 degree model, Its going upto 209 degrees...

thinkyhead, 2016-10-24:
Sorry, I don't know how to help you except to suggest you keep trying stuff and find someone close by who can come over and physically interact with your machine. After all, I'm just another RepRap enthusiast like yourself, and very new to all this electronics mumbo-jumbo.

Shrikant888, 2016-10-24:
I am done PID AUTOTUNING nearly 15-20 times for different different temperatures... All are coming as same...
What to do?

Kaibob2, 2016-10-24:
In Configuratin.h
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L289-L301
try changing
#define PID_MAX BANG_MAX to #define PID_MAX BANG_MAX 200
this will reduce the current to your heater when in PID regulation
and
#define PID_FUNCTIONAL_RANGE 10 to #define PID_FUNCTIONAL_RANGE 30
this will make the PID regulation kick in 30° before reaching target temperature

**And once again: OVERSHOOTING OF 5-6° ON HEATUP IS NO PROBLEM!!!** 
It is only important that the temperature stays in a few degrees range around target temperature **DURING PRINT**

thinkyhead, 2016-10-24:
I wish I knew enough to help you, but I just write the code. For the most part, the electronics side is all voodoo to me.

While we're waiting for the cavalry to arrive, open Repetier Host and turn on temperature monitoring, then do a heat-up while watching the plotted curve up until the point where it stabilizes. Once you have a good detailed picture of the heating curve, showing the overshoot, capture the screen and post the image here.

The way Repetier draws the graph, you can't see small fluctuations easily, so it might be hard to see them until it's been going for a few minutes. You can try having Repetier check the temperature more often.

Anyway, the next thing is to watch that graph. If it continually fluctuates up and down, then you may have a problem. But otherwise, you're doing fine.

thinkyhead, 2016-10-24:
> OVERSHOOTING OF 5-6° ON HEATUP IS NO PROBLEM

@KaiBob2 From the original post I'm under the impression that the temperature is staying 3 or more degrees too high throughout the print.

Is that not correct, @Shrikant888 ?

Kaibob2, 2016-10-24:
>the impression that the temperature is staying 5 or more degrees too high throughout the print
If this is the case i have to appologise for pointing only to 
> And once again: OVERSHOOTING OF 5-6° ON HEATUP IS NO PROBLEM!!!
But looking at the cube image from the original thread i would not think, that this is caused by a 3° difference.

thinkyhead, 2016-10-24:
> i would not think, that this is caused by a 3° difference.

I agree. In that case I think we were seeing an issue of poor extrusion or too few solid layers.

Shrikant888, 2016-10-24:
Ok done PID and got values nearly 1 degree difference.. that is Nozzle set for 205 but showing upto 206...

Now the problem is here...

Its dropping from 206 to 205 204 203 then coming to 205......

Kaibob2, 2016-10-24:
This is also normal! You cannot expect from the software regulation that it will EXACTLY climb to 205° and stay there perfectly. This is not possible, not with PID like implemented in Marlin. 
So, nevermind if it overshoots, and drops right after. If it will stabilize around the target temperature, 205° in your szenario, shortly after it touched the traget temperature once everything works as intended!!

Kaibob2, 2016-10-24:
Only for example:
[image]
If you take a look at the red plot it describes what you're experiencing in your setup. This image is taken from the [Wikipedia PID explanation](https://en.wikipedia.org/wiki/PID_controller) so you're experiencing nothing unusual or problematic temperature behaviour.

thinkyhead, 2016-10-24:
So there you have it. Situation normal, as already explained once before. Now, I'm going to close this issue. _@Shrikant888 **Don't repost this question.**_

closed 2016-10-24 by thinkyhead