MarlinFirmware/Marlin #5077: Release 1.1.0-RC8

thinkyhead, 2016-10-24:
### Changes since 1.1.1-RC7

#### New / Updated Features
- #4595, #4606, #4608, #4620, #4625, #4718 : Improved i2c messaging
- #4667 : Add M211 to Enable/Disable Software Endstops
- #4722 : Added MINIMUM_STEPPER_PULSE option
- #4832 : Enable M0/M1 with Emergency Parser
- #4833 Remove SCARA axis_scaling
- #4840 Add support for G2/G3 with R parameter
- #4900 Cleanup of G38.2 / G38.3
- #4933 Squashed - Removal of PID functional range
- #4955, #4974 PINS_DEBUGGING and M43: Read pin states
- #4983 Revert "Removal of PID functional range"

#### Code Cleanup & Documentation
- #4537 : Patch up some flaws here and there
- #4602 : Minor cleanup of SdFatUtil.h, MarlinSerial.cpp
- #4655 : Optimize, reduce size of MarlinSerial
- #4665 : Clean up ultralcd code, dependencies on ULTIPANEL
- #4691 : Suppress warnings
- #4693 : Cleanup of pins files, Conditionals
- #4703 : Add note suggesting RAMBo users use the RAMBo board type or platformio environment
- #4721 : Rc temperature isr comments
- #4749 : Fix a misspelling of minimum
- #4779 : Minor code cleanup, tweak M109/M190
- #4797 : Fix comment on FILAMENT_RUNOUT_SENSOR
- #4836 : Some comments, const args, debug output tweaks
- #4868 : Cleanups for the Autumn release
- #4893 : Define status LED pins without condition
- #4895 : Include requirements in "Implemented Codes"
- #4941 : Add note for M205
- #4993 : Sort out some header dependencies
- #5053 : Patch BAUDRATE line in configs
- #4650 : extruder_multiplier => flow_percentage

#### Planner & Stepper
- #4611 : Fix acceleration and feedrate following bq/Marlin
- #4720 : Patch stepper.cpp to allow omitting steppers
- #4738 : Optimize stepper ISRs, plus cleanup, shorthand
- #4754 : Unify counter_m and counter_M
- #4852 : Fix for advance extrusion algorithms
- #4936 : Put max_jerk vars in array, clean up jerk code
- #4963 : Move routine of direction signal of ADVANCE and LIN_ADVANCE from Stepper::isr() to Stepper::advance_isr()
- #4980 : LIN_ADVANCE bug fix and optimization
- #4984 : Small fix for the PR #4980 (LIN_ADVANCE bug fix and optimization)
- #5056 : LIN_ADVANCE optimizations
- #4982 : Fix planner with kinematics, delta ABL
- #4997 : Adapt Jerk / Speed code from Prusa MK2
- #5037 : Fix for the commit dc2cb84 (Revert acceleration limiting to avoid overflow)

#### Performance & Stability
- #4554 : Repair SPI-pins, some other pins
- #4564 : MISO is an input
- #4569 : Always disable SLOWDOWN for DELTA/SCARA
- #4572 : Show up to 99:59 in digital time
- #4573 : Throw MINTEMP BED error only if heating
- #4594 : Fix serial output for MBL, M303
- #4605 : Fixup 3DRAG pins based on updated RAMPS pins
- #4610 : Fix stopSDPrint so it works when paused
- #4621 : Init PWM-able AUTO_FAN pins with SET_OUTPUT
- #4688 : Repair MarlinSerial with TX-buffer
- #4748 : Patch duration_t toDigital method
- #4804 : Fix some regressive typos in G29
- #4820 : Fixing Pin Definition Files to Define USBCON
- #4834 : Standardize code_value_bool
- #4859 : Kinematic and SCARA patches
- #4883 : Follow-up the PR #4860 (Fix BLTOUCH string and SERVO_DELAY)
- #4887 : Improve MINIMUM_STEPPER_PULSE
- #4888 : Suppress warnings, fix nonlinear_z_offset
- #4889 : Fix HEATERS_PARALLEL
- #4914 : Pid unconstrained itemp
- #4954 : Fix buzzer when both i2c and beeper pin are set
- #4991 : Patches for Stepper DAC and MAX31855
- #5033 : Fix freezing and MINTEMP Error with MAX31855 thermocouple
- #5036 : Fix DAC current code

#### Configuration
- #4535 : Move tests for old symbols to the top of sanity check
- #4666 : dropsegments => MIN_SEGMENTS_FOR_MOVE
- #4669 : PREVENT_DANGEROUS_EXTRUDE => PREVENT_COLD_EXTRUSION
- #4673 : MIN_SEGMENTS_FOR_MOVE => MIN_STEPS_PER_SEGMENT
- #4727 : Add heading for movement settings
- #4765 : Sanitycheck endstop plugs based on homing direction
- #4805 : Additional documentation of Configuration.h
- #4821 : Z Raise/Clearance rather than Height
- #4842 : Drop DISABLE_Z_MIN_PROBE_ENDSTOP, clean up probe config
- #4919 : K8400 has 1xXY and 2xZ endstop plugs

#### Homing and Bed Leveling
- #4551 : Make Allen Key Probe stow
- #4562 : Fix Z_DUAL_ENDSTOPS with Z MIN Homing
- #4563 : Explain that Z_PROBE_*_HEIGHT options are nozzle-relative
- #4651 : Improvements to homing / leveling
- #4660 : Fix_Zig_Zag
- #4709 : Log whether homing with probe or endstop
- #4710 : Bed leveling that accounts for home XYZ
- #4719 : Marlin patches for homing, esp. Delta
- #4781 : Use simplified Z correction in G29 when possible
- #4789 : Big cleanup of leveling code
- #4835 : Allow non-square Auto Bed Leveling grid
- #4837 : Handle nonlinear bed-leveling in Planner
- #4839, #4856 : Better BLTOUCH support
- #4851 : Patch G29 for 3-point leveling
- #4875 : Use probe clearance for bump when homing Z with probe
- #4899 : ABL: Enable by type. Bilinear for all.
- #4918 : Add PROBE_Y_FIRST option. Arduino 1.6.8 required.
- #4927 : Don't check Z_MAX on raise when the probe pwns the pin
- #4958 : Improve M48 output; Add min, max, range, etc.

#### Mesh (Manual) Bed Leveling
- #4555 : Remove premature int-cast from MBL cell index methods
- #5008 : Save some PROGMEM in MBL G28
- #5057 : Fix manual leveling coordinates

#### LCD Controllers
- #4553 : Save bytes for custom chars (Hitachi LCD)
- #4570 : Make DELTA_CALIBRATION_MENU more universal
- #4571 : Patch command order in delta calibration menu
- #4584 : Use configured feedrates, center in DELTA_CALIBRATION_MENU
- #4574 : Include days in Graphical LCD print timer
- #4597 : Added hyphenated strings for full graphic display
- #4747 : Option to show SD percent on Graphical LCD
- #4803 : Clear LCD button state, apply timer to all
- #4967 : Revised DAC Drive Strength Menu
- #5059 : Set defer delay back to false when exiting babystepping.

#### Languages
- #4603 : Strip never-translated strings from language files
- #4634 : Set language display charset in language.h
- #4530, #4542, #4578, #4806, #4822, #5043 : Updated German language
- #4624 : Updated Spanish language
- #4642 : Updated Russian language
- #4649, #4723 : Updated Danish language
- #4685 : Updated French language
- #4815 : Updated Italian language
- #4885, #4943, #4981 : Updated Japanese language

#### For Developers
- #4196 : Add working PlatformIO support
- #4556 : Within Marlin, maintain most feed rates in mm/s
- #4652 : Add Rambo support/env to platformio.ini; properly map extended pins
- #4717 : Debug logging of nozzle type and offsets
- #4725 : Log machine info in G28 and G29
- #4756 : Makefile fix for Arduino 1.6.9 unzipped
- #4857 : Arduino 1.6.10, direct download U8glib in Travis CI
- #4934 : More detailed debugging of G28 delta
- #4972 : Move platformio directories out of source tree

jackshi618, 2016-10-28:
What time published

thinkyhead, 2016-10-28:
Not yet published.

boelle, 2016-11-20:
@thinkyhead what can we test at this point? and how to do it.... ?

i was thinking of a kind of script of things to put the printer through that tests as much a given setup can do

Blue-Marlin, 2016-11-20:
@boelle ? Testing? You are scaring me. 
From a useful tester i expect 
a minimum ability to decide if a problems root is in Marlin or not, 
a minimum ability to look up problems in the available sources, 
a minimum idea about how a feature should work,
the ability to search and read,
the ability to stand more or less on its own feet.

Please just try to use your printer. That will keep us busy enough with just answering your not Marlin specific questions.

Writing a test script seems to be completely beyond your abilities.

adamfilip, 2016-11-20:
Your response is mean and inappropriate.

Blue-Marlin, 2016-11-20:
That entirely depends on the question of how many of his issues you have read and answered.

adamfilip, 2016-11-20:
Sure, but everyone comes from different backgrounds with different skills. You assume alot.

thinkyhead, 2016-11-21:
@boelle I can almost imagine a GCode file with just about every command in it, and a configuration with every possible option enabled. But truly, it would be daunting to test everything in one go. Plus we have to test many kinds of printers.

In that vein, however, I was trying to recruit the PDX 3DP Lab —the user group in Portland, Oregon, USA— to help with that very thing. There are many kinds of printers, from the most standard to some very unusual. But it is like herding cats.

I think our best bet is to put out the release candidate and, as usual, we'll get a big surge in feedback. The next round will be all about fixing any remaining bugs. And, Crom willing, RC8 will be the last RC.

christianh17, 2016-11-22:
Hi!
I would love seeing a new RC. In the last weeks I often thought about moving to a new version, yesterday I found the new M43-option by chance - and would like to have it.

But: In the issues-list there are so many potential or confirmed bugs and I am not sure if it is a good idea to switch to a version with bugs as I do not see any relevant bugs in my setting. On the other hand I want to support the development and part of it is to use the actual version, test it and give feedback.

So: as soon as you developers think the version is stable enough I would install it asap.

Hopefully this post helps in a decision to do the next RC.

Kind regards
Christian

stigjoergensen, 2016-11-22:
Hi Christian

If i were you i would download the RCBugFix and use it - it may have bugs, but chances are you will never see them, or if you do, you can report them as well - and then disable the feature that are causing the bug - you are not worse off then with the RC7, on the contrary, it have a lot of benefits.. and even in worst case you can reinstall RC7.. just backup your folder with RC7 in... or even better download the RCBugFix into a separate folder and go from there, easy peachy... 

Trust me, its not that unstable - and i think most here actually run the RCBugFix (if not all - all is just a huge word ;)

rgds
/Stig

christianh17, 2016-11-22:
Hi Stig!

I already use RCbugfix from August, 8th and had some issues afterwards. Normally I think "do not change a running system". But I want to support the development and think it would be a sign from the developers if there would be a new RC and perhaps a feature freeze afterwards to get a real release. With this strategy there will perhaps a bigger community to install it and give feedback.

And: Of course I do have a good system for my personal releases so I can switch back very easily to every running version from the past. 

Regards
Christian

thinkyhead, 2016-11-23:
@christianh17 It's worth giving `RCBugFix` another try. A good number of things have been fixed since August. The main pitfall is the Z max speed and acceleration, which may need to be lowered for `RCBugFix` because some acceleration code was fixed. And of course, `M43` is brand new and deserves some testing.

christianh17, 2016-11-23:
Hi Thinkyhead!
I just did it today in the afternoon. My printer works, M43 helped me already a bit connecting my new full graphics display (I am afraid the connectors are turned around compared to my old fashioned display).

First tests looked good! There is a very (!) small thing with M115 - but I already opened a new issue. 

I will test further on and report as soon I have new ideas for improving!

Regards
Christian

landodragon141, 2016-11-26:
Scott how close do you think we are to releasing RC8, it seems like the line needs to be drawn somewhere or we'll just keep fixing ;-)

thinkyhead, 2016-11-27:
@landodragon141 Dual X Carriage issues with Duplication Mode need to be solved. Please contribute, if you can.

landodragon141, 2016-11-27:
What's the issue # for that one?

thinkyhead, 2016-11-27:
Related issues are #4694, #4695, #5162, #5175

landodragon141, 2016-11-27:
Great thanks! I'm not a phenomenal coder but I'm an excellent troubleshooter. I dig into that and see what I can find.

thinkyhead, 2016-11-28:
I believe the last few PRs in the queue should cover the most serious issues. If there's no other blockers, I will release 1.1.0-RC8 on Monday or Tuesday.

Grogyan, 2016-11-28:
/fingerscrossed that this is the last RC

thinkyhead, 2016-11-28:
I think so. We'll leave it out there for a couple of weeks, patch up any serious bugs, and release 1.1.0 in time for Kringlemas.

VanessaE, 2016-11-28:
Don't forget the linear-advance/lost E-steps problem. :-)

thinkyhead, 2016-11-28:
@VanessaE What's the issue number again?

VanessaE, 2016-11-28:
We kinda talked around it in #5092 and #5222, but neither of those was explicitly meant to address that glitch.  The short version is that `LIN_ADVANCE` causes loss of E steps on retract, even with K=0.  While turning up `MINIMUM_STEPPER_PULSE` can help, it interferes with serial timing above 4µS (on my bot anyway), causing checksum/line number errors.

Roxy-3D, 2016-11-28:
> I think so. We'll leave it out there for a couple of weeks, patch up any serious bugs, and release 1.1.0 just in time for Kringelmas.

Actually...  I think it would be helpful if we could get confident enough to commit to that.   The reason I say that is a lot of people have time off from work over the Christmas and New Year's holiday.   It would be nice timing for a lot of people to have the V1.1.0 available so they can spend the time required to bring it up.

landodragon141, 2016-11-28:
I see an added advantage of potential bugs being found from a large uptake.

Sebastianv650, 2016-11-28:
@VanessaE #5259 should fix your serial comunication issue.
Regarding `LIN_ADVANCE`, I will try to integrate the estepper ISR into the main stepper ISR. I hope that the higher resolution from the timer it's using will solve some problems.

boelle, 2016-12-02:
@adamfilip no worries... i have kind of an internal "pissed off gauge", @jbrazio made that gauge go in to the red a LOT..... @Blue-Marlin have gotten it to the yellow area... and then again i have grown up and are just ignoring certain names and avatars completely. ie i don't reed or respond, its like they are not even there.

and should all fail etc etc... bla bla.... Marlin is not the only firmware arround :-D

boelle, 2016-12-02:
@thinkyhead yes its a big task with a common list of G-codes and all the various configs

an old thought came to mind... 2 mega boards linked together... one is the "printer" with the config and options we want to test. the other is the "tester" that fires off commands and measures what the other do etc.

or maybe travis could do this?

just thoughts

jbrazio, 2016-12-02:
@boelle I'm such a peaceful person.. :-/

Doing unity testing would greatly improve Q&A for Marlin, doing it in the physical world would be possible but hard to automate to a certain degree IMO.

Building upon your suggestion and @daid's firmware simulator could be a good way to go forward into such direction, because we could automate it and feed in test scenarios because we control all the inputs into the simulator and can validate all the outputs.

boelle, 2016-12-02:
hehe

i guess my nordic blood does not mix well with your Portuguese.

but oh well, its nice to see that you are still among us mortals :-D

thinkyhead, 2016-12-03:
> and @daid's firmware simulator

I have gotten it to build in XCode, but I have not gotten it to actually do anything — I am presented with a blank window. I would love to be able to get it working.

> i guess my nordic blood does not mix well with your Portuguese.

My Finnish blood is supposed to have me drinking clear liquors and ice fishing.

jbrazio, 2016-12-03:
I some how was able to make it run with the included Marlin firmware, I couldn't make it run with the latest marlin version. There are build time dependencies on the Marlin source code, which means we build the simulator at the same time as we build the Marlin firmware.

In order to be useful we would need somehow to modify it to simulate the Arduino bootloader and boot .hex files.

thinkyhead, 2016-12-03:
Ah, that's right, too. I couldn't build it with XCode. I had to get and use CodeBlocks… I will have to try it some day soon with the included Marlin source. I probably just dropped in some recent version.

Roxy-3D, 2016-12-05:
@thinkyhead  Are we close to tagging RC-8 ?      

The reason I'm asking is I need to move the UBL code forward to the current code base.   I'm hitting a number of bugs that are now fixed RCBugFix and it is a lot of work to cross those bug fixes back to UBL.   So, instead, I was thinking I would spend a couple of days moving forward.  But I wanted to do that to something nice like 'RC-8'.   (I'm losing characters on the serial line and there are a number of LCD Panel bugs that are currently fixed that are causing problems.)

thinkyhead, 2016-12-05:
@Roxy-3D I'm working on debugging the Dual X auto-park issue right now. As soon as I have that resolved I will merge the code into `RC` and tag `1.1.0-RC8`.

Roxy-3D, 2016-12-05:
> I'm working on debugging the Dual X auto-park issue right now. As soon as I have that resolved I will merge the code into RC and tag 1.1.0-RC8.

Thanks for the info!   As soon as I see RC-8 get tagged...  I'll start moving the UBL code to it.

thinkyhead, 2016-12-05:
Just waiting for confirmation that Dual X is working better.

thinkyhead, 2016-12-06:
Ok, It's official! https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0-RC8

closed 2016-12-06 by thinkyhead

Roxy-3D, 2016-12-08:
And can we have a 'Stable Release' on December 24th ?      Then I'd have a 'Happy New Year' !

thinkyhead, 2016-12-08:
Test, report, and help fix, and it'll be done in no time.

h4nc, 2017-01-14:
I'm want to configure RC8 for my Diamond Hotend. It uses three extruders and I'm using Ramps 1.4, so I have to use a Stepper Expander Board. 

My problem is that in pins_Ramps.h there are new used pins, like:
#define Z_CS_PIN   40
#define E0_CS_PIN 42
etc. 

Can someone explain what CS does? 
I checked the circuit diagrams. For me it looks like all of those pins end up not connected to anything. Most of them end in AUX-2.
So what are those pins doing, when they are not connected?

Blue-Marlin, 2017-01-14:
```
/**
 * Enable this for SilentStepStick Trinamic TMC2130 SPI-configurable stepper drivers.
 *
 * To use TMC2130 drivers in SPI mode, you'll also need the TMC2130 Arduino library
 * (https://github.com/makertum/Trinamic_TMC2130).
 *
 * To use TMC2130 stepper drivers in SPI mode connect your SPI2130 pins to
 * the hardware SPI interface on your board and define the required CS pins
 * in your `pins_MYBOARD.h` file. (e.g., RAMPS 1.4 uses AUX3 pins `X_CS_PIN 53`, `Y_CS_PIN 49`, etc.).
 */
```
Probably you can ignore them.

h4nc, 2017-01-16:
Thx, that worked. I commented those out.