MarlinFirmware/Marlin #5079: Speed, acceleration, and kinematics

thinkyhead, 2016-10-24:
On DELTA the acceleration and speed of stepper movement is actually related to the carriages on the towers and not to the effector in cartesian space.

On SCARA the units of the XY axes are actually degrees, not mm, so speed and acceleration on these machines is related to degrees per second, not millimeters per second.

Since the point of speed and acceleration is to prevent loss of steps and keep the machine from jumping off the table, it makes sense to apply these limits to the moving parts. However, especially on SCARA, this neglects the rate and inertia of the end-effector itself. Plus, the distance traveled by a stepper motor has only indirect relation to the distance traveled by the effector.

Would it make sense to re-orient the acceleration code in some way around the cartesian coordinates on kinematic machines, and to account for the degrees-to-millimeters conversion on SCARA machines? Would it make sense in a SCARA setup to roughly determine the angle-to-millimeters distance in a given move and dynamically adjust the acceleration limits based on that Or, when configuring, do we just have to find acceleration rates that work, keeping within reasonable bounds?

Roxy-3D, 2016-10-24:
> Would it make sense to re-orient the acceleration code in some way around the cartesian coordinates on kinematic machines, and to account for the degrees-to-millimeters conversion on SCARA machines? 

Can we do this in Conditionals.h ???   That might be a clean way to hide the difference while keeping the code the more or less the same ???

thinkyhead, 2016-10-26:
Elements of any such re-orientation would be in many places, I'm sure. In general, it's probably too complicated a matter to shunt into the current code-base. We may look at this as part of the bottom-up rewrite that we'll be doing for 1.2.0.

boelle, 2019-10-24:
@thinkyhead is this one still relavant or has it served its purpose?

just dusting it off (2 years with no activity)

boelle, 2019-11-24:
another stale one

and we could as well use discord:

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-24 by boelle