MarlinFirmware/Marlin #5080: Cycles eaten by type casts in stepper and planner.cpp

Sebastianv650, 2016-10-24:
I'm searching for the next speed up possibilities. Therefore I had a look at variables used in planner.cpp and especialy inside the stepper ISR - if we can safe a little bit of time in a ISR called up to 10000x a second, a little bit may become "a little bit more".

On a first look, Marlin often stores variables in data types that is not equal to the ones it uses for calculations later on. For example accelerate_until and decelerate_after, which are stored as int32_t but always casted to uint32_t. Other ones are initial_rate and nominal_rate (only to name two), which are stored as uint32_t but nearly always casted to float.

Before searching and changing a lot of data types and learning afterwards that my work was bulshit during the PR, I want to ask what's your oppinion about that? I found [this Table](http://forum.arduino.cc/index.php?topic=200585.0) which lists some numbers. Assuming float(x) is the same as writing (float)x (is it?), especialy a cast from long to float and vice versa is quite time consuming. I found no numbers for casts between uint32 and int32, I guess it will not do much but no cast might be faster than a cast in any way.

So, continue to dig through the most-called variables or leave them alone?

Roxy-3D, 2016-10-24:
>  I found no numbers for casts between uint32 and int32, I guess it will not do much but no cast might be faster than a cast in any way.

I don't believe any conversion is happening.   The format of the number is going to be the same.  The difference is whether the high order bit is treated as a sign bit during subsequent calculations.     But the best thing would be to look at the compiled code and see if the number is just being used without the sign bit.    Still...   If the number is case to uint_32_t, we should just be declaring it that way and storing it that way.

thinkyhead, 2016-10-26:
There might be some cause for `accelerate_until` and `decelerate_after` to be signed, if they're used in signed math expressions, but if, as you say, they are always converted to `unsigned` then they can be made `uint32_t` instead.

thinkyhead, 2017-10-31:
Let's grab the fixed-point version of the planner/stepper ISR from one of the forks (Prusa MK2) which includes it. Obviously it will need to be adapted for the latest code, and obviously it cannot apply to all bed sizes and microsteps combinations, but it would be great to automatically use it for all machines where the point space is small enough.

landodragon141, 2018-03-18:
Correct me if I'm wrong but I believe this has been done and the issue could be closed.

closed 2018-03-19 by thinkyhead