MarlinFirmware/Marlin #5081: G29 seems to work incorrect

mgroene, 2016-10-24:
Hi, 
currently using Marlin RCbugFix from 2.Oct.16 on my printer. I always felt that the bed leveling not seems to work properly, since the distance of the nozzle is not really constant over the plate.
After reading the post from szafran81 in #4995 I tested with an unlevelled bed. He felt that the correction is made in the wrong direction, which is not true for me, but something other strange happens.
See video: http://sendvid.com/xra8c8e3

The bed was leveled, reported with
> Recv: G29 Auto Bed Leveling
> Recv: Bilinear Leveling Grid:
> Recv:       0     1     2
> Recv:   +2.40 +0.08 -2.66
> Recv:   +2.61 +0.25 -2.47
> Recv:   +2.58 +0.36 -2.27

If making 10mm moves, there is a huge jump in z. Demanding 100mm moves this is not happening. Of course it should never happen? Any idea what to test or what might be wrong?

Fulg, 2016-10-24:
Did you try a later version of RCBugFix? This was obviously broken on Oct 2. :)

Roxy-3D, 2016-10-25:
Your bed has 5mm of tilt form side to side.  Come on...   You have to give the ABL code a fighting chance.

JohnOCFII, 2016-10-25:
@mgroene Are you trying to replicate this heavily tilted bed test?  https://www.youtube.com/watch?v=x8eqSQNAyro

mgroene, 2016-10-25:
Oh no, I don't want to print on that, but cool to see that Marlin would make even that possible ;)
It's just a plate temporarily taped on my usual bed for testing and making the effect of automatic bed leveling better visible. The underlying bed is tilted by about +-0.1mm only, which makes it hard for me to see the effect of the automatic bed leveling. Maybe with my 5mm-tilt I shot a bit over.
What I actually wanted to show here is the sudden jump in Z, which should not occur (In the video at about 0:05s). Will try later with the currentmost RCbugFix commits and see what happens.

Fulg, 2016-10-25:
@Roxy-3D, is there some new limitation to the ABL code? I know this setup is extreme but if the probe is raised enough to clear the tilted bed during `G29`, if the bed is flat, why wouldn't it work? It certainly is a fun trick. :)

Roxy-3D, 2016-10-25:
> Maybe with my 5mm-tilt I shot a bit over.  What I actually wanted to show here is the sudden jump in Z, which should not occur (In the video at about 0:05s). Will try later with the currentmost RCbugFix commits and see what happens.

That sudden jump obviously should not be happening.   And may I ask a question:  Are you using the Grid Based ABL on RCBugFix?    Because that behaviour looks very similar to some of the bugs I had to work through on the Mesh foundation of UBL.   It looks like the nozzle is crossing into an area where the Mesh is poorly defined.

> is there some new limitation to the ABL code? I know this setup is extreme but if the probe is raised enough to clear the tilted bed during G29, if the bed is flat, why wouldn't it work? It certainly is a fun trick. :)

Well...   In theory, the tilt shouldn't matter.  But as the probe gets more and more offset, errors compound and it gets harder and harder to correct accurately.    If you want to play with an interesting situation in the UBL code, you can do this:   Do a G28   followed by a G29 P0 A  followed by a G29 Q 1  O   (There are 4 test patterns)   And then move the nozzle down to 1mm above the bed.   As you move across the diagonal of the bed, you will see the nozzle correct.

mgroene, 2016-10-25:
> That sudden jump obviously should not be happening. And may I ask a question: Are you using the Grid Based ABL on RCBugFix? Because that behaviour looks very similar to some of the bugs I had to [...]

But please see the different behaviour of not jumping on 100mm travels (in the video such a move is triggered at 0:13s). The position, where the jump occurs in 10mm travels is passed by that 100mm move, too but here is no jump to be seen. Can this be explained by a poor mesh, too? I'm using RCBugFix from 2.Oct. with bilinear 3x3 mesh leveling.
Update firmware this evening to the latest RCBugFix and try again, maybe with debugging activated?

Roxy-3D, 2016-10-25:
It would be good to turn on the Communication Debug in PronterFace and duplicate that big Z-Raise.   I don't have any facts to back up this hunch, but I've seen this kind of thing a number of times.   What I'm curious about is:   What is the final coordinate you are moving to when you see the big Z-Raise?    I'm thinking it might be off the bed (and therefor, outside of the mesh).   

If you cant get a record of what is being sent to Marlin via the Communication Debug in Pronterface, you can do it the hard way and send G1 X50 Y50   followed by G1 X100, Y100 followed by G1 X160 Y160 followed by G1 X170 Y160  followed by G1 X180 Y160  etc.   until you get to where it happens.

mgroene, 2016-10-25:
Just came home, the printer is still in the state of yesterday, so I tried to find the x-coord where it jumps in z-direction. Its a quick jump if moving between X179 and X180 forth and/or back. The z-coordinate displayed on the LCD remains the same while jumping. Traveling longer outgoing from X180 e.g. to X170, to X160 and so on, the movement in z-dir. slows down, but is occuring.
A border-case seems to be the travel from X180 to about X100. In this move the z-axis stands completely still. Outgoing from X179 to whatever I like, the software compensates the z-axis for the tilted bed correctly, it seems.

I'm using octoprint to control the printer.  I don't know whether there would be more information to debug communication if using pronterface or repetier host? The history in octoprint is as following:
> Send: G1 X180
> Recv: ok
> Send: G1 X179
> Recv: ok
These are absolute, but it also happens in relative positioning mode.

mgroene, 2016-10-25:
Strange things happen between X0 and X20, too. The bed compensation seems to be doing nothing in this area though no movement in z is recogniseable. Above X20 it lowers the axis, as it should. Maybe it has something to do with LEFT_PROBE_BED_POSITION and RIGHT_PROBE_BED_POSITION?

here some details from configuration.h

```
#define X_PROBE_OFFSET_FROM_EXTRUDER -35  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -11  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.35  // Z offset: -below +above  [the nozzle]

#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20

// Travel limits after homing (units are in mm)
#define X_MIN_POS -1
#define Y_MIN_POS -23
#define Z_MIN_POS 0
#define X_MAX_POS 215
#define Y_MAX_POS 205
#define Z_MAX_POS 140

#define AUTO_BED_LEVELING_BILINEAR

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
#define DEBUG_LEVELING_FEATURE

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define ABL_GRID_POINTS_X 3
  #define ABL_GRID_POINTS_Y ABL_GRID_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 20
  #define RIGHT_PROBE_BED_POSITION 180
  #define FRONT_PROBE_BED_POSITION 20
  #define BACK_PROBE_BED_POSITION 190

  // The Z probe minimum outer margin (to validate G29 parameters).
  #define MIN_PROBE_EDGE 20

  // Probe along the Y axis, advancing X after each column
  //#define PROBE_Y_FIRST

#elif ENABLED(AUTO_BED_LEVELING_3POINT)

  // 3 arbitrary points to probe.
  // A simple cross-product is used to estimate the plane of the bed.
  #define ABL_PROBE_PT_1_X 15
  #define ABL_PROBE_PT_1_Y 180
  #define ABL_PROBE_PT_2_X 15
  #define ABL_PROBE_PT_2_Y 20
  #define ABL_PROBE_PT_3_X 170
  #define ABL_PROBE_PT_3_Y 20

#endif

#define MANUAL_X_HOME_POS -5
#define MANUAL_Y_HOME_POS -27
#define MANUAL_Z_HOME_POS 0 // Distance between the nozzle to printbed after homing

```

mgroene, 2016-10-25:
good evening to everybody.
just updated to recent RCBugFix, the issue remains. Additionally the debugging information of G29:

> Send: M111 S32
> Recv: echo:DEBUG:LEVELING
> Recv: ok
> [...]
> Recv:   current_position=(215.00, 201.00, 9.12) : >>> gcode_G29
> Recv: Machine Type: Cartesian
> Recv: Probe: FIX_MOUNTED_PROBE
> Recv: Probe Offset X:-35 Y:-11 Z:-1.35 (Left-Front & Below Nozzle)
> Recv: Auto Bed Leveling: BILINEAR (enabled)
> Recv: ABL Adjustment Z0.23
> Recv: G29 Auto Bed Leveling
> Recv:   current_position=(215.00, 201.00, 9.35) : sync_plan_position
> Recv:   current_position=(215.00, 201.00, 9.35) : setup_for_endstop_or_probe_move
> Recv:   current_position=(215.00, 201.00, 9.35) : set_probe_deployed
> Recv: deploy: 1
> Recv: do_probe_raise(8.00)
> Recv: >>> do_blocking_move_to(215.00, 201.00, 9.35)
> Recv: <<< do_blocking_move_to
> Recv: >>> do_blocking_move_to(215.00, 201.00, 9.35)
> Recv: <<< do_blocking_move_to
> Recv: >>> probe_pt(20.00, 20.00, no stow)
> Recv:   current_position=(215.00, 201.00, 9.35) :
> Recv: do_probe_raise(5.00)
> Recv: >>> do_blocking_move_to(55.00, 31.00, 9.35)
> Recv: <<< do_blocking_move_to
> Recv:   current_position=(55.00, 31.00, 9.35) : set_probe_deployed
> Recv: deploy: 1
> Recv:   current_position=(55.00, 31.00, 9.35) : >>> run_z_probe
> Recv: >>> do_blocking_move_to(55.00, 31.00, 6.35)
> [...]
> Recv: <<< do_blocking_move_to
> Recv:   current_position=(55.00, 31.00, 6.35) : >>> do_probe_move
> Recv: >>> do_blocking_move_to(55.00, 31.00, -150.00)
> Recv: <<< do_blocking_move_to
> Recv:   current_position=(55.00, 31.00, 3.73) : sync_plan_position
> Recv:   current_position=(55.00, 31.00, 3.73) : <<< do_probe_move
> Recv:   current_position=(55.00, 31.00, 3.73) : <<< run_z_probe
> Recv: do_probe_raise(5.00)
> Recv: >>> do_blocking_move_to(55.00, 31.00, 6.35)
> Recv: <<< do_blocking_move_to
> Recv: <<< probe_pt
> Recv: >>> probe_pt(100.00, 20.00, no stow)
> Recv:   current_position=(55.00, 31.00, 6.35) :
> Recv: do_probe_raise(5.00)
> Recv: >>> do_blocking_move_to(135.00, 31.00, 6.35)
> [...]
> Recv: <<< do_blocking_move_to
> Recv:   current_position=(135.00, 31.00, 6.35) : set_probe_deployed
> Recv: deploy: 1
> Recv:   current_position=(135.00, 31.00, 6.35) : >>> run_z_probe
> Recv:   current_position=(135.00, 31.00, 6.35) : >>> do_probe_move
> Recv: >>> do_blocking_move_to(135.00, 31.00, -150.00)
> Recv: <<< do_blocking_move_to
> Recv:   current_position=(135.00, 31.00, 1.41) : sync_plan_position
> Recv:   current_position=(135.00, 31.00, 1.41) : <<< do_probe_move
> Recv:   current_position=(135.00, 31.00, 1.41) : <<< run_z_probe
> Recv: do_probe_raise(5.00)
> Recv: >>> do_blocking_move_to(135.00, 31.00, 6.35)
> [...]
> Recv: <<< do_blocking_move_to
> Recv: <<< probe_pt
> Recv: >>> probe_pt(180.00, 20.00, no stow)
> Recv:   current_position=(135.00, 31.00, 6.35) :
> Recv: do_probe_raise(5.00)
> Recv: >>> do_blocking_move_to(215.00, 31.00, 6.35)
> Recv: <<< do_blocking_move_to
> Recv:   current_position=(215.00, 31.00, 6.35) : set_probe_deployed
[... 142 lines cut ...]
> Recv: Extrapolate [ 1-  1-] (done)
> Recv: Extrapolate [ 1+  0+] (done)
> Recv: Extrapolate [ 1-  0+] (done)
> Recv: Extrapolate [ 1+  2-] (done)
> Recv: Extrapolate [ 1-  2-] (done)
> Recv: Extrapolate [ 0+  1+] (done)
> Recv: Extrapolate [ 2-  1+] (done)
> Recv: Extrapolate [ 0+  1-] (done)
> Recv: Extrapolate [ 2-  1-] (done)
> Recv: Extrapolate [ 0+  0+] (done)
> Recv: Extrapolate [ 2-  0+] (done)
> Recv: Extrapolate [ 0+  2-] (done)
> Recv: Extrapolate [ 2-  2-] (done)
> Recv: Bilinear Leveling Grid:
> Recv:       0     1     2
> Recv:  0 +2.38 +0.06 -2.66
> Recv:  1 +2.61 +0.23 -2.50
> Recv:  2 +2.56 +0.34 -2.35
> Recv: 
> Recv: G29 uncorrected Z:9.35
> Recv:  corrected Z:9.12
> Recv: <<< gcode_G29
> Recv: X:215.00 Y:201.00 Z:9.12 E:0.00 Count X: 17200 Y:16080 Z:3740
> Recv:   current_position=(215.00, 201.00, 9.12) : sync_plan_position
> Recv: ok

mgroene, 2016-10-25:
Just tested in y-direction, its the same here. Jump between Y189 <-> Y190. That Y-value is near the value for BACK_PROBE_BED_POSITION. Have I set that wrong?
>  // Set the boundaries for probing (where the probe can reach).
>   #define LEFT_PROBE_BED_POSITION 20
>   #define RIGHT_PROBE_BED_POSITION 180
>   #define FRONT_PROBE_BED_POSITION 20
>   #define BACK_PROBE_BED_POSITION 190

thinkyhead, 2016-10-26:
I recently added logic to handle points outside the probed area. I think what you're seeing is a case where the logic should be getting applied, but it's failing right near the edge, probably due to a `float` rounding error. I'll see about patching that.

Blue-Marlin, 2016-10-26:
It begins here:
```
          xProbe = floor(xBase + (xBase < 0 ? 0 : 0.5));
          yProbe = floor(yBase + (yBase < 0 ? 0 : 0.5));
```
with the interval resulting in 0 being only half as big as the others.

thinkyhead, 2016-10-26:
> Strange things happen between X0 and X20, too.

As the code is now, there will be no _additional_ compensation below X20. The Z offset is the same as the level that was measured at the closest edge.

thinkyhead, 2016-10-26:
#5090 prevents a bad grid index from being used at the upper bounds.

mgroene, 2016-10-26:
I made a pic just for my understanding... So no correction is made between X0...20. Is this true for X>180, too? This would explain that jump, doesn't it?
Just another question: Why is it necessary to define the boundaries for probing (x_PROBE_BED_POSITION)? Couldn't these be derived from travel-limits and the offset between nozzle to probe?
[image]

thinkyhead, 2016-10-26:
Nope. No **_ADDITIONAL_** compensation outside the bounds. In other words the tilt within the grid box is not extended outside the bounds.

Blue-Marlin, 2016-10-26:
Can't find the reason for 'xProbe' being different from 'xBase'. The calculation with floor() and adding an half, could make a bit of sense when the indices for the array would be affected, but here the coordinate, where to probe is shifted by 0.5mm. Even with a very good probe you probably can't measure a different z between two points so cloth apart.
The array indices are determined by the loop counters.

mgroene, 2016-10-26:
okay, just tried the latest RCbugFix. Now above X > 100 there is no compensation (z-axis stands still), below X=100 everything as before.

thinkyhead, 2016-10-26:
> Can't find the reason for 'xProbe' being different from 'xBase'.

I don't recall either. Probably the rounding was done in the previous code, and the newer code aimed to preserve it _just in case_.

thinkyhead, 2016-10-26:
> Now above X > 100 there is no compensation

How odd. There's debug code inside of `bilinear_z_offset` that you can uncomment to see what's going on. Try that and see if it reveals the problem.

closed 2016-10-29 by thinkyhead