MarlinFirmware/Marlin #5083: G28 makes BLTouch go in its error state, cannot home Z

Fulg, 2016-10-24:
Current RCBugFix, 3fcf91580872bf0c7a415a8ff539e5f0cf01a244. BLTouch is enabled, with `Z_SAFE_HOMING` and `PROBE_DOUBLE_TOUCH`. 

I start the printer, do a `G28`, X and Y home correctly, BLTouch deploys and Z is fine for the first probe. Then the bed retracts, the BLTouch deploys for the slow probe, but it does not trigger. Amazingly the bed stops soon after Z is negative and prevents crashing the head (I did not expect it to fail if the first probe worked!). Then the BLTouch becomes in its internal error state, and further homing (correctly) results in kill() being called since the probe is no longer working.

This did not happen a few days ago, hopefully this can be fixed before RC8!

[Configuration attached.](https://github.com/MarlinFirmware/Marlin/files/549101/Configuration.zip)

Fulg, 2016-10-24:
I am unable to reproduce this now? After a bisect, all revisions are good, and going back to 3fcf91580872bf0c7a415a8ff539e5f0cf01a244 works without issue.

I should have recorded video when I had the chance...

Roxy-3D, 2016-10-25:
> This did not happen a few days ago, hopefully this can be fixed before RC8!

I'm starting to be concerned about the quality and reliability of the BLTouch probes.   A repeatability of .047mm (for me) isn't very good.   But then all these reports of failures are very troubling.   

I'm not sure I expect anybody to do it, but it would be interesting to go back to the start of RC7 before the BLTouch easy setup options were added and just do the work to declare you have one servo (and set up the angles correctly for it to stow and deploy).     If I were force to bet on how this plays out, I think we will still see some BLTouch probes fail to deploy.    (And we do know the probe legs on a real servo do kick down reliably!)

Fulg, 2016-10-25:
I am only just remembering that I switched from AVR Tools 1.6.14 to 1.6.11 because of another occurrence of #5004. My current theory is that going back to 1.6.14 (assuming I can get it to compile) will cause this error to come back. If not, I'll close this until I can find a repro, if ever.

In this specific case I don't think the reliability of the probe is in question, as the problem was repeatable after multiple power cycles, and went away after flashing new firmware (no hardware changes).

On the bright side this means Marlin correctly detects BLTouch error conditions and will cancel a print if needed, this was always an issue before it was natively supported.

> A repeatability of .047mm (for me) isn't very good

For me `M48` gives a standard deviation of around .0069 with BLTouch, subsequent repeatability tests give similar results so I'm happy with it.

Fulg, 2016-10-25:
Important detail I forgot to add: at no point during this bug did the BLTouch fail to deploy. It looked as if the second probe simply failed to stop at the right Z after the pin was retracted due to contact with the bed, and the software endstop saved me from destroying my hotend.

First things first though, I need to reproduce this, I'll try again later today.

Fulg, 2016-10-25:
I cannot reproduce this again with AVR Tools 1.6.14 and the config I originally posted, so I don't know what happened. Closing until I see this again, if ever.

closed 2016-10-25 by Fulg

thinkyhead, 2016-10-26:
@Fulg As extra insurance for BLTouch probe reliability, we've found it helps to set the Z homing bump divisor to a smaller value, such as 3 or 2.
```cpp
#define HOMING_BUMP_DIVISOR {2, 2, 4}  // Re-Bump Speed Divisor (Divides the Homing Feedrate)
```