MarlinFirmware/Marlin #5084: Dual bed probes?

Nocturnal42, 2016-10-25:
I'm wondering if anyone has ever tried using dual probes for bed leveling? Bed leveling is the bane of my 3D printing experience. As I see it, the problem with using a probe for auto leveling is that the probe is offset from the nozzle, meaning it can never go everywhere the head goes (in my case, that's about 1/3 of the print area). It seems to me that the second simplest solution to this, manual probing being the simplest, would be to add a second probe diagonally opposite to the first probe.

You'd have to get the X and Y offset between the two probes pretty much exactly right, but the Z offset can be done simply by having one correct (offset from the head), and then dropping the two probes in the same spot on the bed to determine the difference in height between the two.

Roxy-3D, 2016-10-25:
Well...    A big probe offset does impact the quality of the bed leveling.   But if you bring up the devel_ubl branch, a probe isn't even necessary to use it.    You can just zero your Mesh and jump to the Mesh Validation Tool and see what areas of the bed need adjustment.   And you just move the nozzle to those areas and start editing the mesh up or down.

Nocturnal42, 2016-10-25:
The devel_ubl branch is on my list of things to try out. However, my ideal end setup is one that does not involve manual intervention in order to level the bed. If I have to manually adjust the mesh, I am still manually leveling the bed.

adamfilip, 2016-10-25:
Use the nozzle to trigger your probe switch :)
I have this setup and it's sweet.
You need to add a simple tilt mechanism to your hotend/extruder that triggers a switch when light pressure is applied.

You end up with no offset xy

I will send a link to my setup shortly

adamfilip, 2016-10-25:
Check this out
https://www.youtube.com/watch?v=9E_alNXIwnI

Nocturnal42, 2016-10-26:
That is very cool. Reminds me of the CNC touch probes. It's more work than just adding a second probe, and I don't think it will work well with the weight or bulk of my current direct drive extruder. It's definitely a better overall solution though.

I assume the spring on top is to load the tilt mechanism and keep it from shifting during normal operation. Did you consider trying a small servo with a cam to lock in into place when not leveling?

adamfilip, 2016-10-26:
havent felt the need. the spring has a decent amount of force, keeping it down. I am currently redesigning the head for use with a Direct drive, E3D Titan Extruder. I should have that running in a few days.

also the spring and tilt movement is  a good safety incase the head plunges into the bed. I have a physical stop on my printer to prevent it from diving more then 1/4" down.

thinkyhead, 2016-10-26:
> more work than just adding a second probe

But less work in software. ;-)

Nocturnal42, 2016-10-27:
True, but working on hardware on my printer on benefits me. Adding something to the software potentially benefits many.

thinkyhead, 2016-10-27:
> Adding something to the software potentially benefits many.

So true. Did I mention I have a Patreon? 😀

adamfilip, 2016-10-27:
Link?

esenapaj, 2016-10-27:
These?
[Scott is creating Firmware | Patreon](https://www.patreon.com/thinkyhead)
[Support Marlin Development by Scott Lahteine - GoFundMe](https://www.gofundme.com/zmh9h9vc)

Nocturnal42, 2016-10-27:
Now if only I could remember my patreon password... 

You might also want to consider adding some low level "reward" tiers like $1 $2 $5, most people don't go in for custom amounts, and $10 is a large ask at the lowest level. If you take a look at Gina Häußge's OctoPrint patreon, nearly half of that is from people giving under $5.

thinkyhead, 2016-10-27:
@Nocturnal42 The thing about Gina's Patreon is that she has received a lot of help and pubicity, both through Tom Sanladerer's YouTube channel, and through, of course, having an up-front product that is really, wholly her own. My angle is more of a call to the beneficiaries of the software rather than myself being sort-of like the publisher of the software. Given the disposition of Marlin and its aim to be as "invisible" as possible, I've been aiming more towards the vendors who use the firmware and profit from it more than the community. But perhaps I should consider that level more.

boelle, 2019-02-18:
@Nocturnal42 is this not more a question? if so:

Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/ 
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

Nocturnal42, 2019-02-19:
While phrased as a question, it is in fact a request for dual bed probe support.

Ergo, a feature request :-P