MarlinFirmware/Marlin #5085: FWRETRACT bug?

apballard, 2016-10-25:
Hi

I've been testing out FWRETRACT with G10/G11.

It works well, however I can reliably reproduce an issue and wondering if you can give insight?

I start a print with hop mm = 0. During a print I then increase it to 0.5mm. On clicking it in its evident that the setting has been applied as the z axis starts moving at the correct and anticipated times.

The problem I'm experiencing is then returning the hop back to 0mm.

When I click it in, it's as if the z axis stays 0.5mm down, therefor my extruded extrudes too high and starts making a mess.

I can repeat this every time.  Using rcbugfix from 21 October. Did not see any changes to fwretract.

Blue-Marlin, 2016-10-25:
Is there any possibility you are setting the new retract-parameters while retracted? Don't do that.

apballard, 2016-10-25:
Ok, it's definitely because of that. I would have to time it at just the right time to ensure it's not during a retract. Smaller parts with lots of retracts may need a bit of concentration. Lol

thinkyhead, 2016-10-26:
Best approach, if you need to make these changes, is to add it to the GCode file at the appropriate points.

apballard, 2016-10-26:
Perfect, thanks for feedback!

closed 2016-10-26 by apballard