MarlinFirmware/Marlin #5086: Help needed - Defining E1, E2 and E3 as Z axis

hello3dprintingworld, 2016-10-25:
Hi!

I would like to set my 3 extra extruders to be Z axis so I can have 4 motors dealing with the massive weight of a bed I am building. How can I do it?
Would be awesome if marlin had in configurations.h a stepper to axis set up. It would open the doors to more exotic setups!

Bob-the-Kuhn, 2016-10-26:
It looks like you'll need to do some custom programming to do this without using external stepper drivers.

You can get some idea of what's involved if you search for **Z2_** in the files **stepper.cpp** & **stepper_indirection.h**

The main issue with custom programming is you'd have to re-do the programming each time you moved to a newer version of firmware.

hello3dprintingworld, 2016-10-26:
Hi Bob-the-Kuhn, if its the case than i got to do so, but I'll problably only update it when it's a big release.
Do you have any tutorial you can point me?
Thanks for the feedback!

Bob-the-Kuhn, 2016-10-26:
I take it you're not a seasoned C++ programmer.

For the last 3 months I've been using [AstroGrep](http://astrogrep.sourceforge.net/) and [NotePad++](https://notepad-plus-plus.org/) to look around the code and find examples of what I want to do.  I still have a hard time remembering to put the semicolon on the end of statements.

I'm sure the professionals can point us to better tools.

hello3dprintingworld, 2016-10-26:
I use also notepad++ and I have really light programming knowledge, i've studied more python.
I'll wait and see if someone else did this that can give a walkthrow. Anyway thanks for the help!

thinkyhead, 2016-10-26:
@hello3dprintingworld Take a look at the `Z_DUAL_STEPPER_DRIVERS` feature – just global search for the term. If you wanted to do a `Z_QUAD_STEPPER_DRIVERS` feature just extend everything you find following precisely the same methodology.

dosono, 2016-12-10:
Great idea,
It could be great to have a z tripple stepper drivers as well.
I have a mid size bed based on three motors and one driver. Still calibrating with a paper... :(
It would be great to put an end to this manual calibrarion n 2017.

thinkyhead, 2016-12-11:
I've implemented this kind of setup using two RAMPS boards. The second RAMPS drives the 4 Z steppers, handles the Z endstops, and manages its own Z corner leveling. It's controlled from the main board over I2C.

dosono, 2016-12-11:
Very very interesting ! I would guess you would simply go for extended extruder boards but a slave boards is even better. I can imagine now many more extruders. Prusa went on multiplexer for four extruders but this is much bett approach.
Is this code available for testing? Do you think it will work with three motors as well?

thinkyhead, 2016-12-13:
@dosono Take a look at this overview that I wrote on how to set up an I2C slave on a second RAMPS board: https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879

That thread has all the essential details on how master-slave communication works.

For a specialized slave instance of Marlin, it is possible to go through the source and strip out all the unused code, leaving just the features you need. For example, in my Z axis slave implementation I just removed all the source files pertaining to SD, LCDs, languages, pins, etc., then I removed every unnecessary GCode handler from `Marlin_main.cpp`. Then, for every disabled feature, I just collapsed its `ENABLED()` blocks in the editor and deleted them.

That process tends to freeze the slave version in time, making it hard to bring in fixes from the newest codebase. It's a laborious process to strip down Marlin to only the parts needed. But it is satisfying to see how little code is really needed in the end.

Of course, it's important to keep the serial terminal and GCode processor intact in any slave, so you can control it independently, test it, and so on. But once you have it fully working, you can even pull out those parts.

When it comes to the kinds of things a slave is good for, I2C is fast enough for most purposes. Consider a slave board that only knows how to handle the kinematics, the movement commands (`G1`, `G4`, `G28`, `G29`, etc.), and the endstops. The stepper drivers can move to the slave board (the motion controller), while the main board manages the LCD, SD card, I/O, sensors, heaters, etc. The main board talks to the host and the SD card. It does many of the commands, but all the movement commands go to the slave.

Once you have a well-defined messaging protocol, you can divide up the firmware in any way that works for your needs. The most compact protocol is the best one, as messages are "officially" limited to 32 bytes apiece, and the slave has to handle commands and return quickly.

A devoted motion controller returns a lot of resources to the main board. Since I2C slave communication is handled in an interrupt, as long as the stepper ISR leaves a tiny slice of time, it will keep receiving commands.

With two boards, you can take advantage of the extra power of the main board to do delta calculations or any other useful work, since the stepper ISR will be completely absent. The display can be assured of having more time to refresh without interruption. Presumably the planner buffer will be refreshed at a faster rate, and higher baud rates should be more reliable.

Overall, a win-win, especially with boards being so pricey and commodified. Being able to just hook up a second RAMPS (sans stepper drivers) and suddenly boost your printer's performance is pretty cool.

In the future we should do the same treatment for SPI, which is both faster and bi-directional, but only works over short distances. It could be useful for making smarter hardware modules in the future.

dosono, 2016-12-13:
Just cool ??? :) Check the prices of the 7 or worse the 8 drivers boards...
Josef Prusa went far and created a special 4 extruder multiplexer board due to that limitation (my guess).
This feature opens up a LOT of new possibilities.... mmm wow.

Anyway, how about the 4 or in my case 3 steppers for the bed ? how do you manage that, not by killing more source code lines I guess ;)
Any example somewhere ?

dosono, 2016-12-28:
Hi, is it working for anyone?

thinkyhead, 2017-02-26:
> Is this code available for testing? Do you think it will work with three motors as well?

https://github.com/thinkyhead/Marlin/tree/a3d_main_controller
https://github.com/thinkyhead/Marlin/tree/a3d_zaxis_controller

closed 2017-10-31 by thinkyhead