MarlinFirmware/Marlin #5096: X not homing

brylandry, 2016-10-27:
So i was told to get to bug fix. My problem is i installed a BLTouch everything was working fine excepr i printed a 10mmx10mmx10mm cube x and y were ok z was 15.81 so thinkyhead told me to download the bugfix my marlin is 1.0.2 i have a hictop 3DP08 usins simplify3d for slicer. not sure why x is not homing. oh and using arduino 1.6.12 
[Marlin-RCBugFix.zip](https://github.com/MarlinFirmware/Marlin/files/554735/Marlin-RCBugFix.zip)
[marlin bltouch.zip](https://github.com/MarlinFirmware/Marlin/files/554751/marlin.bltouch.zip)

thinkyhead, 2016-10-27:
Just to clarify: X is failing to home with `RCBugFix`?

Roxy-3D, 2016-10-27:
> z was 15.81 

That is easy to fix.    Take what ever number is in your Z number (the third one) and
```cpp
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,52.2}   
```
multiply it by 10/15.81.    Rebuild the firmware.  Do a M502 and M500.   And print a new calibration cube.

thinkyhead, 2016-10-27:
Somehow I missed that this was about calibrating Z steps. This may also help http://prusaprinters.org/calculator/#stepspermmlead

brylandry, 2016-10-27:
It's not the z steps I changed it from 640 to 750 just to see what would happen it still printed 15.81 no difference that's my old firmware.  Now the bug fix firmware I just changed this firmware to match my machine and I can't get x to home it just sits there and buzzes for a few seconds then homes y and z without homing x

thinkyhead, 2016-10-27:
> I changed it from 640 to 750 just to see what would happen it still printed 15.81 no difference

Did you reset the EEPROM too? `M502` followed by `M500` to make sure it's not using the old stored value.

thinkyhead, 2016-10-27:
Please post your modified configuration files.
The items you posted above don't seem complete.

brylandry, 2016-10-27:
So i did what Roxy-3D said which should have changed z height but it's exactly the same 640 is the number i calculated from the prusa calculator it worked be for i installed the bltouch now granted Iv'e changed to different firmware but it did work when i printed out my bltouch mount couldn't get the bltouch  to work with the one i used to make the mount been messing with different firmware ever since 
[hictop bltouch.zip](https://github.com/MarlinFirmware/Marlin/files/557183/hictop.bltouch.zip)

thinkyhead, 2016-10-28:
That's one old-ass version of Marlin.

Netoperz, 2016-10-28:
Hello,

Same issue i think.

manually moving to a position sets that position as  0.
When i was moving manually using Pronterface the axis (all 3, XYZ) were moving in + direction but not reacting on - direction.

So if i have moved the X +20 it has moved and when i wanted to go back to -20 it was staying in +20, just i've heard small "thick" like a micro step  in stepper, but not seen any actual move.

Strange thing was that when i was checking endstops

SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED
echo:endstops hit:  X:-0.00
echo:endstops hit:  X:-0.01
echo:endstops hit:  X:-0.01
echo:endstops hit:  X:-0.02
echo:endstops hit:  X:-0.02
echo:endstops hit:  X:-0.03
echo:endstops hit:  X:-0.03
echo:endstops hit:  X:-0.04
echo:endstops hit:  X:-0.04
echo:endstops hit:  X:-0.05
echo:endstops hit:  X:-0.05
echo:endstops hit:  X:2.94
echo:endstops hit:  X:2.94
echo:endstops hit:  X:2.93

So after every move i've had endstop hit, even if min software endstops in config was not uncommented, and commented.

No change.

than i have reset the settings with M502 and M500.

SENDING:M503
echo:Steps per unit:
echo:  M92 X200.00 Y200.00 Z8000.00 E500.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0

Now i can move the all axis back and forward  + - , but they are ignoring the minimal end stops on my ramps, 

ENdstops are fine, connections are fine too.

This is one freaky kind of marlin i have.

and configuration as follows:

`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
[... 1299 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H`

Netoperz, 2016-10-28:
Update:

Right now just after refreshing the same firmware with the same config i cannot take - (minus) direction using menu entry move X when choosing any amount, 1mm or 10mm.

but + (plus) direction works fine.

On LCD  I see HOME X, (HOME is my label in translation for X min endstop triggered, but print head is in max position where i do not have any endstop, i just can't understand that situation.

when using menu entry Home Z the steers do not work correctly, just like wrong frequency is pended to them.
Y when using Home Y is going in wrong direction (+ not - as it should) but it moves only a few millimeters and stops.

when triggering Home all in LCD menu Z is going up not down, so it is going in a different way than just triggering only Z home.

x is homing to max

and y (build plate) as before in MAX (+) direction only a few millimeters and stops, 

and Z is going also to max and stops after a 2 sec or so.

Toptal mess....

when using Pronterface to manually move x to MIN, that is what i get
echo:endstops hit:  X:-0.00
echo:endstops hit:  X:-0.01
echo:endstops hit:  X:-0.01
echo:endstops hit:  X:-0.02
echo:endstops hit:  X:-0.02
echo:endstops hit:  X:-0.03
echo:endstops hit:  X:-0.03
echo:endstops hit:  X:-0.04
echo:endstops hit:  X:-0.04
echo:endstops hit:  X:-0.05
echo:endstops hit:  X:-0.05
echo:endstops hit:  X:-0.06
echo:endstops hit:  X:-0.07
echo:endstops hit:  X:0.93
echo:endstops hit:  X:0.92
echo:endstops hit:  X:0.92
echo:endstops hit:  X:0.91
echo:endstops hit:  X:0.91
echo:endstops hit:  X:0.90

after every try the home is set to the amount i have tried to move but the stepper holds in place and is not moving.

Netoperz, 2016-10-28:
I've tested the bug fix branch.

no changes in config, just enabling the controller and sd card, and changing max positions from 200,200,200, to 190,190,170 (xyz)
when i use in menu of my full graphic controller option HOME X it is trying to home Z and X , when i try home x it is homing only x, trees, but same problem like in RC.

also there is problem with Z steppers they sound strange like wrong frequency and not moving well and stopping with no homing.

When i do try to move the steppers using Pronterface, they work fine but only in + direction.

in negative direction (MIN) they do not move at all.

Netoperz, 2016-10-28:
Solved.

#define min_software_endstops false // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops false  // If true, axis won't move to coordinates greater than the defined lengths below.

I suggest setting those to false by default.

I do not know a person who needs it enabled when building a printer.

That 2 lines have ruined my 3 days...

brylandry, 2016-10-28:
My old ass firmware. i'd love to thar deugg or rc working

SENT: M503
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X80.00 Y80.00 Z405.06 E94.50
M92 X80.00 Y80.00 Z405.06 E94.50
READ: echo:Maximum feedrates (mm/s):
Maximum feedrates (mm/s):
READ: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
M203 X500.00 Y500.00 Z5.00 E25.00
READ: echo:Maximum Acceleration (mm/s2):
Maximum Acceleration (mm/s2):
READ: echo:  M201 X3000 Y3000 Z100 E10000
M201 X3000 Y3000 Z100 E10000
READ: echo:Acceleration: S=acceleration, T=retract acceleration
Acceleration: S=acceleration, T=retract acceleration
READ: echo:  M204 S3000.00 T3000.00
M204 S3000.00 T3000.00
READ: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
READ: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
READ: echo:Home offset (mm):
Home offset (mm):
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:PID settings:
PID settings:
READ: echo:   M301 P49.10 I6.32 D95.33
M301 P49.10 I6.32 D95.33
READ: ok

brylandry, 2016-10-28:
ok 10X10X10 cube printed great it seem that M502 and M500 is what i wasn't doing when i flashhed firmware it's still that old ass firmware though but it's working

Thanks Guy's 

Bryan

thinkyhead, 2016-10-29:
> I suggest setting those to false by default.

@Netoperz Actually, we got rid of them.

closed 2017-10-31 by thinkyhead