MarlinFirmware/Marlin #5101: ubl-devel won't compile

Nocturnal42, 2016-10-28:
It could just be that I screwed up somewhere when switching branches, but it throws a ton of warnings during compile, and complains that chirp_at_user() is non existent.  Since it's only called in two places, I commented it out for now.

I am filled with sadness as I imagine all the chirps I am missing out on.

JohnnyTheOne, 2016-10-28:
You're not alone

Roxy-3D, 2016-10-28:
I tried to update a file that I thought had no other dependencies.   The extra function is in the ultralcd.cpp file.   It compiles clean now.

SORRY ABOUT THAT!!!

Nocturnal42, 2016-10-28:
Still get all the warnings, but it compiles now.

Roxy-3D, 2016-10-28:
Warnings???    Can you post them please?   I have more changes on the file set I'm working with (that are not ready to be uploaded).   But it compiles clean here.    I must have missed something.

Nocturnal42, 2016-10-28:
Enjoy. Are you regretting encouraging me to try the ubl branch yet :)

<details><summary>Warnings</summary><p>
```
In file included from sketch\Bed_Leveling.cpp:23:0:

sketch\Marlin.h:349:0: warning: "RAW_CURRENT_POSITION" redefined

 #define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)

 ^

In file included from sketch\Bed_Leveling.cpp:23:0:

sketch\Marlin.h:140:0: note: this is the location of the previous definition

 #define RAW_CURRENT_POSITION(AXIS) (RAW_POSITION(current_position[AXIS], AXIS))

 ^

In file included from sketch\G25.cpp:6:0:

sketch\Marlin.h:349:0: warning: "RAW_CURRENT_POSITION" redefined

 #define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)

 ^

In file included from sketch\G25.cpp:6:0:

sketch\Marlin.h:140:0: note: this is the location of the previous definition

 #define RAW_CURRENT_POSITION(AXIS) (RAW_POSITION(current_position[AXIS], AXIS))

 ^

In file included from sketch\G26_Mesh_Validation_Tool.cpp:49:0:

sketch\Marlin.h:349:0: warning: "RAW_CURRENT_POSITION" redefined

 #define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)

 ^

In file included from sketch\G26_Mesh_Validation_Tool.cpp:49:0:

sketch\Marlin.h:140:0: note: this is the location of the previous definition

 #define RAW_CURRENT_POSITION(AXIS) (RAW_POSITION(current_position[AXIS], AXIS))

 ^

sketch\G26_Mesh_Validation_Tool.cpp: In function 'void gcode_G26()':

sketch\G26_Mesh_Validation_Tool.cpp:236:72: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   debug_current_and_destination("Starting G26 Mesh Validation Pattern.");

                                                                        ^

sketch\G26_Mesh_Validation_Tool.cpp:354:6: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 cptr = "01234567890ABCDEF????????";

      ^

sketch\G26_Mesh_Validation_Tool.cpp:369:69: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

        debug_current_and_destination("Looking for lines to connect.");

[... 724 lines cut ...]
In file included from sketch\vector_3.cpp:42:0:

sketch\Marlin.h:140:0: note: this is the location of the previous definition

 #define RAW_CURRENT_POSITION(AXIS) (RAW_POSITION(current_position[AXIS], AXIS))

 ^

In file included from sketch\watchdog.cpp:23:0:

sketch\Marlin.h:349:0: warning: "RAW_CURRENT_POSITION" redefined

 #define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)

 ^

In file included from sketch\watchdog.cpp:23:0:

sketch\Marlin.h:140:0: note: this is the location of the previous definition

 #define RAW_CURRENT_POSITION(AXIS) (RAW_POSITION(current_position[AXIS], AXIS))

 ^
```
</p></details>

Roxy-3D, 2016-10-28:
I'm using Arduino 1.6.9 and I'm not getting those warnings.    What version of Arduino are you using?

Nocturnal42, 2016-10-28:
I'm using 1.6.11.

Roxy-3D, 2016-10-28:
I forget what the reason was...   I could have this wrong...   But I think Arduino versions past 1.6.9 expect C++11    and Marlin is definitely not at that level.    I believe (but don't know for sure) you can just ignore the warnings if you really want to use 1.6.11.    But my recommendation is to back peddle to 1.6.9 until Marlin is compliant with C++11.

Nocturnal42, 2016-10-28:
:) I am familiar with ignoring warnings. Your reasoning is wrong however, out of all the branches I have tried (including 1.0.2-1), only yours is generating warnings.

Roxy-3D, 2016-10-28:
Well...   I'm not sure what to tell you.   I have that reason stuck in my head for why I'm using 1.6.9.   And it is possible that @thinkyhead has brought the other branches into C++11 compliance.   He does like doing that stuff.    Maybe I need to beg for a favor and see if he can get devel_ubl to be C++11 compliant?

Nocturnal42, 2016-10-28:
Delete the duplicate `#define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)` from Marlin.h and change all instances of `void debug_current_and_destination(char *title);` to `(const char title[])`  (in G26_Mesh_Validation_Tool.cpp, Marlin_main.cpp, Mesh_Buffer_Line.cpp and planner.cpp if I am remembering correctly). That will get rid of all the warnings... and I'm mostly certain it will still function correctly.

thinkyhead, 2016-10-29:
@Roxy-3D You will see the warnings if you enable the option to see warnings.

Roxy-3D, 2016-10-29:
> Delete the duplicate #define RAW_CURRENT_POSITION(AXIS) RAW_POSITION(current_position[AXIS], AXIS) from Marlin.h and change all instances of void debug_current_and_destination(char *title); to (const char title[]

>  Well, more like, I'm taking advantage of C++11 features like constexpr and static_assert.

I'll do all of this in the morning!  Thank you for pointing me in the right direction!   (I've been playing darts and won enough beers I'm not touching the code right now!)

JohnnyTheOne, 2016-10-29:
a wise Decision

closed 2017-10-31 by thinkyhead