MarlinFirmware/Marlin #5103: FIrmware does not read / use predefined home offsets and bed size from config

Netoperz, 2016-10-28:
I've noticed that i have set 190 190 170
and when checking from pronterface it is showing default setting from default config, even after resetting the eprom.

Also i have noticed that firmware will remember home position when using LCD menu .

When you manualy set the bed, the head and the rods (prusa i3) in min position 
(in my case home is all in MIN, and endstops are all in MIN)

And you will trigger the setting offsets in menu, than firmware remembers it.

But... still it is ignoring endstop signals.

I'm using RC bugfix.

Nocturnal42, 2016-10-28:
I can't help you with most of your problems, but with the endstops, you probably haven't enabled `ENDSTOPS_ALWAYS_ON_DEFAULT` in your Configuration_adv.h.

thinkyhead, 2016-10-29:
Resetting the EEPROM requires `M502` followed by `M500`. Did you do both?

thinkyhead, 2016-10-29:
> firmware will remember home position when using LCD menu

Which command in the LCD menu? (The "Set home offset" command is the same as `M428` or `M206`.)

closed 2017-10-31 by thinkyhead