MarlinFirmware/Marlin #5105: UBL obeying M206 when probing?

Nocturnal42, 2016-10-28:
It seems that the grid is being calculated using the hard coded axis minimum / maximum, but at the same time, while probing, its obeying the offset specified with M206. The end result being it tries to probe past the end of the axis.

Also is the LCD meant to be totally blank during probing? I really fell current position, and tally of grid points probed / remaining, would be extremely useful.

Roxy-3D, 2016-10-28:
> It seems that the grid is being calculated using the hard coded axis minimum / maximum, but at the same time, while probing, its obeying the offset specified with M206. The end result being it tries to probe past the end of the axis.

I think you are correct!   I think I need @thinkyhead to weigh in on what the correct resolution would be.   The problem is all of the bed leveling (including UBL) uses the lower level probe functions.    It would seem to me that physically probing the bed should be done in uncorrected coordinates.   

Until we can get consensus on how home offsets should be handled, they probably should be avoided.

> Also is the LCD meant to be totally blank during probing? I really fell current position, and tally of grid points probed / remaining, would be extremely useful.

That can be added to the LCD Display if it really would be useful.   But doing something like G29 P1 O serves the same purpose and gives much more detailed information about what is happening.

Nocturnal42, 2016-10-28:
When I probed the bed using G29 P1 M, it took more than 30 minute (Mainly due to the speed the Z axis is moving). During which time the screen was blank. That's bad. It means I have no visual indication of progress or ability to estimate time remaining.

When determing if M206 was being obeyed, being able to see the coordiantes that it was attempting to reach would have made things a lot easier. In general, I feel that being able to see where the printer thinks it is at any give time is a good thing.

Just showing the standard screen with current XYZ and temperature would be better than a blank screen.

Nocturnal42, 2016-10-28:
So now when probing I am getting the default screen, with the "Doing G29 UBL !" message. Which makes most of what I was complaining about irrelevant. So why was it blank previously? Related to #5106 perhaps?

Roxy-3D, 2016-10-28:
I don't know the answer of what happened in your case to cause the LCD Panel to be blank.   But I'm wondering if something other than UBL caused it to display a blank screen?    

The reason I'm puzzled is this:   When the UBL System needs access to the Encoder Wheel, it sets a flag to keep the LCD Panel code from seeing any values being generated there.  But the rest of the LCD Panel (status) code is running as it normally does.   The only thing that has changed from its perspective is it is being lied to and not told about any Encoder Wheel twisting and clicking.   The LCD Panel code thinks nothing is happening with its controls and just keeps displaying stuff as it normally would.

Nocturnal42, 2016-10-28:
It's not the cause my screen being blank that's concerning me right now, I'm positive that it's the quick access to z babystepping. What's concerning me is that it wasn't blank when I first loaded the firmware, and my attempts to recreate that (so I can test it with and without quick access enabled), have thus far failed.

Roxy-3D, 2016-10-28:
Do you have a RepRap Discount Full Graphics LCD Panel (for example) ????   Because the Kill pin is floating on that LCD Panel.   (Somehow, they would have to keep it from having an impact on Marlin's idle() function, maybe by doing a #undef or something.)    But the QUICK_ACCESS code isn't checking for anything other than a Kill pin right now.

Blue-Marlin, 2016-10-28:
https://github.com/MarlinFirmware/Marlin/issues/5106#issuecomment-256980275

Roxy-3D, 2016-10-28:
So does anybody know the reason why the Kill pin is looking like it is active?

Nocturnal42, 2016-10-28:
I can't look at the code right now, so I'm just going to ask the most obvious question, does it take into account the pin is active low?

Roxy-3D, 2016-10-28:
The KILL_PIN is coded to always be active low:
```cpp
  #if HAS_KILL
    #ifndef QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN

    // Check if the kill button was pressed and wait just in case it was an accidental
    // key kill key press
    // -------------------------------------------------------------------------------
    static int killCount = 0;   // make the inactivity button a bit less responsive
    const int KILL_DELAY = 750;
    if (!READ(KILL_PIN))
      killCount++;
    else if (killCount > 0)
      killCount--;

    // Exceeded threshold and we can confirm that it was not accidental
    // KILL the machine
    // ----------------------------------------------------------------
    if (killCount >= KILL_DELAY) kill(PSTR(MSG_KILLED));
    #endif
  #endif
```

Nocturnal42, 2016-10-28:
Could be, but Since it kills the printer, you really want to be sure. 

However, I was referring to the quick baby stepping code. This code isn't where the problem lies.

Roxy-3D, 2016-10-28:
Yes, the QUICK ACCESS code to start Baby Stepping just hijacks the same code that is used to detect the Kill Pin.   In Marlin_main.cpp it is set up as having a pull up so that should help with the noise:
```cpp
void setup_killpin() {
  #if HAS_KILL
    SET_INPUT(KILL_PIN);
    WRITE(KILL_PIN, HIGH);
  #endif
}
```
In the UBL code it is being looked for like this:
```cpp
#if ENABLED(BABYSTEPPING)
#ifdef QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN

  if ( !digitalRead(KILL_PIN)) {	// assume an active low pin
	  if ( currentScreen != _lcd_babystep_z)  {
		lcd_implementation_clear();
		lcd_babystep_z();
	  }
  }
#endif
#endif
```
I think we could try something like this to see if that gets rid of the noise:
```cpp
#if HAS_KILL
#if ENABLED(BABYSTEPPING)
#ifdef QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN
  static int killCount = 0;   // Filter noise on the Kill Button
  const int KILL_DELAY = 7;
  if (!READ(KILL_PIN))
    killCount++;
  else if (killCount > 0)
    killCount--;

  if ( killCount >= KILL_DELAY) )  {
    killCount = 0;
    if ( currentScreen != _lcd_babystep_z)  {
      lcd_implementation_clear();
      lcd_babystep_z();
    }
  }
#endif
#endif
#endif
```

thinkyhead, 2016-10-29:
> weigh in on what the correct resolution

There's been a lot of fixing and cleanup around this. Search the current codebase for `LOGICAL_POSITION`, `LOGICAL_X_POSITION` (etc.), `RAW_POSITION`, `RAW_X_POSITION` (etc.), and `RAW_CURRENT_POSITION` and you'll see all the places where the `home_offset`/`position_shift` need to be removed or added.

Regular expression: `(LOGICAL|RAW)_((CURRENT|[XYZ])_)?POSITION`

Nocturnal42, 2016-10-29:
Ah... Now I may have no idea what most of these really are, or what they are used for, but... if LOGICAL_POSITION is POS + offsets and shift, wouldn't that make RAW_POSITION just POS? Otherwise its just returning POS with negative offsets. Making RAW_POSITION negative 2*offsets from LOGICAL_POSITION.

Roxy-3D, 2016-10-29:
@Nocturnal42   That comment wasn't aimed at you.  @thinkyhead is explaining why the problem showed up and a path towards resolution.   

But given the fact the current thinking is we migrate the UBL System into an already fixed RC-8, does it make sense to make the changes to devel_ubl?   Or maybe we don't even bother because we know we will have a a devel_ubl branch based off of RC-8 soon?   

Scott, without really studying the problem, my inclination is to just not support M206 right now.   It will be supported soon enough when UBL is merged back into an RC-8 branch.   Right now...   It feels like a distraction to moving the UBL code forward.   Can you shake your head up and down or left and right?

Nocturnal42, 2016-10-29:
I know its not directed at me, I'm just trying to wrap my head around it. I suppose it makes sense if you are using them to derive the raw position from the logical position and vice versa, but that still seems very strange.

thinkyhead, 2016-10-29:
> my inclination is to just not support M206 right now

Sorry, but you kinda have to. It's not just `M206` but `G92` as well. Marlin supports an arbitrarily-shifted coordinate space, and coordinates given to commands like `G1` apply to the current coordinate space. Thus, any time we store coordinates for later use (i.e., in a generic sense) we must remove the offsets to the coordinate space. Only then can they be applied to the coordinate space later, when it may have been shifted by any number.

thinkyhead, 2016-10-29:
>  if LOGICAL_POSITION is POS + offsets and shift, wouldn't that make RAW_POSITION just POS? 

@Nocturnal42 It's actually like this: `LOGICAL_POSITION` is a raw position with offsets re-applied. `RAW_POSITION` is a logical position with the offsets removed.

Nocturnal42, 2016-10-29:
Yeah, where I was going astray mentally is that I was stuck thinking in the context of current position. Where the current position is always available in one form. What I wasn't thinking about was moving and planing moves.

closed 2017-10-31 by thinkyhead