MarlinFirmware/Marlin #5106: UBL 'Babystepping' at powerup

Nocturnal42, 2016-10-28:
So at the moment when I power up my printer, I get the marlin logo, then the screen shows something like "Babystepping 0.00" and after a few seconds goes blank (and stays blank).

I'm assuming this is because I started the process of creating a mesh, but reset the printer part way through the probing (since it was grinding on the end of the axis), and the UBL system is confused.

Other than the blank display, everything else seems fine (still responding appropriately to gcodes).

Roxy-3D, 2016-10-28:
This almost sounds like something is triggering the Baby Step mode incorrectly.   It might make sense to turn off the Z_BABYSTEPPING stuff in Configuration_adv.h until you get a Mesh defined.   

Did you turn on the QUICK_ACCESS_TO_Z_BABYSTEPPING in Configuration.h ?   It maybe that is falsely detecting your Kill button as being pressed?    

But let's get you up and running on a well defined mesh first.   And then we can figure out what is wrong with the Z_BabyStepping turning on when it shouldn't.

Nocturnal42, 2016-10-28:
I'm fairly certain I stuck pretty close to the included configuration, It is enabled, I don't recall if I did that or if it was the default. Reading the description, it doesn't sound like something I would enable. I tend to do dumb things, so being able to interrupt things when they go wrong is something I like. That being said, my board doesn't have a KILL pin, so no idea what it's thinking.

Nocturnal42, 2016-10-28:
Disabling it definitely fixes the issue. It definitely didn't do it the first time loaded the firmware, so is it the combination of the lack of a mesh (or enabling of the ubl?) and the lack of a KILL pin? 

Either way, it probably deserves a sanity check to prevent enabling without a KILL pin.

adamfilip, 2016-10-28:
I ran into this as well. enabling QUICK_ACCESS_TO_Z_BABYSTEPPING , leads to babystepping being the default screen, and you cant get rid of it. I ended up turning it off.

Nocturnal42, 2016-10-28:
What board are you using? does it have a KILL pin?

adamfilip, 2016-10-28:
Using a Rambo, not sure about the kill pin.

Nocturnal42, 2016-10-28:
according to pins_RAMBO.h, you do.

Roxy-3D, 2016-10-28:
This is valuable feedback!    A lot of boards have a Kill pin defined on them.   But some LCD Panels do not have a Kill button.   For example the RepRap Discount Full Graphics board.    It just leaves the Kill Switch floating:   

http://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf

Roxy-3D, 2016-10-28:
I guess the 'Correct' thing to do is turn off the QUICK_ACCESS_TO_Z_BABYSTEPPING if your LCD Panel does not have a Kill switch.    And I'll hurry up and try to get the "Press and Hold" of the Encoder Wheel functionality to turn on Z-BabyStepping going.

Roxy-3D, 2016-10-28:
> Either way, it probably deserves a sanity check to prevent enabling without a KILL pin.

Agreed.   But there is only one of me and I'm working to get the "Press and Hold" functionality going.   If you like to code, please take a look at how hard it would be to detect that both the LCD Panel in use and the controller board have a Kill function.    The Quick Access stuff is only a couple lines of code in a couple of places.   It might not be too hard to further qualify it with what ever logic makes sense.

Nocturnal42, 2016-10-28:
Sanity checks are pretty well implemented in the current RC branch, so it's not like this requires reinventing the wheel. Just stick this in pretty much anywhere and it should be good.

```
/**
 * Babystepping
 */
#if ENABLED(BABYSTEPPING)
  #if DISABLED(ULTRA_LCD)
    #error "BABYSTEPPING requires an LCD controller."
  #endif
  #if ENABLED(SCARA)
    #error "BABYSTEPPING is not implemented for SCARA yet."
  #endif
  #if ENABLED(DELTA) && ENABLED(BABYSTEP_XY)
    #error "BABYSTEPPING only implemented for Z axis on deltabots."
  #endif
  #if ENABLED(QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN)
    #if !defined(KILL_PIN)
       #error "QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN requires a board with a kill button"
    #endif
    #if ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
       #error "QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN is not compatible with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER"
    #endif
  #endif
#endif
```

Of course I am totally unfamiliar with hardware other than my own, so I have no idea of the other hardware than doesn't deal well with the kill switch, but it shouldn't be hard to add it on a case by case basis.

Roxy-3D, 2016-10-28:
I've added it to my Sanitycheck.h file.    The next time bump the code level, it should be on the GitHub server.    That probably happens when I get the "Press and Hold" Z-Baby Stepping working how I want it.

Blue-Marlin, 2016-10-28:
>  But some LCD Panels do not have a Kill button. For example the RepRap Discount Full Graphics board. It just leaves the Kill Switch floating:
http://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf

@Roxy-3D 
Please have a closer look. 
The RAMPS adapter:
http://reprap.org/mediawiki/images/7/79/LCD_connect_SCHDOC.pdf
tells us the one button is connected to from EXT2-8 to D41.
And pins_RAMPS.h is telling us: https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/pins_RAMPS.h#L241
We have a KILL_PIN at D41.
So the button labeled ''reset on schematics of the `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` is connected and used as the kill-button.

Same for 'REPRAP_DISCOUNT_SMART_CONTROLLER'.

Nocturnal42, 2016-10-28:
Wow. That adapter schematic is terrible. It looks like EXT1-10 is connected to D43 and EXT2-7 is connected to 5V. Which would be very wrong, except the labeling is clearly reversed on AUX3.

Blue-Marlin, 2016-10-28:
@Nocturnal42 
The hardware is pretty robust. :-)
So far the displays and the Megas survived all of my errors. Confusing EXT1 with EXT2, as well as crimping line 1 to 10 and 10 to 1 on the plugs. (But obviously it did not work as expected.)

Nocturnal42, 2016-10-28:
That's because it's all pretty much 5V tollerant. You can usually even get away with reversing GND and +5V, but Try accidentally hooking up VIN to something, and see how fast the magic smoke escapes.

Blue-Marlin, 2016-10-28:
About the risk of connecting backwards.
[image]
Find the difference.

Nocturnal42, 2016-10-28:
That's not a problem, just use IDC connectors without a registration peg. Then you can be certain that you will always plug them in backwards no matter what. 

I don't suppose anyone else is old enough to remember when this used to be an issue plugging in floppy disk drives.

Blue-Marlin, 2016-10-28:
Right. If plugged wrong the LED is permanently on. (VIC20 was my first darling.)

Roxy-3D, 2016-10-28:
Sort of posted to the 'wrong' thread.   I guess this really belongs in this thread.  Oh well...

https://github.com/MarlinFirmware/Marlin/issues/5105#issuecomment-257003159

Nocturnal42, 2016-10-29:
Testing with the linked code, I no longer get the babystepping on startup with QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN enabled. I can't test how it functions for someone with a KILL_PIN though, since my attempts to simulate a kill pin using a spare pin have not been successful. Hopefully @adamfilip can give it a try and see.

Nocturnal42, 2016-10-29:
Finally got my KILL_PIN working, seems to be working correctly with that as well.

Roxy-3D, 2016-10-29:
> Finally got my KILL_PIN working, seems to be working correctly with that as well.

Please pay attention to any random activations of Z-BabyStepping (due to noise problems).   But the problem is, Z-BabyStepping is set to timeout after 15 seconds so if it happens during a print, you may not see that when you look at the LCD Panel.   Specifically, I'm wondering if 7 is big enough to filter the noise?   How was the delay?   Was it unacceptably long?    If so, that 7 could probably be changed to 2 or 3.   It just depends on how noisy that signal line is.

Nocturnal42, 2016-10-31:
I haven't spotted it popping up, but then I haven't printed anything yet either. The delay was unnoticeable,  so it could probably be bumped out further if needed.

closed 2017-10-31 by thinkyhead