MarlinFirmware/Marlin #5107: Babystepping and Z offset

adamfilip, 2016-10-28:
Is there a way so that when you use baby stepping it will also show you what the new Z offset would be.

for example if my Z offset is -1.0 and I go into babystepping and make a -0.5mm adjustment it will show on the screen, underneath that my new Z offset will be  -1.5. once the screen turns off.

perhaps a feature to have babystepping adjust Z offset once the babystepping screen disappears.

Roxy-3D, 2016-10-28:
Hi Adam!    Right now I'm trying different things with the  Z-BabyStepping.    The problem with the BabyStepping is it really isn't integrated into anything else in Marlin.   Very literally, what it does is it goes and adds (or subtracts) stepper pulses to the motors.   Its almost as if you physically grabbed a motor and made it lose (or gain) steps in some direction.

It seems to me the amount you have Baby Stepped an axis should be cumulative on the display.   But the way the BabyStep code works right now is every time it is entered, it starts at 0.0000.

I ran into some weird issues trying to get the "Press and Hold" of the Encoder Wheel to turn on Z-BabyStepping.   I'm working through those issues right now but I'm definitely behind on the schedule of what I thought would be possible.

Once I get the "Press and Hold" function to work and trigger the Z-BabyStepping there are a number of things I want to try.    One idea is to shift the whole Mesh up or down (instead of messing with the Stepper Motors).   That way, if you like how things are printing, you can just save the updated Mesh.   

The other idea is we could update the Z-Offset instead of the Mesh or the Stepper Motors.  That would do exactly what you are asking for.   The LCD Panel would be displaying the updated Z-Offset as you change it.   I'm willing to try that, but my feeling is the Z-Offset should just go away.   It adds a lot of complexity and floating point calculations for no real benefit.   The one reason I'm hesitant to take it away is this:   "What happens if somebody's nozzle is above the 'Fade Height' and they try to Z-BabyStep up or down?"   The Mesh won't be able to contribute any correction because at that point it isn't being used.   So maybe in that case we still go right to the Stepper Motors and adjust them?   I don't know what the right way to handle the situation.

thinkyhead, 2016-10-29:
> it really isn't integrated into anything else

True. The purpose of `BABYSTEPPING` is to adjust the steppers in a way that's completely self-contained and transparent to everything else, as an on-the-fly adjustment.

closed 2017-10-31 by thinkyhead