MarlinFirmware/Marlin #5111: marlin 1.0 - compilation problem on win10 AR IDE 1.6.11

Netoperz, 2016-10-29:
Clean code downloaded from repo, just changed the bed dimensions and enabled full graphic smart controler, nothing alse.

So no one have noticed that bug Yet ?

In file included from sketch\SdFile.h:27:0,

                 from sketch\cardreader.h:8,

                 from sketch\Marlin_main.cpp:44:

SdBaseFile.h:38: error: using typedef-name 'fpos_t' after 'struct'

 struct fpos_t {

        ^

In file included from sketch\Marlin.h:10:0,

                 from sketch\Marlin_main.cpp:30:

c:\users\miros\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2\avr\include\stdio.h:950:33: note: 'fpos_t' has a previous declaration here

 __extension__ typedef long long fpos_t;

                                 ^

exit status 1
using typedef-name 'fpos_t' after 'struct'

esenapaj, 2016-10-29:
It's similar to this problem.
[Compiling issue - Arduino Mega](https://github.com/MarlinFirmware/Marlin/issues/4935)

thinkyhead, 2016-10-29:
This has been patched in the latest code, but was never patched in `1.0.x` because earlier versions of Arduino didn't have this conflict. I'll apply the patch to `1.0.x` and release it as `1.0.2-2`.

closed 2016-10-29 by thinkyhead

Netoperz, 2016-10-30:
Thanx, that will make many people happy :) 

Ilove you guys (i'm hetero) but still love you :)

cheers :D