MarlinFirmware/Marlin #5116: Feature Request: X-Y-Axis Squaring Adjustment

S3DCrations, 2016-10-30:
Hi Folks,

at first sorry for my poor english - but I will try my best. I'm completly new here but have several experience in 3D printing and building up my own cartesian maschines. So here ist my request:

Due to the fact, that a real 100 percentage squaring of the X and Y axis could not be really done by hand I thought it would be really practical to have a kind of an **Angle Offset Parameter** to be cahangable in Firmware. This parameter could be calibrated just like the steps per units parameters in Marlin by doing some messurements.

Doing a little math, this could easily achieved by generating new coordinates for X and Y from the GCode feedet to the printer:

**X_new = X / cos(alpha)
Y_new = Y - X * tan(alpha)**

alpha is the angle the Y-Axis versus X-Axis differ in positive movement direction (could be perfectly messured with a 90 dregree testprint). For example:

_The angle between Y and X ist 92 degrees --> alpha = 2_

Or

_The angle between Y and X ist 89 degrees --> alpha = -1_

I think it would be much helpful to define this Offset-Angle (alpha) in firmware for doing a fine calibration of squaring the Y-X-Axis. If alpha = 0 there would be no changes to the coordinates at all.

Hoping for resonations in any form and have a nice day
S3D

Blue-Marlin, 2016-10-30:
Duplicate of #4904, #3014

S3DCrations, 2016-10-30:
Hi,

sorry, didnt know about these oher requests. My opinion to #4904 is that the Prusa MK2-Way ist too specific to the printing machine itself for the moment. Once heated beds with induction points will become more available this could be interessting.

At the moment I was wondering why "the easy way" like I discribebed above and in #3014 wasnt already implemented. I searched for this way to calibrate so I started this request.

Adjusting the software_max_endstops could be done with this little calculation also very easy. Perhaps its possible to deliver a standard stl-file for any user for the XY-calibration (I could design something easy to print an meassure). The big adavantage ist that this method can be used by every printer just by giving the offset angle to marlin and youre done...

Im not this deep in Marlin to implement it by myself so I'm hoping ther are possibilities to implement it on the clean way just througt the team.

best regards
S3D

clefranc, 2016-10-30:
@S3DCrations Can it be done in the slicer instead?

S3DCrations, 2016-10-30:
@clefranc I have not found any option in any slicer I know for doing something like maipluate every single coordinate...

But I think also its a calibration problem witch depends on the printer istself and not to the stuff to print. Its just like the calibration of one single axis or the extruder. This time its the angle of X-Y.

Have attached the short calculation I made... sorry for the poor quality.
[image]

clefranc, 2016-10-30:
@S3DCreations I did not want to imply that slicers can do it right now, but 8-bit processors are already overused, so then asking the slicers communities is perhaps another option. It's only a suggestion, I did not thought about it very deeply.

Roxy-3D, 2016-10-30:
> I did not want to imply that slicers can do it right now, but 8-bit processors are already overused, so then asking the slicers communities is perhaps another option. It's only a suggestion, I did not thought about it very deeply.

This is an interesting way to solve the problem.   After all...  The Slic3r's are running on a processor with almost infinite processing power compared to the 16 MHz AVR processor!

S3DCreations, 2016-10-30:
@clefranc This is a good point, but I it schould use much less calculation power as the transform process to delta coordinates for example which can also be done easily.

Perhaps its possible to change the GCode with a script, but I think this would only be one step more unnecessary in the printing routine. But if someone ist able to write such a script I would be very interessted in testing ist :)

clefranc, 2016-10-30:
@alexrj Is a skew correction a viable option in slicers?

Blue-Marlin, 2016-10-30:
Theoretically the correction can be made in every step from CAD over slicer to host and firmware, or as a pre or post processing step of each of them. Most sense it makes in the firmware, because the 'error' is part of the machine.

looks as if it would cost a lot of time, but it is not. 
Because the angle is constant the sin/cos/tan can be precalculated. For angles below 2° the difference to 1 is less than 1/°°. If short on processing power this can be assumed as 1. What then remains is a multiplication and a subtraction per transformed coordinate.

(DELTA and SCARA do not have perpendicular x and y axes - they can not have __this__ error.)

S3DCreations, 2016-10-30:
@Blue-Marlin That sounds very positive :) But keep in mind that theese corrections only make sense for angles below 2°, because for bigger angles we could calibrate the printer mechanical.

Indeet the X-Correction is always muss less than the Y-Corretion... So for the X-coordinates taking cos(<2)=0,999=1 schould be not as noticable as for the Y-coordinates -> tan(2)=0,035

Greetings
S3D

thinkyhead, 2016-10-30:
> What then remains is a multiplication and a subtraction per transformed coordinate.

Currently the planner/stepper still works with an offset coordinate space (applied by `M206` and `G92`), but I hope to make it so that they only work with the "raw" coordinates for 1.1.1. At that time it will be slightly less expensive to handle the skew, and it will be done transparently at the planner level.

gcormier, 2016-10-31:
> Because the angle is constant the sin/cos/tan can be precalculated.

Reminds me of the good ol days. SIN/COS lookup tables in 3D gaming engines to speed things up :)

thinkyhead, 2016-10-31:
> Reminds me of the good ol days. SIN/COS lookup tables in 3D gaming engines to speed things up.

Indeed. Reminds me of the days of programming OpenGL directly using transformation matrices.

Blue-Marlin, 2016-10-31:
Maybe transformations in `Homogeneous coordinates` are not a that bad idea. Scaling of too big or small values is easy then.

Looking up sin() & co. is overkill here - just calculate once with the normal function and store the result. The angle changes are rare.

thinkyhead, 2016-10-31:
> just calculate once

Just like OpenGL transformation matrices. The reference point for applying the rotation factors has to be in the center of the coordinate space for simplicity. The Prusa MK2 branch has us covered. We just need to adapt it.

S3DCreations, 2016-10-31:
If the coordinates used in Marlin are absolute ones (like the GCode travel positions) the rotation factors should not depend on any position at all. Even for negative coordinates the result is correct (The difference in the Y-Calculation turns in a sum for negative X's).
The origin of my drawing above is the absolute position of X=0 and Y=0 for any Printer, no matter if this position ist the center of the printbed or the left front corner...

thinkyhead, 2016-10-31:
> the rotation factors should not depend on any position at all

I know how to rotate a coordinate system. In the general and symmetrical case, the fulcrum should be the center.

Blue-Marlin, 2016-10-31:
'Turning' is wrong anyway.
We don't want to turn a rectangle. We want to modify a trapezoid to a rectangle or the other way around. So only one axis has to be modified to fight the "x/y-axis are not perpendicular to each other" error.

Suppose x and y to be perpendicular. Now bolt the bed at the midpoint to the y-cariage. Now turn the bed by some angle. Now a part of bed-size will not fit on the bed anymore. That's what you fight with a rotation.

In the case we want to handle here, not perpendicular axes, you will get a turned trapezoid if turning both axis.

thinkyhead, 2016-10-31:
@Blue-Marlin Yes, quite right. The X axis will still be cross-wise, while the Y axis will be slanted.

S3DCreations, 2016-10-31:
>  I believe we still need to choose a reference point for where the slanted Y axis is centered

Im not a software engeneer but look at my post above. Mathematical an reference point should be obsolete. In any case there will be given a little(!) lost of room in Y-direction given by X_max * tan(alpha) due to the correction.

> Then there's the question of how to establish the amount of skew, whether by probing or manual input.

Is it possible to give both ways of Input. Manual input will work on every machine out there.
For the probing variant and since the original Prusa MK2 Heatbed is not offered stand alone for this time, I'd really like an open source variant of an Heatbed with induction-points.

a little bit off-topic:
In the next month I try to engeneer an open soucre PCB Heatbed based on the induction-points method. Therefore I have to figure out how many and where is the optimal position for these points. Since I'm not to deep in programming I'd need a little help of what is the best way of routine the probing through marlin.
But one Problem I have with this method of this "Prusa-Way": Whats if the heatbed isnt perfectly square to the Y-axis itself. Doest the Prusa code algorithm stand this an will correct in "both ways"?

Greetings
S3D

Blue-Marlin, 2016-11-01:
Usually the entire bed is reachable by the nozzle. Because inductive sensors can not be integrated directly into the nozzle they have an offset to the nozzle. Ideally the spots are all reachable by the probe. That means they can not be very close to the edges of the bed. On the other hand, best accuracy is reached when the spots are as far apart as possible - big distances amplify the displacement for small angular errors. There is no simple way to fulfil both. You have to set a value. Maybe 40mm from the edges is a good value.
You  can minimize the mess, when the probe offset is too larger, by having at least 9 spots. Regardless of how big and in witch direction the offset is, at least 4 spots are reachable. 3 is the minimum to determine the displacement and rotation angles. More is better because then you can average the measurements.
At the spots you can't heat. So minimizing the amount of spots is a good for the heat distribution.
One point should be in the centre of the bed. In a skewed only (not bent) bed the point in the middle has automatically the average (middle) height.
The positioning of the spots is critical. Their positions are the reference for the measurements. The don't have to be symmetrical to something or each other, but their positions have to be known as exact as possible. 0.1mm error is for sure to much. Errors below one micro-steps length don't make sense. We can't measure the difference. The exactness of the outer shape of the bed is completely uncritical.
The size of the spots is unclear to me. With my M8 inductive probe i can find one of the iron m3 mounting bolts in my alu bed (the others are not reachable). But centring on this ~6mm diameter spot is hard till impossible. A bit larger would be helpful. The M6 probe arrived but has no mount jet - so it is untested. I don't know if it will make a difference. My guess is, with smaller probe diameters and bigger spots centring will be easier. 10 - 12mm diameter could be usable. On the other hand big spots disturb the heat distribution more than smaller ones.
The gap between the spots and the next heating traces should be as large as possible to get the maximum signal difference - but as small as possible to not produce cold places.
If money would not matter i'd experiment with multilayer heater boars. Spots on the top side, hater traces at the bottom. Or a additional layer, in solid copper, to distribute the heat better. (No clue if this disturbs the spot detection.)
Varying the heater trace density around the spots and the edges of the bed may make the heat distribution better - or worse.
A solid alu bed with isolated copper traces like the Mk3 is probably a bad idea.
Glas on the bed is probably a bad idea. The datasheet trigger distance for the M6 probes is only 2mm. 2mm glas will be a bit brittle.
The PEI covers for the Prusa bed have been available from Prusa when i looked them up the last time. But in theory i'd like it a bit more rigid (used to a solid alu beds (Dauerdruckplatte)).
Thickness of the FIR? No clue. More increases the mechanical rigidity but probably will give more thermal warp, when one side is hotter than the other.

Sorry i don't have ready plan. But at least you should now know about most of the parameters to consider.

Blue-Marlin, 2016-11-01:
@clexpert 
How is your experience with the MK42 (new Prusa bed)? Durability of the PEI? Adhesion? Does the levelling always work? Is the heat distribution as good as promised?

Blue-Marlin, 2016-11-01:
The algorithm for calculating the correction angles is about that:
Take two points that theoretically be on the same x. With the known distance of the spots and their x difference build a triangle with a right angle and calculate the difference angle to the ideal line (alpha). 
Do the same for y and call it beta. If now alpha == beta we have a rotated only bed. If they are different, the difference is the error of the both axis to 90°. 
For the complete correction turn x by alpha and y by beta. Or if you want to save some time and the common rotation does not matter, turn only one axis by the difference between alpha and beta.

With the known distance between 2 spots and the measured distance between them you can also scale the steps/mm value. 

For how to find the spots centres take a look at Prusa article (http://prusaprinters.org/first-printer-to-automatically-correct-geometry-in-all-axes/).

Blue-Marlin, 2016-11-01:
If the bed is flat when unmounted there are only a few ways to bend it, if mounted near the edges. with 3 or 4 rigid mounting points.
Thermal expansion: If the bed is hotter on the upper side then on the lower side we get a convex bool shape at the upper side.
Mechanical compression between the mounting points results usually in a roughly cylinder wall shape. The direction is undetermined. If the compression is in both directions the result can be bool shaped, but usually does not, because the bow in the one direction is stabilising the other.
Load in z direction results in a roughly bool shape. If convex or concave depends on if the load is applied between or outside the mounting points.

All the possible shapes have in common:
There is only one maximum or minimum.
They steady. (else the bed is broken)
There is no turning point (f''(x,y) = 0)

Together that means the surface is relative good describable by 9 points.

A additional mounting point in the middle of the bed limits the max amplitude of the unflatnesses because the unsupported length are shorter, but makes the possible shapes much more complex. I guess you need about 25 points to get a rough description.

If the mid mount is the only rigid one, but the edge mounts are flexible in x/y direction the compression cases can not occur, but the possible shapes are not much simplified.

thinkyhead, 2016-11-01:
@S3DCreations @Blue-Marlin I presume we've all seen Prusa's videos on the subject of bed skew compensation and mesh leveling. If not… https://www.youtube.com/watch?v=rYrLT5G-a9I

Blue-Marlin, 2016-11-01:
One day, when we have unlimited processing power, we should try to use splines to connect the measured spots. The steadiness (in f'(x,y)) of the surface would be much improved, compared to connected straight lines.  More like the real worlds physics. No sharp folds anymore. (just to prevent patents - if not already invented ;-))

fiveangle, 2016-11-01:
Skeptical myself about the MK42 HB,  and Prusa MK2 in general, now having toyed with it for about a month, I can say I am thoroughly impressed how Josef has embraced the notion of the old hardware engineer adage, "we'll fix it in software."

The MK42  heated bed isn't even flat on delivery, and  only marginally better once mounted onto the machined Y carriage frame.  Yet with just nine induction points across a 250mm x 210mm  surface, this thing prints a flatter and more accurate first later than my old Printrbot Simple Metal (aka, "the tank") with 1/4" thick Mic 6  machined aluminum 150mm x 150mm heated bed.

After assembly I had to take the XY Frame apart to replace  something, and as a test ( or more a test of my patience), I put it back together purposely skewed about 2mm on each axis.  The result was a nasty, "Axes severely skewed!  Will compensate…"  after calibration routine.  This thing is skewed in every way except that the bed is square with the Y axis movement, and yet produces absolutely perfect prints  indistinguishable from when it's assembled perfectly (which is only possible after eight hours of continuous fiddling).

Having assembled this i3 design, I don't see how he could've fixed it any other way:  The design is crap;  sure, it includes the ability to adjust every possible angle, but it also gives no fine control to do so.  If you make one small adjustment to any of the grossly course threaded rods,  every other angle is thrown off as well.  It is nearly impossible to get it perfect without some kind of jig, or the aforementioned tenacity to keep trying until it's right. 

I really have to hand it to him:  he fixed his raprap design in the only way it could have been while still keeping with the spirit of the reprap  community.

Of course this let me directly to disappointment, in that there's absolutely no way to run vanilla Marlin without a severe downgrade in print quality (my interest in LINEARADV  is what brought me here).

In summary, automatic skew correction and mesh bed leveling is the magic bullet target everyone should strive to hit:  it just works.

adamfilip, 2016-11-01:
So how do it accomplish this. So everyone can benefit even those without inductive probes

thinkyhead, 2016-11-01:
> no way to run vanilla Marlin without a severe downgrade in print quality

@fiveangle Of course, now we're incorporating aspects of MK2 into the main branch. The mesh-based `AUTO_BED_LEVELING_BILINEAR` now supports cartesians. Their jerk and acceleration code enhancements have been adapted. And `LIN_ADVANCE` as of today is much more solid. I'd be interested to know how well `RCBugFix` (soon to be `1.1.0-RC8`) works on the MK2. I have a self-sourced Prusa i3 myself, and am looking forward to trying the latest code on it when my travels are done.

S3DCreations, 2016-11-01:
The question apart from the software calibration process for me is: is there a need for an open source heatbed with induction points or not. I'm thinking here on other sizes (200x200, 200x300, 300x300) but it has to be open source and setting a standard for the induction probe method. Therefore should be given a method to define the type of "induction-spot-satndard" in firmware. Perhaps just like setting the right thermisor also setting the right probe-method (matrix for finding the points; or just "copy" the Prusa points?) for the heatbed. And that would be firmware related stuff.

I have a lot of ideas how to design such a pcb and how a standard could be established. If there ist enough interrest we should maybe start a subdiscussion which is related to this discussion.

Sorry for beeing off-topic again, but I think the devellopment of a heatbed has to go hand in hand with the calibration process at all.

casper2002casper, 2016-11-01:
Couldn't you get the probe points by finding the corners of the printbed? The distance from those points should be known.

thinkyhead, 2016-11-01:
It would have limited utility due to the wide variety of bed sizes. But I guess that Prusa Research will probably be selling their custom PCB on its own without a printer, in which case we can at least add support for that specific arrangement.

S3DCreations, 2016-11-01:
@thinkyhead therefore I said lets find a standard for the points... for example 9 points arranged in a square with known distances.

for example: 
Bed 200x200: 5cm (point-to-piont)
Bed 300x300: 8cm (point-to-piont)
Prusa M42: ??cm

These standards could be added to firmware and just need to be defined through the user. Just like setting the right thermistor. Or even a matrix where the user can define the coordinats for the points by himself from center as reference...

This would open up the developping of these kind of new heatbed PCBs for the community and speed up developpment of 3D-printing technology... nearby making Marlin the first choice - even for the original Prusa MK2...

epatel, 2016-11-01:
@S3DCreations I think it actually depends on the bed. A perfect flat bed doesn't need any mesh points at all. And I think its also a matter of _inconvenience_. The MK2 seem to do a probe cycle before avery print but I use EEPROM to store the mesh so I maybe do a probe sequence once a month. More points takes longer time etc.

S3DCreations, 2016-11-01:
@epatel inconvenience could be a reason for some users. For me its a way of fine-tunig the squaring process for getting the maxium out of the printer when adjusting by hand is no more possible. Just like the Z-correction bedlevelling.

Anyway... inconvenience or not... it should help making the prints more exact an that would help the technology to spred out in the low-cost segment... And nobody has to use ist if not wanted, just like Z-probing...

adamfilip, 2016-11-01:
Could it be a removable board with special dots on it, you lay it on top of your print bed.. clip it down to your glass and do a measurement, then remove when no longer needed..

I assume you wouldnt need to do this very often.
and this way many people can easily add it to their machine

Piece of glass with Screenprinted metal dots?

"Skew Calibration Plate" :)

S3DCreations, 2016-11-01:
> Could it be a removable board with special dots on it, you lay it on top of your print bed.. clip it down to your glass and do a measurement, then remove when no longer needed..

Interessting idea of a surface just for calibration... but we want the this procedure to get the maximum accuracy. If positioned such a Plat manually to the bed we have again the human factor and its imperfections... in other words: you would have to calibrate the calibration plate on the heatbed even better as your axis...

but as i said: nobody has to use this... and you can tell the firmware by hand to calibrate in the printers menu or in an intervall of serveral prints - dont have to use it every time.

adamfilip, 2016-11-01:
if you square up two edges of the calibration plate with the glass bed 
it "should" be accurate enough.., 

couldnt you do the same with just a carpenters metal Square
put it on the two edges and have a prox probe detect the edge. and see how square it is??

thinkyhead, 2016-11-01:
> These standards could be added to firmware and just need to be defined through the user.

Surely. When I do the integration that feature from the Prusa MK2 code I will follow the usual procedure – make it a more general feature and make it fully configurable.

S3DCreations, 2016-11-01:
> if you square up two edges of the calibration plate with the glass bed 
it "should" be accurate enough..,

the question is what is "enough" wenn software cann do a way more exactly for the last 1/10 degrees...

> couldnt you do the same with just a carpenters metal Square
put it on the two edges and have a prox probe detect the edge. and see how square it is??

Of course you can... and you should! Get the maximum by hand and let the software do the rest wich you cant...

adamfilip, 2016-11-01:
I wasn't really talking about do what you can by hand to make the machine as Square as possible even though you should but I meant is put something down on the bed that is square and then have the machine read how Square it is to compare to its own movement

S3DCreations, 2016-11-01:
@adamfilip Okay, now I got your point... Let me just think about it:

Since rotation of the heatbed and therefor also the rotation of the "Calibration Plate" is not relevant for the calibration processitself we schould be able to messure the squareness of this plate even if it is rotated... So if there is is meassured a rhombus we could calculate the offset-angle for the printer...

is this the idea?

If so we only need an algorythm for finding the first induction spot, then do a circular search (since we know the spot distance) around this spot for finding the next two ones and so on...

Roxy-3D, 2016-11-01:
> Could it be a removable board with special dots on it, you lay it on top of your print bed.. clip it down to your glass and do a measurement, then remove when no longer needed..

I think this is a good idea!   Because if the calibration plate can be removed,  the standard heat beds can still be used and the heat distribution won't be affected.    And I'm all for automated probing, but it would seem a machine without a probe could still be calibrated this way.  You would just very carefully move the nozzle to each calibration point and say "Its there!"    

Wouldn't it be nice if you could just print the "Calibration Plate" on your Paper Printer and tape it to the bed to get your machine's numbers?   You do it once, and just lock the numbers into the firmware for future use.

S3DCreations, 2016-11-01:
Yeah, thinking about this the last half an hour I'm also on this way to go... especally becaus all kinds of printsurfaces would be possible...

and it can nearly be fully automated... Just put this plate near (not exact) to the middle of the heatbed. The Printer serarches for the predefined points on the calibration plate... calculates out the rotation if needed and gets the angle of the rhombus... thats it...

The plate has to be a flat (therefore little bit thicker against bending over time) PCB with etched-in induction points... only problem here is the z-offset for the thickness of this PCB, but there schould be a secured offset parameter just for the probing to be figured out...

The printer can also remind you to do the calibration probe after a defind numer of prints too, to keep his his offset angle in firmware up to date....

adamfilip, 2016-11-01:
I use roland wide format vinyl printers/cutters at my work and these use optical sensors to detect printed ink black dots. It measures 4 of these dots to do a skew/scale correction. I'm not sure exactly what these sensors are. Seems like a very crude device that can see contrast. Doesn't look like a camera

S3DCreations, 2016-11-01:
Meassuring a printed sheet of paper would be fantasic... but i have a little bit of fear of optical sensors...

-how acurrate are they?
-how will the ambient light affect the meassuring?
-need for an extra sensor on the prinhead.

I will have to get aound of the possiblilties of optical sensoring ;)

Roxy-3D, 2016-11-01:
> Measuring a printed sheet of paper would be fantastic... but i have a little bit of fear of optical sensors...

All valid concerns...   But what this would do is enable even the very lowest level machines to have their skew corrected if the owner was willing to print (on paper) a calibration sheet and run the nozzle tip around the bed to 'sample' points.   It really would be great to have a solution that handles the entire spectrum of machines.

adamfilip, 2016-11-01:
Aligning the nozzle over 4 points on a printed pieces of paper that is taped down to the bed. After aligning one side of it.  Can be tricky perhaps if it was a pin or something very sharp.  It can make it easier to do very precise positioning

Roxy-3D, 2016-11-01:
In the crude, very primitive case...  Even taping a toothpick to the nozzle would probably work. (You could not let the taped toothpick touch the bed because it would shift its position!)   And regardless of how the 'Calibration Plate' was taped to the bed, it would seem the algorithm should come up with the same correction numbers for the skew.

S3DCreations, 2016-11-01:
Perhaps it would be better to do this manually... manual find the spots with the nozzle and let the printer collect the coordinats by pressing a button... or type the coordinats of the dots directly into the firmware...

so there is no need for a automated search for the paper method...

Roxy-3D, 2016-11-01:
> so there is no need for a automated search for the paper method...

Agreed...   And in fact, without a probe...  The printer really can't do an automated search.   The whole idea with taping a toothpick or needle to the nozzle was aimed at still letting the user be very precise because they could see exactly where the 'indicator' was pointing.

S3DCreations, 2016-11-01:
By the way... manually or automated... why we dont calculate correction facors for travel ways of X and Y at the same time within the squaring calculation?

adamfilip, 2016-11-01:
This sounds great. Lets get the Manual Skew correction working first with paper and then once the guts of the code down. then we can revisit automating it with a probe or optical or whatever

epatel, 2016-11-01:
How about this? Print a number of lines in a "fan" way, and then entering which line that lines up best from the edges of a piece of paper? Maybe first print a fan with a higher "tilt" per line, measure and then print a new pattern with a smaller tilt until one is done? UBL can print patterns?

[image]

Roxy-3D, 2016-11-02:
>Maybe first print a fan with a higher "tilt" per line, measure and then print a new pattern with a smaller tilt until one is done? UBL can print patterns?

Yup!  UBL can print patterns.   This is very 'do-able'.

epatel, 2016-11-02:
If one starts with 7 lines as above and have a 2˚ spread per line the first fan would cover ±6˚. Then the next pattern could cover ±2˚ (0.6666667˚ per line) etc. The middle line and pattern should be adjusted from the previous "round". Not sure how many times would be good, but until one can't tell them apart I guess.

Roxy-3D, 2016-11-02:
For 1° spread per line on a 200mm x 200mm bed, you get:   cos(89°)= .01745 and sin(89°)= .9998477 
 
.01745 * 200m = .349mm   We could print 'dashed' lines and stagger the dashes so they don't merge and interfere with each other.

epatel, 2016-11-02:
though I think 0.01745 * 200mm is 3.49mm (thats a lot so maybe sufficient to start there ±1˚?)

adamfilip, 2016-11-02:
Doing the line method seems like it would be more steps. Do you believe this would be more accurate then the paper/toothpick method?

epatel, 2016-11-02:
@adamfilip Sure it might be more steps, but it is less "fiddly" I imagine. The pattern should print very quickly and be easy to remove. Maybe more "robust" as one narrow in on a satisfactory skew. Well, just my opinion.

[47 more replies]

closed 2018-03-19 by thinkyhead